PX4/PX4-Autopilot #21255: gazebo(ignition) error :

xmfxhddl, 2023-03-07:
hello. I am running a gazebo(ignition) simulation in environment docker container(11.8.0_cudnn8_devel_ubuntu22.04 base image), and the problem occurs. 

### error code ↓↓↓↓
I tried this command
`root@lig-MS-7B17:~/PX4-Autopilot# make px4_sitl gz_x500`

and occur error like this.

`libEGL warning: egl: failed to create dri2 screen`

how to solve the problem?...

I want someone to help me. please...

TheotimeBalaguer, 2023-03-15:
Hello,
This is not a PX4 warning, but a Gazebo warning. You can refer to [this issue](https://github.com/gazebosim/gz-rendering/issues/587), which states that the problem comes from Ogre. 
However, this is only a warning and doesn't affect Gazebo's behavior !

PoyrazTahan, 2023-11-16:
@xmfxhddl did you have a documentation on how you handle the issue?