PX4/PX4-Autopilot #21284: PX4 SITL offboard control problem when using different SITL instances [ROS2 humble & gz-garden]

zp-yang, 2023-03-10:
I am having problems with offboard control when launching px4-sitl with an instance number other than 0 i.e. `./bin/px4 -i 2`. The offboard control script, (for example [[cpp](https://github.com/PX4/px4_ros_com/tree/main/src/examples/offboardl) and [python](https://github.com/PX4/px4_ros_com/tree/main/src/examples/offboard_py)]) could not arm or change flight mode on the SITL vehicle.

However, if I use commander to change flight mode to offboard and takeoff, then run the offboard control script, the SITL vehicle moves to the expected trajectory setpoint. 

I am not sure what could cause this problem (DDS agent? or px4_sitl?) and hope someone could point me to a good direction. Thanks!

## To Reproduce
- Clone [px4-ros-com](https://github.com/PX4/px4_ros_com) and change all instances of `/fmu` topic publication to `/px4_2/fmu` in `offboard_control.cpp`. Build and source the package with [px4_msgs](https://github.com/PX4/px4_msgs) 
- Build px4_sitl
- Start MicroXRCEAgent on another terminal with `MicroXRCEAgent udp4 --port 8888`
- Launch px4_sitl with `./build/px4_sitl_default/biin/px4 -i 2`
- Run the offboard control node `ros2 run px4_ros_com offboard_control`. SITL vehicle is not moving

## Expected behavior
If start px4_sitl with `./build/px4_sitl_default/biin/px4 -i 0`, the vehicle in gz simulation can be seen being armed and move to the trajectory setpoint. This should be the expected behavior when starting with `-i 2` as well.

## Drone (please complete the following information):
- SITL gz-garden x500

beniaminopozzan, 2023-03-10:
Hi @zp-yang !

Please see https://github.com/PX4/PX4-Autopilot/issues/21085

You are sending commands with the wrong `target_system` value.

zp-yang, 2023-03-10:
Thanks, that explains it. How is `target_system` determined? So for instance 0 `target_system=1`, for instance `target_system=2` and so on so forth?

beniaminopozzan, 2023-03-10:
precisely!

closed 2023-03-11 by zp-yang