PX4/PX4-Autopilot #21290: pixhawk 6c Failsafe Enable: No local position estimate

amovlgf, 2023-03-11:
Using pixhawk 6c flight control brush to write px4 v1.13.1 firmware, the following error:
check for a valid position estimate
Failsafe enable :No local position estimate
Accel 1 clipping, not safe to fly!
Land now, and check the vehicle srtup. Clipping can lead to fly-aways.
Stopping compass use! Land now and calibrate the compass

[image]

[image]

ryanjAA, 2023-03-23:
I’d jump over to the discuss server and upload the log link.

amovlgf, 2023-05-03:
The log file is large, please download it here：
link：https://pan.baidu.com/s/1zVFNuegf2NaUBpm7PH9t4g?pwd=1111 
password：1111

jyf-ai, 2024-01-28:
I also encountered this situation,has this problem been solved?