PX4/PX4-Autopilot #21304: Gazebo (ignition) speed factor for faster than realtime

ghfue, 2023-03-13:
## Describe problem solved by the proposed feature
jMAVSim and Gazebo Classic support the `PX4_SIM_SPEED_FACTOR` environment variable for faster than realtime simulation. The documentation on running [faster than realtime](https://docs.px4.io/main/en/simulation/#run-simulation-faster-than-realtime) says jMAVSim and Gazebo Classic are the only simulators supported. Initial testing with Gazebo showed no speed improvement with the environment variable. It would be nice to have this for Gazebo (Ignition).

## Additional context
I tested with `PX4_SIM_SPEED_FACTOR=10 make px4_sitl gz_x500` and saw no speed difference in simulation.

Jaeyoung-Lim, 2023-03-14:
> I tested with PX4_SIM_SPEED_FACTOR=10 make px4_sitl gz_x500 and saw no speed difference in simulation.

@ghfue This is not a supported feature for gazebo yet

fabrizioschiano, 2023-08-31:
Hi @Jaeyoung-Lim , could you confirm that this would not work even with a `PX4_SIM_SPEED_FACTOR < 1 ` ? 

When you say that this is not "supported for gazebo yet", I have two questions: 
1. Is there someone working on this already? 
2. I remember this working for older versions of Gazebo, is that correct? 

Btw, I think that this should be somehow specified in the PX4 Dev User Guide because we struggled a bit before finding this issue and understanding that maybe this is not our mistake but it is just not supported.

Jaeyoung-Lim, 2023-09-01:
> 1. Is there someone working on this already?

Unfortunately Not, but would be a great contribution.

> 2. I remember this working for older versions of Gazebo, is that correct?

Yes, however the way PX4 is integrated to the two simulations are very different. Gazebo classic uses lockstep with the `MAVLINK_HIL_* ` messages. Gazebo (new gazebo) directly talks to ignition transport through ignition transport

Jaeyoung-Lim, 2025-02-05:
@GreenPenguin77 Unfortunately there is not

nshahrestani, 2025-07-29:
Are there plans to support this feature in the future?

dakejahl, 2025-07-29:
https://docs.px4.io/main/en/sim_gazebo_gz/#change-simulation-speed

closed 2025-07-29 by dakejahl