PX4/PX4-Autopilot #21315: ADSB-IN   configuration for ORANGE CUBE PIXHAWK  under QGC GCS

sibujacob, 2023-03-15:
As discussed today...

Thanks!

dagar, 2023-03-15:
@julianoes @mrpollo let's make sure the Cube Orange (and Orange plus presumably) is configured so that ADSB works by default (forwarded to GCS).

julianoes, 2023-03-21:
I can't verify this right now but it looks like it's all set up correctly.

There is a mavlink instance listening:
https://github.com/PX4/PX4-Autopilot/blob/main/boards/cubepilot/cubeorange/init/rc.board_mavlink
https://github.com/PX4/PX4-Autopilot/blob/main/boards/cubepilot/cubeorangeplus/init/rc.board_mavlink

And the [ADSB_VEHICLE](https://mavlink.io/en/messages/common.html#ADSB_VEHICLE) message should be received:

https://github.com/PX4/PX4-Autopilot/blob/22c94f805a372783d670bcaf53ca736c86dcd7a5/src/modules/mavlink/mavlink_receiver.cpp#L2462-L2502

And then sent off again:
https://github.com/PX4/PX4-Autopilot/blob/22c94f805a372783d670bcaf53ca736c86dcd7a5/src/modules/mavlink/streams/ADSB_VEHICLE.hpp#L68-L104

Was there a specific report where this was not working as expected?

sibujacob, 2023-03-21:
Hi.....Nice to hear the ADSB work....   Is this ADSB  integrated with PX4 v1.13.3 or PX4 v1.13.2  ....thank you

julianoes, 2023-03-21:
When I get a chance next week, I'll try this.

julianoes, 2023-03-26:
[image]

Tested with future v1.14 and v1.13.3

closed 2023-03-26 by julianoes

sibujacob, 2023-03-27:
thank you dev- team. Great work....