PX4/PX4-Autopilot #21327: Flyaway after loiter to altitude

kibidev, 2023-03-17:
## Describe the bug
While flying a VTOL mission in simulator with WPs with large altitude difference close together (to provoke loiter to altitude), I experienced that the drone after completing loiter to altitude (WP 4) selected the WP after the next (WP 6, but currently unsure if WP 5 was briefly selected in between) but continued straight east, but adjusting the altitude to that given in WP 5.

## To Reproduce
I have not been able to reproduce yet, but the steps I did when got the error is given below.

Steps to reproduce the behavior:
1. Started simulator `make px4_sitl gazebo_standard_vtol`
2. Uploaded mission [bumpy.zip](https://github.com/PX4/PX4-Autopilot/files/11000338/bumpy.zip)
3. Started mission
4. Error after WP 4, as described/shown in screenshot

## Expected behavior
The drone should have gone to WP 5 and loitered to corrected altitude after WP 4.

## Log Files and Screenshots
https://logs.px4.io/plot_app?log=d5fb6ec1-4393-406c-904d-77450f2c6b4f
[image]

## Drone (please complete the following information):
- Simulator, gazebo_standard_vtol

## Additional context
Used latest main branch version:
[0ae296bfe21add6276f5c28a97884059fb317999](https://github.com/PX4/PX4-Autopilot/commit/0ae296bfe21add6276f5c28a97884059fb317999)
Build/run in px4io/px4-dev-simulation-focal docker image.

After the screenshot was taken, I selected WP 6 and continued mission from there. No unexpected behavior after this.

Jaeyoung-Lim, 2023-03-17:
@kibidev Behaviorwise this looks like a similar problem that https://github.com/PX4/PX4-Autopilot/pull/18996 fixed

kibidev, 2023-03-17:
Got an other data point:
[image]
[image]

It actually changes WPs here, but is flying at wrong position in a line, when it should be loitering.

I will update logs and plan soon.

kibidev, 2023-03-17:
> @kibidev Behaviorwise this looks like a similar problem that #18996 fixed

Thanks @Jaeyoung-Lim, this does indeed seem relevant.
I will investigate a bit more, and try to pinpoint which versions are impacted.

kibidev, 2023-03-17:
I have not been able to reproduce this in 1.13.2, so I believe some change after this is the cause, but I can not be 100 % sure.

kibidev, 2023-03-17:
Using bisect, I have found the commit/PR that first causes this behavior:
https://github.com/PX4/PX4-Autopilot/pull/21168

I have not studied it in details, but I suspect that there is some other older code that is the actual error, and this just stopped hiding the problem.

closed 2023-05-26 by sfuhrer