PX4/PX4-Autopilot #21335: Mavros/imu/data rostopic giving empty messages on pixhawk cube orange + when barometer and magnetometer are turned off

Astik-2002, 2023-03-20:
I am currently working on a drone which relies on VIO for state estimation. Since this uav has to operate in highly magnetic environment, we have to turn-off the magnetometer and barometer to avoid compass error. 

The VIO to px4 bridge (Fast-Drone-250/src/realflight_modules/px4ctrl at master · ZJU-FAST-Lab/Fast-Drone-250 · GitHub) uses mavros/imu/data rostopic to recieve rotation quaternions from imu. For older pixhawks (cube black, orange), this rostopic is publishing data (without orientation), which is augmented by orientation recieved from odometry. But for cube orange +, the messages are just empty.

This issue is only happening if compass and barometer are turned off in cube orange +. 

Pinned below is the drive link of videos showcasing the issue. Mavros/imu/data is publishing messages (for cube orange with magnetometer off) but is empty for cube orange + under same condition.

https://drive.google.com/drive/folders/1DLsRW9gPwCZH5RYpC8scv6ZJoK9JXVVW?usp=sharing

We are using px4 version 13 for this. Kindly help me solve this issue. Thanks

julianoes, 2023-03-20:
Is this issue related to the Orange+ or does the same happen on Orange?

EhrazImam, 2023-03-20:
This issue is with cube orange plus and pixhawk 6c
Everything runs good with cube orange and pixhawk 4

julianoes, 2023-03-20:
Ok, it's not easy for me to replicate the setup so I suggest to figure out what exactly is sent differently from the CubeOrange+ and Pixhawk 6C.

Astik-2002, 2023-03-23:
How do you suggest we do this? It is not exactly clear what is causing this difference from cube orange to orange +/6c

julianoes, 2023-03-23:
I would try to compare the topics published using `uorb top` for both cases.

EhrazImam, 2023-03-27:
@julianoes Any update??

julianoes, 2023-03-28:
See what I wrote last.

julianoes, 2023-04-10:
Another way would be to create to ulg files for both cases and then upload both to logs.px4.io and paste the two links here and we can compare what is going on.