PX4/PX4-Autopilot #21358: v1.14 Release Log Archive

mrpollo, 2023-03-22:
Hey community I made this issue to archive the flight logs contributed by the community; please reply to this issue with links to flight review flights with v1.14

ryanjAA, 2023-03-22:
https://review.px4.io/plot_app?log=fead427b-23bf-426b-8138-30764197aeaf
https://review.px4.io/plot_app?log=6efcab43-8f73-4e8a-81a4-e67a697f29fd
https://review.px4.io/plot_app?log=a20c78a1-9737-43af-b48b-08cdbe895830
https://review.px4.io/plot_app?log=e3852b37-1f69-481d-97b4-21625e77f09c
https://review.px4.io/plot_app?log=85e7018e-6980-4ab5-9efb-29247b1e8a40
https://review.px4.io/plot_app?log=9fd4e98c-04af-493b-8387-34a6eac902cb
https://review.px4.io/plot_app?log=4baefb1b-15d1-440e-b05f-526d58552fa5
https://review.px4.io/plot_app?log=6ddeef40-9571-4a8c-93a2-92b7d70eae46
https://review.px4.io/plot_app?log=755cb3f0-10c8-4f3f-8c24-c1a79b08ad87
https://review.px4.io/plot_app?log=d0fd56d0-e8f3-46fb-8384-62a2c50a91f7
https://review.px4.io/plot_app?log=77c5b3b3-ea2f-4076-94ae-0dcccb28e71e 
https://review.px4.io/plot_app?log=a03594eb-b403-4b60-8569-3c227f4d07d6
https://review.px4.io/plot_app?log=939e4474-6daa-43a9-8fac-d8d98f07bc17
https://review.px4.io/plot_app?log=57413060-21da-4ab2-974f-016fbc8ba25c
https://review.px4.io/plot_app?log=93c5fae3-9280-4309-9f69-c17c75ee8fa6
https://review.px4.io/plot_app?log=08df4199-bccd-4745-8768-859f3a802ec8
https://review.px4.io/plot_app?log=5c3da1c5-e1ee-4ddd-82ba-7942f3d36b05
https://review.px4.io/plot_app?log=ef2e27f1-7329-4db7-9373-dd2bf2b70ccc
https://review.px4.io/plot_app?log=d32f3cb8-dc3e-46dc-aad1-1607cc1bb895
https://review.px4.io/plot_app?log=353f4df5-1405-4d38-a6ef-17faf974cd4f
https://review.px4.io/plot_app?log=4b9da507-81ac-4afa-8814-c204da6f7b56
https://review.px4.io/plot_app?log=131f84d2-639a-44b8-b87d-09c52dc039fc
https://review.px4.io/plot_app?log=f7130c9e-0a0d-421f-927f-e15f1c9736ee
https://review.px4.io/plot_app?log=48b6def5-fcd1-441f-88e6-a29e1602f3b2 
https://review.px4.io/plot_app?log=300c4368-3353-4eef-940e-b4c1221cb2f7
https://review.px4.io/plot_app?log=ba4f9c24-b70e-4da5-a7c5-805e1e0fcdb9
https://review.px4.io/plot_app?log=a7442044-0512-4f27-9f01-6abe8922d4ee
http://files.appliedaeronautics.com:5006/plot_app?log=3918377b-8858-4582-8956-ceabb818cb97
https://review.px4.io/plot_app?log=582ad5c8-a030-4e1c-83cc-8b9e12e7adaf
https://review.px4.io/plot_app?log=ab7e959e-636c-4667-bf6e-72fc6ee46fdc
https://review.px4.io/plot_app?log=927b658a-da8c-4260-9e2d-ac9461f042be
https://review.px4.io/plot_app?log=300c4368-3353-4eef-940e-b4c1221cb2f7
https://review.px4.io/plot_app?log=4ef94126-6c11-421d-9d9d-91084c0ccad9
https://review.px4.io/plot_app?log=8551e5c9-2262-4ce5-b339-a4c1e4344ba8
https://review.px4.io/plot_app?log=ef794429-7b0c-4452-946c-0024cc90d565
https://review.px4.io/plot_app?log=8551e5c9-2262-4ce5-b339-a4c1e4344ba8
https://review.px4.io/plot_app?log=ef794429-7b0c-4452-946c-0024cc90d565
https://review.px4.io/plot_app?log=0cfb33f6-a8ad-436a-9aa9-dc46408b5766
https://review.px4.io/plot_app?log=ea406bd3-3169-4734-b9af-1ad5303d2984
https://review.px4.io/plot_app?log=ac0775d6-84c2-4def-8641-7ecaafd7547e
https://review.px4.io/plot_app?log=7939dcb5-2a7d-4a49-ac00-d5588ec22934
https://review.px4.io/plot_app?log=700e86d5-b7d9-482f-b7e9-9761c6238048
https://review.px4.io/plot_app?log=6a67a36a-ab61-4672-945d-b4a25b0c0494
https://logs.px4.io/plot_app?log=c52a816e-3cf9-42e4-9384-a3946a59504f
https://review.px4.io/plot_app?log=2053c052-d1d7-4337-878d-7bb9f795c105
https://review.px4.io/plot_app?log=6a67a36a-ab61-4672-945d-b4a25b0c0494
https://review.px4.io/plot_app?log=97186db0-1043-4cb8-95e6-7dd1853bdf81
https://review.px4.io/plot_app?log=f5fd2ecf-8b1f-4c77-baab-ab556bb2a6c0
https://review.px4.io/plot_app?log=5eed8abb-02a9-47c4-bc85-56d0ac8f684d
https://logs.px4.io/plot_app?log=bd506e50-3ba7-4713-b212-7bb9e0f7f6ef
https://logs.px4.io/plot_app?log=aba3ca67-7721-4cc4-ac94-1b3a4c9f639e
https://logs.px4.io/plot_app?log=9cd327e1-9909-4896-9159-b05141cc0b48
https://review.px4.io/plot_app?log=c5e273cf-55d3-4177-b32f-e3d7189ee586
https://logs.px4.io/plot_app?log=5877bc88-3de6-43e8-b72e-14d9711ebd94
https://logs.px4.io/plot_app?log=2b2e71b3-b6cd-4338-bc20-787e38e3c38b
https://review.px4.io/plot_app?log=d5fa325c-526e-4f6d-9784-4553b0aa3412
https://logs.px4.io/plot_app?log=a22c4f4e-8c96-4630-a147-75392b285b14
https://logs.px4.io/plot_app?log=c52a816e-3cf9-42e4-9384-a3946a59504f
https://logs.px4.io/plot_app?log=875a0417-ae32-4711-8eac-168f7dadc345
https://logs.px4.io/plot_app?log=57e5f314-84b3-4504-9818-75c409f64841
https://logs.px4.io/plot_app?log=20aa404c-ca21-4ace-a9ad-e332b4e1ac4a
https://logs.px4.io/plot_app?log=2408c4ea-6203-4e05-ba3d-f5b194f49ed7

ryanjAA, 2023-03-22:
All on 1.14b on FW. :)

junwoo091400, 2023-03-24:
Btw, I would like to note that since we agreed to wait for the rebasing of main branch into release/1.14 until we resolve the outstanding issues in our project board: https://github.com/orgs/PX4/projects/45, I'm not sure if we can consider current beta releases as a valid one or not :shrug:

mrpollo, 2023-03-27:
@junwoo091400 anything tagged v1.14 (beta/rc... etc) or coming from the main branch (before the official release or branching out) should be considered 1.14, and as such the logs are definitely helpful.

jwarnergithub, 2023-04-01:
https://logs.px4.io/plot_app?log=7324a947-33f9-4c63-a5a1-2ce1c7208560

Flew my hex for a quick few minutes in position mode. No problems noted.

AlexKlimaj, 2023-04-01:
[image]

Looks like there is no beta flash-able for the ARKV6X in QGC.

AlexKlimaj, 2023-04-01:
```
Submodule path 'src/lib/events/libevents': checked out '48911e7be752a908a15d945b796ed8d8282b7b02'
fatal: remote error: upload-pack: not our ref bc889afb4c5bf1c0d8ee29ef35eaaf4c8bef8a5d
fatal: Fetched in submodule path 'src/lib/events/libevents/libs/cpp/parse/nlohmann_json', but it did not contain bc889afb4c5bf1c0d8ee29ef35eaaf4c8bef8a5d. Direct fetching of that commit failed.
fatal: 
Submodule path 'src/modules/mavlink/mavlink': checked out '2bdcab78b53d1e349079b43c9d726036abe0617c'
Submodule path 'src/modules/mavlink/mavlink/pymavlink': checked out '2ca2c13b54b4c75dd71c79acafc7ec40d9cb4965'
Submodule path 'src/modules/microdds_client/Micro-XRCE-DDS-Client': checked out '4248559f3b111155c783e524e461ccc83e768103'
fatal: Failed to recurse into submodule path 'src/lib/events/libevents
```
When checking out the beta and updating submodules.

AlexKlimaj, 2023-04-02:
This is a 250 quad with ARKV6X, ARK GPS, and ARK Flow. I think my rates tuning was way off to start. The vehicle always wanted to pitch backwards on takeoff. After some rates tuning it seems okay.

@dagar I was able to arm in position mode with optical flow for some reason this time.

https://review.px4.io/plot_app?log=96d5db15-dbcd-4b5e-aa64-f85d6ca41da4
https://review.px4.io/plot_app?log=41dd1b7e-d7bb-410b-8613-35459d9d511a
https://review.px4.io/plot_app?log=59d81ce1-d0d0-4cd5-9bf3-9fc12aeba2d1
https://review.px4.io/plot_app?log=f849d0fd-ca7d-43c1-9126-66eb3e43229b
https://review.px4.io/plot_app?log=7f14891c-ff1e-43e4-b2a8-03e287a77835

bresch, 2023-04-05:
@MDecarabas Thanks for the tests. I inspected the altitude issue and the drone is constantly climbing because the GNSS altitude tells that the drone is actually descending (compared to the baro that seems to be more reasonable).
[image]

You could change the height reference to baro (`[EKF2_HGT_REF](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#EKF2_HGT_REF)) to lower the effect of the GNSS height drift or even disable the GNSS height fusion to let the baro do the job alone ([EKF2_GPS_CTRL](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#EKF2_GPS_CTRL)). See https://docs.px4.io/main/en/advanced_config/tuning_the_ecl_ekf.html#height for more details.

junwoo091400, 2023-04-12:
From sturner in discord: https://discord.com/channels/1022170275984457759/1022185721450213396/1092830779429625977

Just tested with latest 1.14 and an ARK Flow. Possible that we have other issues with our configuration. This is with external vision inputs enabled where we are able to arm:
https://review.px4.io/plot_app?log=731de912-a5cb-4271-a868-ea927bd73f3b

With external vision inputs disabled, we are unable to arm: https://review.px4.io/plot_app?log=c2b658cc-72ea-4b9f-9153-c13852865406

abarcis, 2023-04-27:
We are preparing for tests with 1.14.0-beta2 and we experienced the same issues as the ones described by @MDecarabas [https://github.com/PX4/PX4-Autopilot/issues/21358#issuecomment-1483580800](https://github.com/PX4/PX4-Autopilot/issues/21358#issuecomment-1483580800) We are working with fixedwings, so I cannot fully reproduce everything, but, in particular, we were also experiencing a similar problem when trying to arm in manual mode: "Unable to arm because throttle is high when using sbus rc.". 

When debugging, we've realized that manual_control_setpoint orb topic contained throttle from 0 to 1 and I understand that in 1.14 it should be from -1 to 1. After some extensive debugging and trying random things, we've managed to make it work; we think the problem was related to the fact that throttle output of RC was inverted. @MDecarabas can you maybe confirm if your RC was also set to invert throttle?

junwoo091400, 2023-05-04:
Tested with the [main](https://github.com/PX4/PX4-Autopilot/commit/9e5420bbbd0c3fe2001ae4a1ebe6c6c3abec830c) branch commit on ATL Mantis Edu (Quadcopter) today.

Yaw estimate error bug test: https://logs.px4.io/plot_app?log=ca3cafd0-2487-4df6-868b-4901eef31808

- The same error didn't occur, meaning [between](https://github.com/PX4/PX4-Autopilot/compare/58a5aa26a09975a00b0953eac12c4b06e399ee98..9e5420bbbd0c3fe2001ae4a1ebe6c6c3abec830c?diff=split) 1.14 beta & main branch, some change happened that fixed this behavior.

Random 100% thrust issue test:

abarcis, 2023-05-05:
A few of our flights with a fixed-wing model. All done with 1.14.0-beta2.
Stabilized + mission mode:
https://logs.px4.io/plot_app?log=7aca6a10-1042-4254-9d9b-988763adc4c1
https://logs.px4.io/plot_app?log=9263f8b8-8ea5-4fab-b2c5-4068ed5a6be9
Stabilized + mission + offboard:
https://logs.px4.io/plot_app?log=ac132a77-7494-42ec-bf05-e940d580967f

During the flights, everything went smoothly. The only problem we had was with calibrating RC with the inverted throttle as mentioned in one of the comments above. Is it a known problem, or should we raise it as an issue?

abarcis, 2023-05-10:
> We are preparing for tests with 1.14.0-beta2 and we experienced the same issues as the ones described by @MDecarabas [#21358 (comment)](https://github.com/PX4/PX4-Autopilot/issues/21358#issuecomment-1483580800) We are working with fixedwings, so I cannot fully reproduce everything, but, in particular, we were also experiencing a similar problem when trying to arm in manual mode: "Unable to arm because throttle is high when using sbus rc.".
> 
> When debugging, we've realized that manual_control_setpoint orb topic contained throttle from 0 to 1 and I understand that in 1.14 it should be from -1 to 1. After some extensive debugging and trying random things, we've managed to make it work; we think the problem was related to the fact that throttle output of RC was inverted. @MDecarabas can you maybe confirm if your RC was also set to invert throttle?

I've raised the PR https://github.com/PX4/PX4-Autopilot/pull/21576 which solves this problem.

kibidev, 2023-05-19:
Not tested on real hardware, but I have tested the beta in simulator (standard VTOL), regarding this bug #21327
Logs: https://logs.px4.io/plot_app?log=259ecb46-4352-4de2-b472-ae53ecca7ef2

The bug is included in the 1.14 beta (tested on commit as of commit 18898f1876ec1be9f86d89aedc56b4918c983afa).

junwoo091400, 2023-05-23:
> I've raised the PR #21576 which solves this problem.

Thanks a lot! I have addressed the PR 🙏

SPuigUAVW, 2023-05-31:
We have been testing the beta, configuring the autopilot for a VTOL. We have the same hardware that we were flying perfectly with 1.12.

Pitot sensor calibration is not working. Through the mavlink console, using the `sdp3x` module, we can see that it does not start automatically and that the i2c bus does not register any address using `i2cdetect -b 2`. If we start manually with `sdp3x start -X`, we see that it is running on address 0x21, which is different from the default 33. After this manual start, calibration can be done. If we perform a hardware reset, everything goes back to being uncalibrated and the sdp3x module doesn't start either.

Previously, the parameter SENS_EN_SDP3X had been configured to be activated.

I'm simply raising this information to see if anyone can shed some light or indicate how we can get more information to resolve it.

Sidzeppelin95, 2023-05-31:
Here are some action items that can be done to automate testing:

   - Troubleshoot why the Pitot sensor calibration is not starting
   automatically
   - Determine why the i2c bus is not registering any address when using
   "i2cdetect -b 2"
   - Obtain more information to resolve the issue

SPuigUAVW, 2023-06-01:
> > We have been testing the beta, configuring the autopilot for a VTOL. We have the same hardware that we were flying perfectly with 1.12. (Cube Black (FMUv3), i2c Sensirion SDP33)
> > Pitot sensor calibration is not working. Through the mavlink console, using the `sdp3x` module, we can see that it does not start automatically and that the i2c bus does not register any address using `i2cdetect -b 2`. If we start manually with `sdp3x start -X`, we see that it is running on address 0x21, which is different from the default 33. After this manual start, calibration can be done. If we perform a hardware reset, everything goes back to being uncalibrated and the sdp3x module doesn't start either.
> > Previously, the parameter SENS_EN_SDP3X had been configured to be activated.
> > I'm simply raising this information to see if anyone can shed some light or indicate how we can get more information to resolve it.
> 
> Which commit are you on? Link to a log if you have it. I’ve been running sdp33 sensors nonstop for the past few months of testing without issue.
> 
> Also, the i2c address is defined by the airspeed board, not px4 side. PX4 side can be altered to register certain addresses as relevant to those specific devices but no changes have been made to the driver so can you confirm your hardware side, etc. Ate you using custom hardware or COTS sdp33 units?

Thanks for your response. 
Regarding the commit I just selected in QGC the Beta yesterday and flash it from there.
We don't have a log because we where just starting to configure parameters and calibration of the sensors and thats when we encountered the issue.
We are using Drotek's SDP33 airspeed sensor (we used it without any problems always).

After some more trying we found that the calibration was not setting **SENS_DPRES_OFF** parameter, we  input it from older parameters back up and the calibration finally saved.

SPuigUAVW, 2023-06-12:
> > After some more trying we found that the calibration was not setting **SENS_DPRES_OFF** parameter, we input it from older parameters back up and the calibration finally saved.
> 
> @SPuigUAVW during [today's maintainers call](https://discuss.px4.io/t/px4-maintainers-call-june-06-2023/32473#v114-release-remaining-issues-5), we discussed maybe your calibration value for `SENS_DPRES_OFF` was getting set to exactly 0.0, for your sensor (and that may cause the code to think that sensor isn't calibrated, because offset of 0.0 implies uncalibrated sensor)
> 
> So I had 2 questions for you:
> 
> 1. What is the value of that old parameter you set manually?
> 2. How did you find out that the `SENS_DPRES_OFF` was not getting set? Which value did it stay on? (Was it 0.0?)

Sorry for my response delay. 

1. We manually set SENS_DPRES_OFF as 0.1. 
2. It was 0.00000000... before, we checked it in a saved Parameter backup. We found some info around similar issues in this thread, where my partner [VTOLDavid](https://github.com/VTOLDavid) reported problems: https://github.com/PX4/PX4-Autopilot/issues/19088

VTOLDavid, 2023-06-12:
Hi, I just found that on 13.3 if you are flying a tailsitter (SITL) on FW in position or hold modes, clicking on the map  QGC offers you to go to location or orbit at location . Go to location works as spected but orbit at location surprisingly works and does a  orbit with the radius you select, you can set a 5m radius. It goes crazy trying to turn in such a small radius and ignores the parameter NAV_LOITER_RAD.  
Log file:
https://logs.px4.io/plot_app?log=48c0670b-e030-45cb-a776-9fa25e013cb1

Edit: Same behavior found using 1.14.0.

QGC pictures:
https://media.discordapp.net/attachments/1028031026947162143/1116030550810054696/photo_2023-06-07_17-39-42.jpg?width=857&height=586

https://cdn.discordapp.com/attachments/1028031026947162143/1116030551070089226/photo_2023-06-07_17-39-34.jpg

https://cdn.discordapp.com/attachments/1028031026947162143/1116030551304962158/photo_2023-06-07_17-39-20.jpg

AlexKlimaj, 2023-06-14:
Just flew my 250 and Rover on 1.14 with my RTCM fixes.

250
https://review.px4.io/plot_app?log=5c742d13-b95e-4df7-91ad-d46d91e50b99
https://review.px4.io/plot_app?log=5c4f81c0-67a4-4566-9c77-b39c946b94fb
https://review.px4.io/plot_app?log=15c06091-a431-457b-bec8-f3d9d55c0f1b

Rover with https://github.com/PX4/PX4-Autopilot/pull/21722
https://review.px4.io/plot_app?log=e3a8cb5e-c974-4141-baf5-6bfcd1baa902
https://review.px4.io/plot_app?log=87c6243a-58be-4666-b2b1-5c543ce60129
https://review.px4.io/plot_app?log=bd52a8cc-c2bb-43d1-82d2-aebc11897367
https://review.px4.io/plot_app?log=a1393406-1e01-4daf-a19d-c6b44626c4b2

DusterTheFirst, 2023-06-17:
Flew a custom frame, had a lot of normal flights but just this afternoon ran into super weird oscillations. I am unsure of the cause.

https://logs.px4.io/plot_app?log=78b46588-80e0-4a89-bae8-d7a10a51ed82

VTOLDavid, 2023-06-23:
Found 2 bugs flying a tailsitter on SITL.

**FW loitering with 0 radius:**

1. Disarmed in position mode
2. Take-off from QGC  
3. When reached selected altitude it changes to hold mode
4. Transition from QGC
5. The plane starts turnign with 0 radius.

This also happens if you select in QGC "orbit here" and set a smaller radius than the minimum the airframe can reach. Se discussion on Discord:
https://discord.com/channels/1022170275984457759/1116030550742925343

Log:

**Hold does not stop a Landing command**

1. Flying in hold in FW mode
2. Hit Land in QGC
3. Transition to MC and starts descending.
4. Try to abort the landing selecting hold mode.
5. Continues descending either in FW or MC mode.

Seems that the altitude setpoint is not set to the current altitude in MC or minimum FW loiter altitude in FW.
log:
https://logs.px4.io/plot_app?log=c6a6d461-3a02-4d8f-a408-80aaf0711801

junwoo091400, 2023-06-26:
> Go to location works as spected but orbit at location surprisingly works and does a orbit with the radius you select, you can set a 5m radius. It goes crazy trying to turn in such a small radius and ignores the parameter NAV_LOITER_RAD.

Thanks for the report @VTOLDavid! I have added this to tomorrow's maintainers meeting notes to discuss

mrpollo, 2023-06-27:
Hey @sfuhrer please take a look at the reports above

jstrebel, 2023-06-28:
ERROR Mission rejected: Landing waypoint/pattern required.
RTL_TYPE 0 has no effect.
With 1.13.3, same Mission all OK

SW v1.14.0 (beta) ([18898f18](https://github.com/PX4/Firmware/commit/18898f1876ec1be9f86d89aedc56b4918c983afa))
HW: pixhawk4
QGC:4.2.6
log:https://review.px4.io/plot_app?log=a2bd448c-ad95-4b88-a6cf-bafb72a37e3a

sfuhrer, 2023-06-29:
> ERROR Mission rejected: Landing waypoint/pattern required. RTL_TYPE 0 has no effect. With 1.13.3, same Mission all OK
>
> SW v1.14.0 (beta) [...]

You can customize requiring takeoff/landings in missions through MIS_TKO_LAND_REQ, see also https://docs.px4.io/main/en/flight_modes/mission.html#mission-feasibility-checks.

sfuhrer, 2023-06-29:
> Found 2 bugs flying a tailsitter on SITL.
>
> **FW loitering with 0 radius:**
>
>     1. Disarmed in position mode
>
>     2. Take-off from QGC
>
>     3. When reached selected altitude it changes to hold [...]

hi @VTOLDavid , I see that the PX4 version you used is [1.13, after this PR](https://github.com/PX4/PX4-Autopilot/pull/21227). Can you check if the same happens on v1.14? I briefly checked in SITL and could not reproduce neither of your reported two issues.

VTOLDavid, 2023-06-29:
Yes it was in 1.13.3, latest stable version. I will check it in 1.14.

VTOLDavid, 2023-06-29:
> > Found 2 bugs flying a tailsitter on SITL.
> 
> hi @VTOLDavid , I see that the PX4 version you used is [1.13, after this PR](https://github.com/PX4/PX4-Autopilot/pull/21227). Can you check if the same happens on v1.14? I briefly checked in SITL and could not reproduce neither of your reported two issues.

Yes it was in 1.13.3, latest stable version. I will check it in 1.14. I am testing both versions and though I was in 1.14.

VTOLDavid, 2023-06-29:
Not happening un 1.14

junwoo091400, 2023-06-29:
> After some more trying we found that the calibration was not setting **SENS_DPRES_OFF** parameter, we input it from older parameters back up and the calibration finally saved.

As I noted in https://github.com/PX4/PX4-Autopilot/issues/19088#issuecomment-1613285885, the airspeed calibration itself should work in v1.14 beta, however as you stated, if the `SENS_DPRES_OFF` is indeed not getting saved, could you provide more information on how the calibration went?

Did the airspeed calibration indeed finish? What did the QGC calibration screen show?

mwshafer, 2023-07-04:
@mrpollo I’ve been testing with 1.14 beta for some time now and have been continually dealing with high vibration levels. I’ve only recently realized that this is a PX4 specific issue, as I loaded Arducopter today and the vibration levels all looked fine for that flight. This morning I completed two back to back flights with my [system](https://uavsystemsinternational.com/products/aurelia-x4-standard?variant=43543652532461) which runs a Cube Orange+. The first flight is with PX4 and the second was with Arducopter. Both log files can be found [here](https://drive.google.com/drive/folders/1ORU0Rf2oy8h3m_W7aB_s35jxQ42yixJe?usp=sharing). The PX4 flight was posted in Flight Review here:
https://review.px4.io/plot_app?log=a9bebd55-10d4-42df-b12b-5781106c4696

The PX4 flight had a terrible oscillation during the decent on RTL that nearly crashed the drone. I was able to recover thankfully.

What is really odd is that the first two IMUs, which are supposed to be mechanically isolated/damped are showing higher vibration levels than the one that is hard mounted in the board:

[image]

I’ve posted this issue here as well: https://discuss.px4.io/t/accel-1-clipping-with-px4-no-problems-with-arducopter/32247/4

AlexKlimaj, 2023-07-06:
release/1.14 with #21789 #21769 #21019

https://review.px4.io/plot_app?log=a1d5cbbb-fe78-425b-9ef6-c99dca2d6139
https://review.px4.io/plot_app?log=f458c74a-898b-4e03-b1e4-50315a1ec2b1
https://review.px4.io/plot_app?log=622f14ec-ec4e-4d45-97cb-475798fb1f84

jstrebel, 2023-07-07:
Fixed Wing FX61 using current 1.14main (QGC 4.2.6)
Fairly good without tuning. (setting MIS_TKO_LAND_REQ as recommended by sfuhrer)
https://review.px4.io/plot_app?log=b9f996e3-43ed-42b2-ad12-5f67942e72c5

mwshafer, 2023-07-10:
I just tried to install the beta version and I am getting this error:
[image]

I have tried this multiple times, all with the same result. I’m using the daily build of QGC. This worked last week when I installed 1.14 beta.

junwoo091400, 2023-07-10:
> I’m using the daily build of QGC. This worked last week when I installed 1.14 beta.

Weird, since the `beta` target shouldn't have been available at all, according to the last beta S3 bucket Jenkins log: http://ci.px4.io/blue/organizations/jenkins/PX4_misc%252FFirmware_s3_deploy_beta/detail/Firmware_s3_deploy_beta/47/pipeline/

Also, I don't have the same make error when I build the target in `release/1.14` branch. For now, could you download the binary from the `main` branch NuttX build artifacts from runs like this one? -> https://github.com/PX4/PX4-Autopilot/actions/runs/5510817520

mwshafer, 2023-07-10:
@junwoo091400 I don’t do a lot of this so excuse my ignorance. Where on that page do I find the Nuttx build artifacts for the CubeOrangePlus?

I was able to build the release/1.14 branch without issue.

junwoo091400, 2023-07-10:
> Where on that page do I find the Nuttx build artifacts for the CubeOrangePlus?

You can find the artifact down in the page here:
[image]

In that artifact you can find the .px4 binary under the target name folder!

julianoes, 2023-07-11:
@mwshafer have you tried a clean build and are all submodules up-to-date? `git submodule update --init --recursive`?

mwshafer, 2023-07-11:
@julianoes that submodule update worked. I haven't tried via QGC again today.

afwilkin, 2023-07-19:
CubeOrange Plus 1.14 fixed wing
https://review.px4.io/plot_app?log=57ca4342-b057-411b-b1be-da1bd853574f

jstrebel, 2023-07-24:
Two flights (beta build at 19.jul. ) The wind was moderate. but the flights are OK. 
Flight one: direct into mission mode to loitering.
https://review.px4.io/plot_app?log=759d2132-dea2-4da6-b686-e72f71e3fe4c
Flight two: direct into mission mode. Survey, then loitering.
https://review.px4.io/plot_app?log=15d55457-68dc-4991-a8dd-d7a534763a8b

I observed errors in the log: hight estimate nor stable and horizontal velocity unstable. I saw this messages several times during flight preparation.

sfuhrer, 2023-07-25:
>Two flights (beta build at 19.jul. ) The wind was moderate. but the flights are OK.

@jstrebel thanks for the testing! Looks alright indeed, beside the estimator warnings that you're also mentioning. 

@bresch fyi for the "horizontal velocity unstable" and "yaw estimate error" preflight failures.

jstrebel, 2023-07-25:
@sfuhrer @bresch Let me know if you need more specific tests. My test setup is now operational.

bresch, 2023-07-27:
@jstrebel @sfuhrer I had a look at the log and both preflight failures are caused by the same thing and not due to a software issue. Basically, between boot and takeoff, the drone was moved and because of the bad mag readings, the yaw estimate is disturbed (triggering "yaw estimate error") and this creates slight issues in the GNSS vel fusion (because the North estimate is a bit off), triggering "horizontal velocity unstable".
After takeoff, the yaw innovations are getting much lower and we don't see issuer anymore. This could indicate that the mag filed near the ground wasn't clean.

@jstrebel Maybe could you just try the same thing but at another location? (boot, wait for GNSS fix, then move and tumble a bit the drone).
[image]

jstrebel, 2023-07-27:
@sfuhrer @bresch had some excellent flights. Attached one log 
https://review.px4.io/plot_app?log=88256746-de50-4240-9661-d859b9f25cfd
Thank you for your tips.
BTW: Autolanding, do I need a ground radar or ground lidar?

sibujacob, 2023-07-31:
**After using PX4v1.14-rc** , I encountered failsafe activated due to low power( indication at 22.6V ).  This is false especially for lithium ion batteries 6s pack shown as low power and the drone tends to land itself .  Is there any way to remove this code !!

ryanjAA, 2023-07-31:
> **After using PX4v1.14-rc** , I encountered failsafe activated due to low power( indication at 22.6V ). This is false especially for lithium ion batteries 6s pack shown as low power and the drone tends to land itself . Is there any way to remove this code !!

Upload the log. It’s most likely an incorrect setting.

sibujacob, 2023-07-31:
> Upload the log. It’s most likely an incorrect setting.

thank you for your feedback.....Please find below the log files ......
https://review.px4.io/plot_app?log=2b50236c-b16a-49e8-9a61-bacbbe4d8091

ryanjAA, 2023-07-31:
> thank you for your feedback.....Please find below the log files ...... https://review.px4.io/plot_app?log=2b50236c-b16a-49e8-9a61-bacbbe4d8091 [...]

Vibration is way too high. Batteries weren't fully charged. You also need batteries with a higher C rating given the voltage sag in the logs (or use larger batteries). You are completely saturated (motor outputs at max) so either weight is too high or motor power is too low for the weight. Fully charged batteries wont really fix that since you are close to full but now full. The issue isn't the flight code on these logs, it's the hardware setup and battery selection for your setup.

sibujacob, 2023-07-31:
> Vibration is way too high. Batteries weren't fully charged. You also need batteries with a higher C rating given the voltage sag in the logs (or use larger batteries). You are completely saturated [...]

I am using the same setup with the same batteries but when I  run PX4 v1.12.3 , i get nearly 20 minutes of flight time !!  flight log for PX4 v1.12.3 done 2 weeks back 
https://review.px4.io/plot_app?log=4947922a-c1ed-4fde-9b3e-1cce79181b70  ( this flight shows the battery graph with 0 battery remaining all the time )

ryanjAA, 2023-07-31:
> I am using the same setup with the same batteries but when I run PX4 v1.12.3 , i get nearly 20 minutes of flight time !! flight log for PX4 v1.12.3 done 2 weeks back [...]

Your battery settings are different enough to trigger this from the older logs to newer. Adjust and retry but you should still change the battery. Voltage sag is much too high...

sibujacob, 2023-08-01:
> Your battery settings are different enough to trigger this from the older logs to newer. Adjust and retry but you should still change the battery. Voltage sag is much too high...

thank you , Ryan.  Will  look into the battery modification . I also suspect the BMS module used with the Li ion pack.

sfuhrer, 2023-08-02:
> @sfuhrer @bresch had some excellent flights. Attached one log https://review.px4.io/plot_app?log=88256746-de50-4240-9661-d859b9f25cfd Thank you for your tips. BTW: Autolanding, do I need a ground radar or ground lidar?

Happy to hear! For autolanding you need a distance sensor of any type, though lidars are much more common in the category of small UAVs. See https://docs.px4.io/main/en/flight_modes/mission.html#automatic-abort for the logic around distance sensor data handling in the landing sequence.

sfuhrer, 2023-08-03:
>would it be possible to do a autolanding test without lidar until I get one?

You can try it without the flaring step by disabling the automatic abort due to no distance sensor present, see https://docs.px4.io/main/en/flight_modes/mission.html#automatic-abort.
It should then just keep the set glide slope without flaring. Shallow glide slope recommended!

jstrebel, 2023-08-07:
Hi below the links to Testflights with 1.14rc1

The Main Goal was to do a Autotuning with our fixedwing Skywalker X6

We were strugling to follow the initial steps of the Documentation. In fact we could not find the descibed part below which belogs to QGC.

If an Enable/Disable Autotune Switch is configured you can just toggle the switch to the "enabled" position. 
1. In QGroundControl, open the menu: Vehicle setup > PID Tuning 2. Select either the Rate Controller or Attitude Controller tabs. 3. Ensure that the Autotune enabled button is enabled (this will display 
the Autotune button and remove the manual tuning selectors). 4. Read the warning popup and click on OK to start tuning.

Flight 1: (check the behaviour of the plane)
https://review.px4.io/plot_app?log=8fb0be4f-a259-42a7-91e0-389773f6c442
Observation:
- Roll axis  oszillation (It looks I changed by mistake in a earlier flight FW_RR_I to 4)
- log: Altitude failure (roll)
- log Failsafe warning triggered

Flight 2: first try autotune in position mode
https://review.px4.io/plot_app?log=3f89d902-3ecf-4cbe-a27a-406ea8931ab3
Observation:
- log: Altitude failure (roll)
- log Failsafe warning triggered

Flight 3: Fixed, roll axis oszillation FW_RR_I to default. 
https://review.px4.io/plot_app?log=8f858806-542f-46c3-94d3-2247ac54157e
Test autotune only for roll
Observation:
- log: Altitude failure (roll)
- log Failsafe warning triggered

Flight 5: it seems autotune for roll worked
https://review.px4.io/plot_app?log=01349a1e-3778-4368-a7f6-9df380ec5997

Observation:
- log many Failsafe warning triggered

We hear quite often the voice message in QGC :
-Horizontal velocity not stable
-Hight estimate not stable

sfuhrer, 2023-08-08:
@bresch could you have a look at the autotuning feedback above?

@jstrebel concerning your high wind testing post above: Did you on purpose disable airspeed fusion into the EKF wind estimate? EKF2_ARSP_THR is set to 0. It's highly recommended to enable airspeed fusion, otherwise the system has a hard time estimating the wind speeds, and if the wind speed isn't known then the controller gets unstable (that's what happened in your flight last week).

jstrebel, 2023-08-08:
FW Test flights 1.14rc1
The intention was to test the “robust wind aware path optimization” from Thomas Stastny
Mission with two surveys. 

In both missions, two surveys are defined. The flight paths for the surveys are deliberately planned perpendicular to the wind direction. The goal is that the trajectories are flown with the same speed.

The first survey with transect, the second without transect. Enclosed are the parameters I changed. For flight two I set FW_RR_IMAX to 0.6. Hoping that the turns will be better. But I do not notice any change.

Flight: 1 https://review.px4.io/plot_app?log=40b2f903-4509-42a6-a54a-c75e562a4067
FW_R_LIM 65 	 (50)
FW_R_RMAX 120	(70)

Flight 2 https://review.px4.io/plot_app?log=0e81a536-dce5-4231-90e0-eaf217a837e7
FW_RR_IMAX 0.6   (0.2)
FW_R_LIM 64	(50)
FW_R_RMAX 120	(70)

I assume there are other parameters I need to touch. **Which ones?** Thank you for your suggestions.

[57 more replies]

closed 2023-11-03 by mrpollo