PX4/PX4-Autopilot #21370: v1.14.0-beta regression: Failsafe Flight Task is being triggered after startup

junwoo091400, 2023-03-25:
## Describe the bug
After the commit https://github.com/PX4/PX4-Autopilot/commit/ea97a38b2b5edaa515d8d6feb836c45f302b3b95, the `main` branch constantly suffered the HITL tests for certain targets (5 out of 8) reporting "Failsafe Flighttask activated" error in Jenkins Hardware rack testings.

Context in the problematic PR: https://github.com/PX4/PX4-Autopilot/pull/19748#issuecomment-1483755408

More context with screenshots: https://github.com/PX4/PX4-Autopilot/pull/21230#issuecomment-1483754432

## To Reproduce
1. Checkout `main` branch, then: `make px4_sitl gazebo-classic` -> Soon after startup, "Failsafe Flighttask activated" error will appear
2. Checkout [commit](https://github.com/PX4/PX4-Autopilot/commit/fe3c1d0a92a202a8f1428aea1028bcc0aca2333e) right before the problematic PR: `git checkout fe3c1d0a92a202a8f1428aea1028bcc0aca2333e && make px4_sitl gazebo-classic` -> There is no Failsafe Flighttask error after startup.

## Expected behavior
Failsafe flighttask error shouldn't arise without a concrete reason. Something in the PR https://github.com/PX4/PX4-Autopilot/pull/19748 introduced a regression.

## Log Files and Screenshots
1. Failsafe Flighttask triggered log: https://logs.px4.io/plot_app?log=e789c03c-1e7a-48e5-b74b-ac3e38ecf986
2. Failsafe not triggered log (one commit before): https://logs.px4.io/plot_app?log=f2e20bfe-fca2-424e-825a-62577c718874

## Additional context
@dagar I always ignored the Jenkins Hardware rack tests, but this one really showed me that it will actually catch the cases we shouldn't ignore. Let's make sure we catch these in the future before it getting merged as well!!

junwoo091400, 2023-03-25:
Note: I have found that before/after that problematic [commit](https://github.com/PX4/PX4-Autopilot/commit/ea97a38b2b5edaa515d8d6feb836c45f302b3b95), the Flight Task activation pattern has differed.

After the commit: Switches between 0, 3, 5 indexes (each Auto, ManualAcceleration, ManualAltitudeSmoothVel)
[image]

Before the commit: Switches between 0, 3 indexes (each Auto, ManualAcceleration)

It is very strange why after that commit the "ManualAltitudeSmoothVel" is suddenly failing to activate/initialize, but at least the problem is getting narrowed down :grimacing:

junwoo091400, 2023-03-25:
[image]

Ah no! I realized there's actually *plenty of commits between these two. Idk why Jenkins didn't run (or preserve log) for commits in between, but it now excludes possibility that Alex's commit actually introduced this bug (as it's only related to UAVCAN, it is not likely. The driver isn't even running in the STIL).

Now investigating all the commits between them :disappointed:

junwoo091400, 2023-03-25:
The answer lies somewhere here: https://github.com/PX4/PX4-Autopilot/compare/fe3c1d0a92a202a8f1428aea1028bcc0aca2333e...ea97a38b2b5edaa515d8d6feb836c45f302b3b95

junwoo091400, 2023-03-25:
Now doing bissect, as @MaEtUgR taught me :wink:

junwoo091400, 2023-03-25:
Found it by bisecting 8 times! It's this commit: https://github.com/PX4/PX4-Autopilot/commit/da4644c20adce8aeedd5be432fc88b1b115d7eb9

MaEtUgR, 2023-03-27:
@junwoo091400 Heeey, thanks for looking into this! That's already really helpful. I noticed the `Failsafe Flighttask activated` in SITL and wanted to see why it comes up in a quiet moment but wasn't aware that it trips CI (a good thing) or someone else found it too.

Let me check if it's easy to reproduce.

MaEtUgR, 2023-03-28:
I added the validity flag to the requirements of using stick input:
https://github.com/PX4/PX4-Autopilot/commit/da4644c20adce8aeedd5be432fc88b1b115d7eb9#diff-a65eff8fc6362bb2a8c4c7495f56778e1896809f6251a9403e57ff18f754076aR66
Maybe there's a glitch.

MaEtUgR, 2023-03-28:
It's because it goes through Altitude mode during boot even when there is no stick input and Altitude mode reports to fail :confused:

junwoo091400, 2023-03-28:
@MaEtUgR Btw I did some analysis (in comments above), feel free to reference them in case that helps as well! :raised_hands: 

What I found weird was that FlightTask was trying to activate 3 different tasks (trying to activate Auto, failing then falling back to ManualAcceleration, failing then falling back to ManualAltitudeSmoothVel, etc"

Is this how Flighttasks are supposed to operate? I was surprised by how many `switchTask` gets called!!

MaEtUgR, 2023-03-28:
Thank you so much @junwoo091400 ! :rocket: I'm on it :pray:

MaEtUgR, 2023-03-28:
It's a fallback but Commander always tries to be in `NAVIGATION_STATE_AUTO_LOITER` from the beginning on but something fails there. I'm guessing because there's no position estimate available yet.

MaEtUgR, 2023-03-28:
I'm pretty sure now the problem is that with the new failsafe state machine it is allowed to switch to a mode before arming even though it cannot run. When doing so e.g. switching to Hold mode `FlightTaskAuto` is ran and reports it cannot run because it requires local position, global reference and a position setpoint triplet. So the question is how to solve that best. :thinking: 
https://github.com/PX4/PX4-Autopilot/blob/319f3db5e1f0f799be45064330d9c9504f5b5bcc/src/modules/flight_mode_manager/tasks/Auto/FlightTaskAuto.cpp#L97-L100

MaEtUgR, 2023-03-28:
It's there from the introduction of the flight tasks. Any task that requires a local position fails to execute if its requirements are not fulfilled: https://github.com/PX4/PX4-Autopilot/commit/afd02aca92564e9e2be05942233f5d31773c797a

I'd rather not make a hole for just `FlightTaskAuto` but change how it works in general.

PR #21458 (2023-04-15): Add descend task to failsafe fallback to avoid "Failsafe task activated"

### Solved Problem
When starting SITL e.g. using `make px4_sitl sihsim_quadx` and don't have any ground station that immediately sends joystick data there's an error message `ERROR [flight_mode_manager] Failsafe task activated`

[image]

This translates to the real vehicle boot without position estimate and RC signal.

Fixes #21370

### Solution
My suggestion for 1.14 is to add the descend task to the flight mode manager fallback which I did in this pr. It's a minimal change that solves the problem like it was after merg [... cut]
closed 2023-04-15 by MaEtUgR