PX4/PX4-Autopilot #21378: SITL x500 motors not working in Gazebo Garden

MichaelBosello, 2023-03-28:
## Describe the bug
I tried to run SITL with Gazebo Garden and Ubuntu 20.04 on the current main, but the vehicle does not take off or arm.

## To Reproduce
Steps to reproduce the behavior:

1) [Install Gazebo Garden](https://gazebosim.org/docs/garden/install_ubuntu)

2) Download PX4 and install the basic requirements

```
git clone https://github.com/PX4/PX4-Autopilot.git --recursive
bash ./PX4-Autopilot/Tools/setup/ubuntu.sh --no-nuttx --no-sim-tools
```
3) Build and run SITL
```
cd PX4-Autopilot
make px4_sitl gz_x500
```

4) In QGC, enable the virtual joystick
5) In QGC, command a takeoff

6) The vehicle does not move

## Expected behavior
The vehicle takes off and goes into hold mode

## Log Files and Screenshots
Error reported in the shell:
```
[Err] [MulticopterMotorModel.cc:255] Please specify a robotNamespace.
[Err] [MulticopterMotorModel.cc:255] Please specify a robotNamespace.
[Err] [MulticopterMotorModel.cc:255] Please specify a robotNamespace.
[Err] [MulticopterMotorModel.cc:255] Please specify a robotNamespace.
```

## Additional context
I made it work by manually defining the `robotNamespace` of the motors plugin in the `model.sdf`:
`<robotNamespace>x500_0</robotNamespace>`
But it isn't a good way to handle the problem

I'm on a clean Ubuntu 20.04 machine.

Jaeyoung-Lim, 2023-03-28:
@MichaelBosello What is your gz version?

MichaelBosello, 2023-03-28:
```
Gazebo Sim, version 7.0.0
```

MichaelBosello, 2023-03-28:
@Jaeyoung-Lim Yes, I saw that. I gave it a try (libignition pointing to 6.14), but it didn't have any effect. I don't think it applies to Garden as it is not ign anymore, but only to Fortress

Jaeyoung-Lim, 2023-03-28:
@MichaelBosello I see, I think Ubuntu 20.04 with Gazebo Garden was not really tested until now. 
It might be caused by one of the ign-* dependencies that might be causing it.

Most of the development/support have been happening on 22.04 for Gazebo Garden

Sitin, 2023-06-24:
I have the same issue on Ubuntu 22.04 and official Gazebo Garden.

It is possible to "fix" this problem with `<robotNamespace>x500_0</robotNamespace>` trick.

PRAN384, 2023-09-03:
Hi all, 
I am running PX4 SITL with Gazebo Garden and I am getting the same error on ubuntu20.

I tried to fix my .sdf file like @MichaelBosello, and I was able to get rid of the error messages for the file, but I still was not able to get quad to takeoff. I then I noticed that I get another error as shown:

`WARN  [init] PX4_GZ_MODEL_NAME or PX4_GZ_MODEL not set using PX4_SIM_MODEL.
Error [Utils.cc:174] Missing element description for [persistent]
` 
Does anyone have an idea of what it could be?

DonLakeFlyer, 2024-04-27:
> Error [Utils.cc:174] Missing element description for [persistent]

This is caused by the ApplyLinkWrench plugin in the default.sdf world ()https://github.com/PX4/PX4-gazebo-models/blob/main/worlds/default.sdf#L15). No idea what this plugin does. But if you remove it. You will no longer get the error and takeoff will work.