PX4/PX4-Autopilot #21387: [WIP] v1.14.0 beta: Yaw estimate error in ATL Mantis Edu

junwoo091400, 2023-03-29:
## Describe the bug
With ATL Mantis Edu flashed with `main` (1.14.0 beta), after the compass calibration, the yaw estimate was off by 180 degrees.

Also, when I inspected `vehicle_magnetometer` and `sensor_mag` data, the data didn't make much sense, so I am guessing it is either:

- Magnetometer orientation detection error
- Magnetometer fault

But in case this is somehow related to PX4's EKF / Magnetometer architecture, I am creating this issue!

## Expected behavior
A clear and concise description of what you expected to happen.

## Log Files and Screenshots
https://logs.px4.io/plot_app?log=10380af7-c173-4795-be6e-b4088224a689

- First I went around the park pointing the drone in the direction of travel (as a groundtruth for the sensor data vs actual heading)
- Then, I had a short Manual mode flight just to test it's flyability (& magnetometer data)

## Drone (please complete the following information):
ATL Mantis Edu

## Additional context
I haven't tested on `v1.13`, so I can't confirm if this is a regression.

But what I know is that this yaw estimate error didn't occur in the past when developing Follow Me mode: https://github.com/PX4/PX4-Autopilot/pull/19260

gustavotbarros, 2023-03-30:
I also flashed a Pixhawk 6X Board with the v1.14 beta and was getting critical errors because of the Yaw Estimate all the time.

gustavotbarros, 2023-05-10:
Hi @junwoo091400, i've flashed my Pixhawk 6X today with 1.14-beta2 and the problems are still there. 

When I move the drone around, several sensor calibration errors appear. 

And this time it even showed up a Crash Dump on SD message, something like that. I went back to 1.13.2 and the problems are gone.

junwoo091400, 2023-05-10:
> i've flashed my Pixhawk 6X today with 1.14-beta2 and the problems are still there.

That's normal! Since we didn't release further beta, the problem persists in `v1.14-beta2`, and is only fixed (as I tested) in the `main` branch.

> And this time it even showed up a Crash Dump on SD message, something like that. I went back to 1.13.2 and the problems are gone.

That sounds worrying, could you create an issue with a SD card dump message & details of the test please? That may be one of the critical bugs we might have in the 1.14 release!

gustavotbarros, 2023-05-10:
> That's normal! Since we didn't release further beta, the problem persists in v1.14-beta2, and is only fixed (as I tested) in the main branch.

I apologize for the confusion earlier. I have tested the main branch as well, it wasn't beta2. Unfortunately, I cannot perform a test at the moment, but I can upload again the main branch tomorrow and see if the Crash Dump message appears.

Just to clarify, I observed the sensor errors when I physically manipulated the drone by hand.

mrpollo, 2023-06-27:
are you planning to fix for v1.14 @junwoo091400 otherwise let's remove from the project board

junwoo091400, 2023-06-27:
> As I detailed in [#21358 (comment)](https://github.com/PX4/PX4-Autopilot/issues/21358#issuecomment-1534226224), the issue seems to have been resolved in the main branch now

As I stated above, now in the `main` branch this has been resolved, thus I think this can be closed!

hugo2410, 2024-11-12:
Hi @junwoo091400 
I'm seeing a similar issue in v1.15.1, just wanted to confirm that v1.15 had been branch off main prior to this fix as this issue seems to be in the v1.16 Release project?
Also, this is preventing me from arming, is there a param I could change to override it ?