PX4/PX4-Autopilot #21393: Pixhawk v6c Micro DDS Client don't connect with Micro XRCE Agent using serial port and ROS2 tutorial

RPS98, 2023-03-29:
## Describe the bug
I have tried to follow [ROS 2 Offboard Control Example](https://docs.px4.io/main/en/ros/ros2_offboard_control.html#ros-2-offboard-control-example), using Pixhawk v6c board connected by serial port.  Micro DDS Client launched in px4 board and Micro XRCE Agent launched on computer doesn't connect. I have tried both PX4 Autopilot v1.13.3 and latest main branch.

It works on [Gazebo Classic Simulation](https://docs.px4.io/main/en/sim_gazebo_classic/#gazebo-classic-simulation). I am using [Ubuntu 22.04.2 LTS](https://releases.ubuntu.com/jammy) with [ROS 2 Humble](https://docs.ros.org/en/humble/index.html).

## To Reproduce
Steps to reproduce the behavior:
1. Follow [PX4-Autopilot ROS 2 User Guide](https://docs.px4.io/main/en/ros/ros2_comm.html)
  1.1. Install [ROS 2 Humble](https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html#)
  1.2. Build [PX4-msgs and PX4-ros-com](https://docs.px4.io/main/en/ros/ros2_comm.html#building-the-workspace)
  1.3. Build [Micro XRCE DDS Agent in ROS within ROS 2 Workspace](https://docs.px4.io/main/en/middleware/xrce_dds.html#build-run-within-ros-2-workspace)
3. Clone [PX4-Autopilot](https://github.com/PX4/PX4-Autopilot) and build px4_fmu-v6c (make px4_fmu-v6c)
4. Open QGroundControl and upload custom board firmware.
5. Open MAVLink Console and [start microdds client](https://docs.px4.io/main/en/modules/modules_system.html#microdds-client):
```
nsh> microdds_client start -t serial -d /dev/ttyS3 -b 921600
nsh> microdds_client status
INFO  [microdds_client] Running, disconnected
INFO  [microdds_client] Using transport: serial
```
6. Connect TELEM2 to computer by serial.
7. Launch Micro XRCE DDS Agent in computer
```
$ MicroXRCEAgent serial -b 921600 --dev /dev/ttyUSB0
waiting for connection...
[1680099912.703669] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1680099912.703845] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4

```

## Expected behavior
A similar output where computer Micro XRCE Agent connect to PX4 Micro DDS Client and creates ROS2 publishers/subscribers.
```
[1680100421.530899] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1680100421.531059] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1680100422.149455] info     | Root.cpp           | create_client            | create                 | client_key: 0x00000001, session_id: 0x81
[1680100422.153600] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x00000001, address: 127.0.0.1:47029
[1680100422.159350] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x00000001, participant_id: 0x001(1)
[1680100422.159629] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3E8(2), participant_id: 0x001(1)
[1680100422.159679] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3E8(4), participant_id: 0x001(1)
[1680100422.160489] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3E8(6), subscriber_id: 0x3E8(4)
[1680100422.160575] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3E9(2), participant_id: 0x001(1)
[1680100422.160590] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3E9(4), participant_id: 0x001(1)
[1680100422.160636] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3E9(6), subscriber_id: 0x3E9(4)
[1680100422.160683] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3EA(2), participant_id: 0x001(1)
[1680100422.160694] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3EA(4), participant_id: 0x001(1)
[1680100422.160732] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3EA(6), subscriber_id: 0x3EA(4)
[1680100422.160775] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3EB(2), participant_id: 0x001(1)
[1680100422.160787] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3EB(4), participant_id: 0x001(1)
[1680100422.160899] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3EB(6), subscriber_id: 0x3EB(4)
[1680100422.160945] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3EC(2), participant_id: 0x001(1)
```

## Log Files and Screenshots
When launch microdds client on PX4 v6c board, the following output show up:
```
nsh> microdds_client start -t serial -d /dev/ttyS3 -b 921600
nsh> microdds_client status
INFO  [microdds_client] Running, disconnected
INFO  [microdds_client] Using transport: serial
```

And when launch Micro XRCE Agent on computer, the following output show up:
```
$ MicroXRCEAgent serial -b 921600 --dev /dev/ttyUSB0
waiting for connection...
[1680099912.703669] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1680099912.703845] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
```

## Drone
- I have used Pixhawk 6c board, with USB port power and TELEM2 port connected to computer. Also try on a [PX4 Vision Autonomy Development Kit](https://docs.px4.io/main/en/complete_vehicles/px4_vision_kit.html).
- The computer uses [Ubuntu 22.04.2 LTS](https://releases.ubuntu.com/jammy) with [ROS 2 Humble](https://docs.ros.org/en/humble/index.html)

mrpollo, 2023-03-29:
Hey @RPS98 the microdds feature is only available for v1.14 and above, please try again with the v1.14-beta2 release tag

RPS98, 2023-03-29:
Hi, thanks for the response.

I have tried it and the same error happens.

The microdds_client log change to:
```
nsh> microdds_client start -t serial -d /dev/ttyS3 -b 921600
nsh> microdds_client status
INFO  [microdds_client] Running, disconnected
INFO  [microdds_client] Payload tx: 0 B/s
INFO  [microdds_client] Payload rx: 0 B/s
```

Have to launch the Agent in the computer and then connect the port, otherwise the following error happens:
```
$ MicroXRCEAgent serial -b 921600 --dev /dev/ttyUSB1
[1680107388.920965] error    | TermiosAgentLinux.cpp | init                     | open device error      | device: /dev/ttyUSB1, errno: 16
Error while starting serial agent!
```

Don`t know if may be a problem.

Also, I have tried to set XRCE_DDS_0_CFG param to TELEM2, to auto-initialize, but it doesn`t stored after reboot.

beniaminopozzan, 2023-03-29:
Hi @RPS98 

> Have to launch the Agent in the computer and then connect the port, otherwise the following error happens:

Could you explain it a bit more? First of all, in your first message you used `/dev/ttyUSB0` while in the second you use `/dev/ttyUSB1`.

1. Are you sure you are using the right one?
2. Are you sure there is no other application that is using that serial port?

Could you try on a **fresh** beta2 firmware (I mean literally with all parameters initialized to the default values) to run again
```sh
microdds_client start -t serial -d /dev/ttyS3 -b 921600
```
If it does not work, could you try the same procedure on beta1?

> Also, I have tried to set XRCE_DDS_0_CFG param to TELEM2, to auto-initialize, but it doesn`t stored after reboot.

That is concerning. Are you sure there is no other module (like mavlink) that is using TELEM2 ?

RPS98, 2023-03-30:
Sorry, I didn't check Mavlink was using Telem1 and Telem2. Thank you.

It is also referred to issue #20634.

I hope a working version of v1.14 will be released soon. Maybe, ROS2 documentation should say v1.13 don't support microdds (It's a bit confusing: https://github.com/PX4/PX4-user_guide/blob/228144bfabe5eb311c46c9cb79d2c2275aabdff0/en/ros/ros2_comm.md?plain=1#L8)

## Describe the bug
I have already these problems, but they may be in another issues/questions:

Using v1.14.0-beta2 branch, I have these issues:
1. When I set XRCE_DDS_0_CFG to TELEM 3, microdds_client is not launched at startup. I have to launch it manually by Mavlink console, which it is a problem for real autonomous flights.
2. Can not launch Micro XRCE Agent after connecting TELEM 3 to the PC using USB cable. May be a [
Micro-XRCE-DDS](https://github.com/eProsima/Micro-XRCE-DDS) issue or Ubuntu problem. So I have to launch Micro XRCE Agent and then connect the USB cable to the PC, which it is also a problem for real autonomous flights. The output is:
```
$ MicroXRCEAgent serial -b 921600 --dev /dev/ttyUSB0
[1680167372.830435] error    | TermiosAgentLinux.cpp | init                     | open device error      | device: /dev/ttyUSB0, errno: 16
Error while starting serial agent!
```
3. When connect Micro XRCE Agent and Micro DDS Client, ROS2 topics are not working properly (see Log Files Section below). Timesync problems appears, some topics don't work and when try to do a ros2 echo, Agent and Client disconnect. I have to debug it.

## To Reproduce
Steps to reproduce the behavior:
1. Clone and build the firmware:
```
git clone --branch v1.14.0-beta2 https://github.com/PX4/PX4-Autopilot.git --recurse-submodules
cd PX4-Autopilot/
make px4_fmu-v6c
```
2. Open QGroundControl
3. Connect the Pixhawk v6C to the PC using the USB cable. The USB is ttyACM0.
4. Update the firmware: px4_fmu-v6c_default.px4
5. In QGroundControl, set parameter XRCE_DDS_0_CFG to TELEM 3 and reboot
6. Check parameter and it has the new value TELEM 3
7. Open Mavlink and check microdds. It is not launched at startup.
```
nsh> microdds_client status
INFO  [microdds_client] not running
nsh> 
```
8. Launch microdds_client at TELEM 3 ([PX4 v6c Serial Port Mapping](https://docs.px4.io/main/en/flight_controller/pixhawk6c.html#serial-port-mapping)):
```
nsh> microdds_client start -t serial -d /dev/ttyS1 -b 921600
nsh> microdds_client status
INFO  [microdds_client] Running, disconnected
INFO  [microdds_client] Payload tx: 0 B/s
INFO  [microdds_client] Payload rx: 0 B/s
nsh> 
```
9. Connect TELEM 3 to the PC using another USB cable.
10. Start Micro XRCE Agent. Checking the port, it is ttyUSB0.
```
MicroXRCEAgent serial -b 921600 --dev /dev/ttyUSB0
```

## Log Files and Screenshots
When launch microdds client on PX4 v6c board, the following output show up:
```
nsh> microdds_client start -t serial -d /dev/ttyS1 -b 921600
nsh> microdds_client status
INFO  [microdds_client] Running, disconnected
INFO  [microdds_client] Payload tx: 0 B/s
INFO  [microdds_client] Payload rx: 0 B/s
nsh> INFO  [microdds_client] synchronized with time offset 0    us
INFO  [microdds_client] successfully created rt/fmu/out/failsafe_flags data writer, topic id: 80
INFO  [microdds_client] successfully created rt/fmu/out/sensor_combined data writer, topic id: 168
INFO  [microdds_client] successfully created rt/fmu/out/timesync_status data writer, topic id: 190
INFO  [microdds_client] successfully created rt/fmu/out/vehicle_attitude data writer, topic id: 207
INFO  [microdds_client] successfully created rt/fmu/out/vehicle_control_mode data writer, topic id: 213
INFO  [microdds_client] successfully created rt/fmu/out/vehicle_local_position data writer, topic id: 220
INFO  [microdds_client] successfully created rt/fmu/out/vehicle_odometry data writer, topic id: 225
[... 106 lines cut ...]
[1680166619.807917] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x00000001, datawriter_id: 0x0CF(5), publisher_id: 0x0CF(3)
[1680166619.843723] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x0D5(2), participant_id: 0x001(1)
[1680166619.843746] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x00000001, publisher_id: 0x0D5(3), participant_id: 0x001(1)
[1680166619.843865] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x00000001, datawriter_id: 0x0D5(5), publisher_id: 0x0D5(3)
[1680166619.864697] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x0DC(2), participant_id: 0x001(1)
[1680166619.864732] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x00000001, publisher_id: 0x0DC(3), participant_id: 0x001(1)
[1680166619.864793] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x00000001, datawriter_id: 0x0DC(5), publisher_id: 0x0DC(3)
[1680166620.062529] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x0E1(2), participant_id: 0x001(1)
[1680166620.062569] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x00000001, publisher_id: 0x0E1(3), participant_id: 0x001(1)
[... cut ...]
[1680166647.427224] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3E8(2), participant_id: 0x001(1)
```

## Drone
- I have used Pixhawk 6c board, with USB port power and TELEM3 port connected to computer. Also try on a [PX4 Vision Autonomy Development Kit](https://docs.px4.io/main/en/complete_vehicles/px4_vision_kit.html).
- The computer uses [Ubuntu 22.04.2 LTS](https://releases.ubuntu.com/jammy) with [ROS 2 Humble](https://docs.ros.org/en/humble/index.html)

beniaminopozzan, 2023-03-31:
@RPS98 , I do not have Pixhawk 6c board but I tested everything on a CUAV Pixhawk 6X. I tried all the possible combination of:

- `main` and `1.14-beta2`.
- `TELEM2` and `TELEM3`.
- starting the client from console and starting it using the parameters.

and all the combinations worked without any problem:

- The client always start when it has to start.
- The agent always start and it never ever show any error.

I used `921600` as baudrate.

--

Please check again

- That your serial to usb adapter works properly. I checked mine starting MAVLINK over it and verifying that QGC can connect
- That your serial to usb adapter supports the baudrate you want to use. My adapter did not work with `3000000` as baudrate for example.
- That there is absolutely no other process in your Ubuntu that is trying to use the usb. The "Error while starting serial agent!" really tells me that something is wrong there. Also check that you have the permission to write on the serial port.
- That there is absolutely no other process in PX4 that is trying to use the serial port.
- That when you set `XRCE_DDS_0_CFG` you reboot the board and then set the proper baudrate and then you reboot again (you need up to 2 reboots to set device and baudrate). Only after the second reboot you the client will fully work.

Do you have any other serial port on the PX4 that are used?

RPS98, 2023-03-31:
Hi @mrpollo and @beniaminopozzan , finally the problem is solved.

I couldn't set TELEM3 baudrate (param SER_TEL3_BAUD does not appear in QGroundControl). Finally, I disabled MAVLink stream on TELEM2 and I enabled Micro XRCE-DDS on TELEM2 (and then change it baudrate to 921600), and it works fine, starting at boot.

Then, the Micro XRCE-DDS Agent problem was that QGroundControl was using TELEM2 too, so as Micro DDS Client was launch on boot, I could connect the Agent without usign QGroundControl.

Thank you very much for your help.

mrpollo, 2023-03-31:
Glad you got it working.

gustavotbarros, 2023-05-10:
Hi @beniaminopozzan, are you getting also problems with sensor calibration on Pixhawk 6X with main or v1.14-beta2?

Ive flashed today the board one more time with main, and even a problem with Crash Dump on SD came, when I moved the flight controller. But then I went back to v1.13.2 and the problems were gone.

beniaminopozzan, 2023-05-10:
Hi @gustavotbarros , please open a new issue or a topic in the forum so we don't go offtopic here.

mrodrigues14, 2023-08-29:
Hi @beniaminopozzan, i need some help with this topic, how can i connect my computer with MicroXRCEAgent to my drone, that is equipped with pixhawk 6c and i made the connection with computer to drone with esp32. I use gazebo normaly but when i try to connect with MicroXRCE that not conect. Is there any configuration that i need to made in esp32 or pixhawk?