PX4/PX4-Autopilot #21421: [EstimatorInterface::setSystemFlagData]:This function seems to have a bug

BreederBai, 2023-04-04:
[image]

It will never be true.
`if (time_us >= static_cast<int64_t>(_system_flag_buffer->get_newest().time_us + _min_obs_interval_us)) `
Variable system_ Flags and_ system_ flag_ buffer->get_ The newest() value is the same. That is to say, A-B>=A+C. When B and C are both positive numbers, this formula cannot hold.

bresch, 2023-04-04:
Thanks for reporting this @BreederBai , it looks like the first buffer push shouldn't be there (the correct one is protected by the `if`): https://github.com/PX4/PX4-Autopilot/pull/21422

BreederBai, 2023-04-04:
@bresch You're welcome. I'm glad to contribute to the improvement of the PX4 code

PR #21422 (2023-04-04): ekf2: remove duplicate buffer push of system flags

fixes #21421
closed 2023-04-04 by dagar