PX4/PX4-Autopilot #21424: Cyphal support for px4_fmu-v6c target

vmsav, 2023-04-04:
It would be great to provide out of box support of Cyphal for px4_fmu-v6c target! At this moment an error occurs when building firmware with enabled Cyphal driver (CONFIG_DRIVERS_CYPHAL=y, CONFIG_DRIVERS_UAVCAN=n). 
`PX4-Autopilot/src/drivers/cyphal/CanardHandle.cpp:52:4: error: #error "No CAN driver enabled for Cyphal build"`
@PonomarevDA

ekononykhin, 2023-04-13:
It will be great if Cyphal support becomes available for v6c in the near future. I'm currently using Cyphal and PWM at the ESC level and plan to switch fully to Cyphal.

PonomarevDA, 2023-04-14:
Besides modifying `default.px4board` or adding `cyphal.px4board`, you should also take care about nuttx-config. You can check the [fmu-v5/nuttx-config/cyphal](https://github.com/PX4/PX4-Autopilot/tree/main/boards/px4/fmu-v5/nuttx-config/cyphal) example. The user guide ([PX4 Board Configuration](https://docs.px4.io/main/en/hardware/porting_guide_config.html) and [NuttX Board Porting Guide](https://docs.px4.io/main/en/hardware/porting_guide_nuttx.html#nuttx-menuconfig-setup)) also makes things a bit clearer.
fmu-v6c has canfd, so the configuration might be a bit different from to fmu-v5, but I see it was considered [on the forum](https://discuss.px4.io/t/nuttx-can-drivers-on-stm32h7x3/29828).

I have fmu-v6c, but I will not be able to try it until the end of next week.