PX4/PX4-Autopilot #21462: list_vmd_make_targets disappeared but is still required.

hamishwillee, 2023-04-12:
## Describe the bug

The `list_vmd_make_targets` make target was/is extremely useful for working out what simulation targets are available. Otherwise you have to know in advance that they exist before you can use them.

The fact that it no longer works is also a big problem for docs - we have to document every option, and retest every option - there is no way to automate working out whether options disappeared or not. 

I understand this disappeared for a reason, but can we get it back again?

## To Reproduce

do

```
make list_vmd_make_targets
```

## Expected behavior

Make toolchain lists all the supported simulator targets

rosiakpiotr, 2023-04-23:
Worth to mention, official PX4 simulation [guide](https://docs.px4.io/main/en/sim_gazebo_classic/) refers to `make list_vmd_make_targets`, but as stated in the issue, it does not exist.

```
➜  PX4-Autopilot git:(main) ✗ make px4_sitl list_vmd_make_targets
ninja: error: unknown target 'list_vmd_make_targets'
make: *** [Makefile:232: px4_sitl] Error 1
```

kanishkanarch, 2023-06-12:
Is there any update on `list_vmd_make_targets`? Can we get it anytime soon?

alnI3S, 2023-09-18:
Hi @hamishwillee,

Do you know what is the reason the px4 dev group removes this `vmd_make_targets`?

hamishwillee, 2023-09-20:
@alnI3S I believe it is an artifact of the way that make files are done now. But I don't really know. I just think it sux.