PX4/PX4-Autopilot #21465: Build Issue on Cygwin Toolchain with PX4 v1.12.3

kginnell, 2023-04-12:
## Description
I downloaded the Cygwin Toolchain 0.8 and cloned PX4 using the Simulink UAV Toolbox, where PX4 v1.12.3 builds right out of the box. When I use the Cygwin terminal to try and build one of the examples, the "make px4_fmu-v5_default" ends in the error below. I even switched it back, so the code should be completely original. but PX4 does not build from terminal.

## Console Output
`$ make px4_fmu-v5_default
[0/1388] git submodule platforms/nuttx/NuttX/nuttx
[9/1388] git submodule platforms/nuttx/NuttX/apps
[11/1388] git submodule src/lib/ecl
[16/1388] git submodule src/drivers/gps/devices
[22/1388] git submodule src/drivers/uavcan/libuavcan
[23/1388] git submodule mavlink/include/mavlink/v2.0
[28/1388] Performing configure step for 'px4io_firmware'
-- PX4 version: v1.12.3
-- PX4 config file: /cygdrive/c/PX4_r2022b/home/Firmware/boards/px4/io-v2/default.cmake
-- PX4 config: px4_io-v2_default
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- build type is MinSizeRel
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-591-gb3fed06
-- Configuring done
-- Generating done
-- Build files have been written to: /cygdrive/c/PX4_r2022b/home/Firmware/build/px4_fmu-v5_default/external/Build/px4io_firmware
[28/1388] Performing build step for 'px4io_firmware'
[1/3] Linking CXX executable px4_io-v2_default.elf
Memory region         Used Size  Region Size  %age Used
           flash:       60436 B        60 KB     98.37%
            sram:        3856 B         8 KB     47.07%
[3/3] Creating /cygdrive/c/PX4_r2022b/home/Firmware/build/px4_fmu-v5_default/external/Build/px4io_firmware/px4_io-v2_default.px4
[34/1388] Generating romfs_extract.stamp
FAILED: ROMFS/romfs_extract.stamp
cd /cygdrive/c/PX4_r2022b/home/Firmware/build/px4_fmu-v5_default/etc && /usr/bin/cmake.exe -E remove_directory /cygdrive/c/PX4_r2022b/home/Firmware/build/px4_fmu-v5_default/etc/* && /usr/bin/cmake.exe -E tar xf /cygdrive/c/PX4_r2022b/home/Firmware/build/px4_fmu-v5_default/romfs_files.tar && /usr/bin/cmake.exe -E touch /cygdrive/c/PX4_r2022b/home/Firmware/build/px4_fmu-v5_default/ROMFS/romfs_extract.stamp
CMake Error: cmake version 3.14.5
Usage: /usr/bin/cmake -E <command> [arguments...]
Available commands:
  capabilities              - Report capabilities built into cmake in JSON format
  chdir dir cmd [args...]   - run command in a given directory
  compare_files [--ignore-eol] file1 file2
                              - check if file1 is same as file2
  copy <file>... destination  - copy files to destination (either file or directory)
  copy_directory <dir>... destination   - copy content of <dir>... directories to 'destination' directory
  copy_if_different <file>... destination  - copy files if it has changed
  echo [<string>...]        - displays arguments as text
  echo_append [<string>...] - displays arguments as text but no new line
  env [--unset=NAME]... [NAME=VALUE]... COMMAND [ARG]...
                            - run command in a modified environment
  environment               - display the current environment
  make_directory <dir>...   - create parent and <dir> directories
  md5sum <file>...          - create MD5 checksum of files
  sha1sum <file>...         - create SHA1 checksum of files
  sha224sum <file>...       - create SHA224 checksum of files
  sha256sum <file>...       - create SHA256 checksum of files
  sha384sum <file>...       - create SHA384 checksum of files
  sha512sum <file>...       - create SHA512 checksum of files
  remove [-f] <file>...     - remove the file(s), use -f to force it
  remove_directory dir      - remove a directory and its contents
  rename oldname newname    - rename a file or directory (on one volume)
  server                    - start cmake in server mode
  sleep <number>...         - sleep for given number of seconds
  tar [cxt][vf][zjJ] file.tar [file/dir1 file/dir2 ...]
                            - create or extract a tar or zip archive
  time command [args...]    - run command and display elapsed time
  touch file                - touch a file.
  touch_nocreate file       - touch a file but do not create it.
  create_symlink old new    - create a symbolic link new -> old

[35/1388] Copying NuttX compressed config nsh and inflating (make olddefconfig)
ninja: build stopped: subcommand failed.
make: *** [Makefile:226: px4_fmu-v5_default] Error 1`

kginnell, 2023-04-12:
# For context. this is the branch which is automatically checked out by Simulink:

`$ git status
HEAD detached at v1.12.3
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

        modified:   ROMFS/px4fmu_common/init.d/rc.mc_apps
        modified:   boards/px4/fmu-v5/default.cmake
        modified:   boards/px4/fmu-v5/nuttx-config/nsh/defconfig

Untracked files:
  (use "git add <file>..." to include in what will be committed)

        ROMFS/px4fmu_common/init.d/rc.mc_apps.original
        Tools/simulation/
        boards/px4/fmu-v5/default.cmake.original
        boards/px4/fmu-v5/init/rc.board_extras
        boards/px4/fmu-v5/nuttx-config/nsh/defconfig.original
        src/drivers/cyphal/
        src/lib/crypto/libtomcrypt/
        src/lib/crypto/libtommath/
        src/lib/events/
        src/modules/mavlink/mavlink/
        src/modules/microdds_client/
        src/modules/px4_simulink_app/

no changes added to commit (use "git add" and/or "git commit -a")`

zondafeng, 2024-08-19:
so ,i have the same problem. did you have solved this problem?

hamishwillee, 2024-08-21:
@zondafeng This isn't being maintained or looked at. IMO we're getting to the point where the Cygwin solution should be removed.

The best way to build on windows is to use WSL2 https://docs.px4.io/main/en/dev_setup/dev_env_windows_wsl.html

zondafeng, 2024-08-21:
A few days ago, I use simulink and wls2 to design my controller. But when i choose "run on board" mode in Simulink, it takes 6-7 minutes to build and deploy. When i choose "monitor & tune" in "run on board" mode in simulink, it takes 15 minutes to buid the firmware. I dont know how to solve this problem, so i give up this way "wsl2+simulink". and I use px4 toolchain in windows and matlab 2022b now.

> @zondafeng This isn't being maintained or looked at. IMO we're getting to the point where the Cygwin solution should be removed.
> 
> The best way to build on windows is to use WSL2 https://docs.px4.io/main/en/dev_setup/dev_env_windows_wsl.html

zondafeng, 2024-08-21:
In this issue, i gauss the problem may be caused by git. Because this issue is happend after i modified some changes in /src/modules. i am new to git. and now i couldnt solve this issue. it makes me confused and upset.

hamishwillee, 2024-08-22:
I'm sorry to hear that. Note again though, not supported from v1.12 and later https://docs.px4.io/main/en/dev_setup/dev_env_windows_cygwin.html#windows-development-environment-cygwin-based so even if you fix this issue you'll likely run into more.