PX4/PX4-Autopilot #21478: make px4_sitl_default gazebo failed

zondafeng, 2023-04-15:
ubuntu 20.04 gazebo9 
[image]
does anybody know how to solve?

JohnTGZ, 2023-04-20:
I faced a similar issue and solved it. You have to make sure to have installed any PX4 SITL dependencies for simulation with Gazebo. This is done after cloning this repo and specifying the "--gazebo" flag to the ubuntu.sh bash script as follows:
```
bash ./PX4-Autopilot/Tools/setup/ubuntu.sh --gazebo
```
Then, the following should build:
```
make px4_sitl gazebo-classic
```

farhangnaderi, 2023-05-01:
glad it solved. Could you close the issue?