PX4/PX4-Autopilot #21479: How to use vx, vy, and roll to control fixed wings through self setting interfaces

niceybc, 2023-04-15:
## How to use vx, vy, and roll to control fixed wings through self setting interfaces

As far as I know, the official control interfaces provided by PX4 only include position control and attitude angle thrust control. I want to use **setpoint_raw** topic in PX4 to introduce the control interface and achieve the control of fixed wings using vx, vy, and roll.[[SET_ATTITUDE_TARGET]](http://docs.px4.io/main/en/flight_modes/offboard.html#fixed-wing).

[[PX4 Internal Control]](http://docs.px4.io/main/en/flight_stack/controller_diagrams.html)

Thanks!

Jaeyoung-Lim, 2023-04-15:
@niceybc I don't see why this is a valid use case. Are you trying to control acceleration through roll? Then why not just provide acceleration setpoints?

niceybc, 2023-04-16:
When the roll angle in a two-dimensional plane does not consider height changes, it can be represented by ay. So can I directly use [[SET_POSITION_TARGET_LOCAL_NED]](https://mavlink.io/en/messages/common.html#SET_POSITION_TARGET_LOCAL_NED) if I want to control vx, vy, and ay? Because I found in the px4 document that it only mentioned *Position setpoint*

Jaeyoung-Lim, 2023-04-16:
You need to set the full reference. What is the expectation of not providing vz? Uncontrolled? Or zero?

niceybc, 2023-04-16:
I want to fly at a fixed altitude. So the pitch angle is controlled by TECS

Jaeyoung-Lim, 2023-04-16:
@niceybc Then set the vertical velocity as zero

niceybc, 2023-04-16:
@Jaeyoung-Lim I hope to use TECS to control the altitude and speed response of the aircraft. Roll angle is used to achieve single machine flight control or multi machine collaborative algorithms, such as L1.Not a constant.

Jaeyoung-Lim, 2023-04-16:
@niceybc My question is why you wouldn't use vx, vy, vz and let the low level control deal with tracking the reference.

I still don't see why you would want to control roll and directly, individual of vx and vy

niceybc, 2023-04-18:
> @niceybc My question is why you wouldn't use vx, vy, vz and let the low level control deal with tracking the reference.
> 
> I still don't see why you would want to control roll directly, individual of vx and vy

Sorry, my basic knowledge is not solid enough. What I really want is to use airspeed and roll angle. 
So, can I convert my expected airspeed and roll angle into vx, vy, and vz for control?