PX4/PX4-Autopilot #21480: My drone flew for some minutes (2 mins) without any issue but after this. It started acting weird . It started rotating and fell down.

pongthang, 2023-04-15:
Here I include the log file also [https://drive.google.com/file/d/1tZbrbUQqcw8FmPMaEFj38y1PwkeS01Dx/view?usp=share_link](url)

This is the results of the log file in flight review [https://drive.google.com/file/d/1IeaRIAfY-BV8wjCTeWZG7GnO9DWr2FO8/view?usp=share_link](url)

Video for the flight [https://youtu.be/CdoeyhWByFw](url)
Please suggest the problem could be and the solution also.
Thanks in advanced.

mrpollo, 2023-04-19:
Here's the link to flight review https://logs.px4.io/plot_app?log=fd42f073-e029-4a70-b239-97fed677d7ba

junwoo091400, 2023-04-19:
[image]

Are you using some sort of an external estimator (optical flow, vio)? The error messages are logged

pongthang, 2023-04-20:
> [image]
> 
> Are you using some sort of an external estimator (optical flow, vio)? The error messages are logged

No, I used only GPS (Pixhawk 4 GPS holybro)

pongthang, 2023-04-20:
What I did was  I added one raspberry pi 4 for forwarding mavlink data from pixhawk to a digital ocean server using mavproxy.py. I added on 4G internet module using SIM card to give internet to raspberry pi also. Is there any issues created by 4G internet module? I don't have any clue.

MaEtUgR, 2023-04-26:
@pongthang Thanks for the report and pinging me.

> Is there any issues created by 4G internet module?

I doubt that the companion or its internet connection cause this flight control issue.

It's not super clear to me at the moment what exactly is the root cause of the crash but what I can see is that even though the drone flew visually acceptable in hover it quickly reached its limits in maneuvers. It must be a fundamental issue either in the hardware setup or basic tuning and configuration. Because what lead to the crash is a cascade of bad rate control tracking caused by what looks like allocation problems.
[image]

I suggest you go through the hardware setup checking all important things like:
- **Is the center of gravity close to the center of the vehicle** and also where the autopilot is mounted? Is any sensor far away from that? If so can it be improved or at least correctly configured such that the autopilot knows?
- Is the autopilot mounted exactly in the right orientation? Is it mounted well to the frame? So pretty solid and not on a thin vibrating plate or similar? Is there anything else on the frame e.g. a long thin GPS mast that can cause vibrations?
- **Are good ESCs in use with fast reaction times? Do they support DSHOT? If not are the PWM limits set up correctly? E.g. in the log PWM goes from 1075 to 1949. Do all motors idle reliably at the same speed with a PWM of 1075 and increase the speed up until a WPM value of 1949?**
- Are the motors and propellers roughly balanced? Do they have enough thrust and are propeller discs not overlapping? **Are the arms slightly tilted the right way to help the yaw authority?**
Only if you checked all these things I'd suggest looking into tuning especially rate control tuning: https://docs.px4.io/main/en/config_mc/pid_tuning_guide_multicopter.html#rate-controller
And during those tests try out all the limits before flying position mode or even autonomous missions.

If you share additional information about the size of the drone, parts in use and/or a picture of the setup I might be able to spot something and give additional advice.

pongthang, 2023-04-26:
@MaEtUgR  , Unfortunately the drone was broken into halves and I did not take images of the drone. But I take some screenshots from our previous video of test flight (which was successful in same route same distance with the same drone ). 

1. I am using tarot hexacopter frame. 
[image]

2. Here how I mounted the flight controller and raspberry pi. Two levels were made, flight controller was placed on the topmost plate.
[image]

3. Two lipo batteries were attached below the frame.
[image]
 
4. Emax’s BLHeli Series 30A ESC with Oneshot was used. 
[image]

5.  Motors and propellers
[image]

[image]

Calibration of all sensors and escs were done properly as mentioned in PX4 documentation. Then I didn't do calibration for past 4 test flight(3 hovering and 1 long range -- 930 meters ). After that, crash happened.

pongthang, 2023-04-26:
> @pongthang Thanks for the report and pinging me.
> 
> > Is there any issues created by 4G internet module?
> 
> I doubt that the companion or its internet connection cause this flight control issue.
> 
> It's not super clear to me at the moment what exactly is the root cause of the crash but what I can see is that even though the drone flew visually acceptable in hover it quickly reached its limits in maneuvers. It must be a fundamental issue either in the hardware setup or basic tuning and configuration. Because what lead to the crash is a cascade of bad rate control tracking caused by what looks like allocation problems. [image]
> 
> I suggest you go through the hardware setup checking all important things like:
> 
> * **Is the center of gravity close to the center of the vehicle** and also where the autopilot is mounted? Is any sensor far away from that? If so can it be improved or at least correctly configured such that the autopilot knows?
> * Is the autopilot mounted exactly in the right orientation? Is it mounted well to the frame? So pretty solid and not on a thin vibrating plate or similar? Is there anything else on the frame e.g. a long thin GPS mast that can cause vibrations?
> * **Are good ESCs in use with fast reaction times? Do they support DSHOT? If not are the PWM limits set up correctly? E.g. in the log PWM goes from 1075 to 1949. Do all motors idle reliably at the same speed with a PWM of 1075 and increase the speed up until a PWM value of 1949?**
> * Are the motors and propellers roughly balanced? **Do they have enough thrust** and are propeller discs not overlapping? **Are the arms slightly tilted the right way to help the yaw authority?**
>   Only if you checked all these things I'd suggest looking into tuning especially rate control tuning: https://docs.px4.io/main/en/config_mc/pid_tuning_guide_multicopter.html#rate-controller
>   And during those tests try out all the limits before flying position mode or even autonomous missions.
> 
> If you share additional information about the size of the drone, parts in use and/or a picture of the setup I might be able to spot something and give additional advice.

I am not able to answer some of the questions that you provided as the drone is broken . But we are making another drone with same components. I will do all the test that you provided above and share you the results here.

mrpollo, 2023-04-26:
Please report back once you have your new vehicle we can help you make sure this doesn't happen again, meanwhile I'm closing this issue, please create a new one or post on our forums with your new vehicle