PX4/PX4-Autopilot #21494: Dangerous takeoff in Manual because PX4 v1.13.2 did not switch to Hold mode

Katawann, 2023-04-19:
## Describe the bug
On our hexacopter we had a crazy takeoff because the switch on the RC for the Flight mode was in Hold mode when power on the drone, but PX4 did not switch automatically to Hold position once we got the GPS position and remained in Manual mode (which is not assigned in our case...). We did not see this and when we took off, the drone climb up like crazy before we take back the control by switching to Position mode !..

## To Reproduce
Steps to reproduce the behavior:
1. Assigned one switch of your RC to Hold mode
2. Power on your drone with the switch of Flight Mode on Hold mode
3. Wait for the drone to start, it will stay in Manual instead to go to Hold mode even when good position is set

## Expected behavior
PX4 should automatically switch to Hold mode once a GPS position is set

## Drone (please complete the following information):
- Pixhawk 4 with PX4 v1.13.2
- Hexacopter

afwilkin, 2023-04-19:
@Katawann can you post flight logs here of this issue?

Katawann, 2023-04-20:
@afwilkin thank you for considering my case ! Please find the logs [here](https://review.px4.io/plot_app?log=8f7f22e7-0c87-4c13-a817-353bb64fce51)

Katawann, 2023-06-06:
Hot fixed by changing the manual mode to hold mode with my companion computer...

closed 2023-06-06 by Katawann