PX4/PX4-Autopilot #21501: VTOL Quad tailsitter without elevons & rudders

DimitrisPericleous, 2023-04-21:
Hi all,
I am currently working on a VTOL Quad Tailsitter drone with no control surfaces like elevons, rudders, etc. More specifically, it’s going to take off like a multirotor drone and then transition to a fixed a wing UAV. The differential thrust of the four motors will be utilized for controlling the fixed wing configuration. I’m using the PX4 as a hardware and the PX4 1.13 as a firmware.
As an airplane reference, I have selected the VTOL Quad Tailsitter with the following specific outputs:
MAIN1: motor 1
MAIN2: motor 2
MAIN3: motor 3
MAIN4: motor 4
MAIN5: elevon left
MAIN6: elevon right
MAIN7: canard surface
MAIN8: rudder
I have already determined the Main 1,2,3,4 and I would like to ask if the rest are optionals in my case or if I should input something there as well? And, if so, what should it be?

sfuhrer, 2023-06-09:
I'm doing a similar project right now, flying the Swan K1 with release/1.14. Flight log: https://review.px4.io/plot_app?log=994f573f-c95b-497e-997a-c5610359b008
Note that I had to make one code adaption in the FW rate controller to not have it constantly reset the integrators (which has a big impact on flight performance).

julianoes, 2023-06-10:
@sfuhrer can you point me to the commit regarding integrator resets? I need to make sure I have it.

sfuhrer, 2023-06-12:
> @sfuhrer can you point me to the commit regarding integrator resets? I need to make sure I have it.

The commit I flew with was a dirty hack to make it fly, here is a somewhat cleaner version: https://github.com/PX4/PX4-Autopilot/pull/21713

junwoo091400, 2023-06-29:
It would be nice to have some documentation on the setup of the Quad Tailsitter, as [hamish suggested](https://discord.com/channels/1022170275984457759/1123615141410373792/1123755440459743324)!

DimitrisPericleous, 2023-09-27:
Could you please provide information regarding the minimum number of required actuators and which among these actuators are considered optional and mandatory in this airplane reference?

sfuhrer, 2023-09-28:
>Could you please provide information regarding the minimum number of required actuators and which among these actuators are considered optional and mandatory in this airplane reference?

DimitrisPericleous, 2023-09-28:
> > Could you please provide information regarding the minimum number of required actuators and which among these actuators are considered optional and mandatory in this airplane reference?
> 
> For an tailsitter without control surfaces? You need 4 motors, and you need to make sure that you have enough FW roll authority by tilting them accordingly. If you additionally want to add control surfaces then you should have more control authority, but tuning it properly may get hard. Never tried that myself.

So, for a surfaceless Quad Tailsitter, the only thing I need to determine is the motors. The airplane, a reference VTOL Quad Tailsitter which I have chosen, has the following specific outputs:
MAIN1: Motor 1
MAIN2: Motor 2
MAIN3: Motor 3
MAIN4: Motor 4
MAIN5: Elevon left
MAIN6: Elevon right
MAIN7: Canard surface
MAIN8: Rudder

Is it okay not to determine MAIN 5-8? Are there any other adjustments I need to make to fly as a surfaceless quad tailsitter?

DimitrisPericleous, 2023-09-28:
> > Is it okay not to determine MAIN 5-8? Are there any other adjustments I need to make to fly as a surfaceless quad tailsitter?
> 
> You mean to remove 5-8? Yes, if you want to fly without surfaces you can remove them. I don't know what your platform looks like so can't really give you hints for adjustments.

I'm using the PX4 as hardware, PX4 1.13 as firmware, and I've selected the VTOL Quad Tailsitter as the airplane reference. My UAV is equipped with 4 motors. I won't be inputting anything at the outputs MAIN 5-8, as you mentioned. Will Pixhawk be capable of controlling the roll, pitch, and yaw of the UAV as well as the transition from VTOL to Fixed Wing without any further adjustments? Or do I need to make changes to the firmware in order to control the UAV's movement using differential thrust?Thank you in advance for your time!!!

sfuhrer, 2023-10-30:
You will need 1.14, but then it should be possible without further code changes.