PX4/PX4-Autopilot #21516: IMU data shows non-zero values for the 2nd element of the quaternion even when the quadcopter is on the ground!

MOCACHICA, 2023-04-26:
## Attention! Please read the note below

We use GitHub issues only to discuss PX4 bugs and new features.

**For questions about using PX4 or related components, please use [PX4 Discuss](http://discuss.px4.io/).**

Thanks!

MOCACHICA, 2023-04-27:
I executed a code using roslaunch and read the data from the IMU output, and it appeared to work flawlessly. However, upon checking the simulation environment (Gazebo in this case), an issue arose. Instead of remaining stable, the quadcopter began to move uncontrollably.

Check the file: 
https://drive.google.com/file/d/1g0BIuFVUck2_H3bD4aiFwYcX9kxK9mcT/view?usp=sharing

bresch, 2023-04-28:
>  Instead of remaining stable, the quadcopter began to move uncontrollably.

In your video, I can see a drone hovering. When is it moving "uncontrollably"?

> The IMU data indicates that all three elements of the quaternions are zero even though the quad has a pitch angle rotation and is in motion.

I just see the drone hovering in the video, so it's expected to have the x an y elements of the quaternion close to 0

MOCACHICA, 2023-04-28:
> In your video, I can see a drone hovering. When is it moving "uncontrollably"?
>
>
> I just see the drone hovering in the video, so it's expected to have the x an y elements of the quaternion close to 0

It does not hover. it moves in the direction of the x-axis.