PX4/PX4-Autopilot #21518: Propellers not spinning in offboard with thrust and torque setpoint

salvamarce, 2023-04-26:
## Describe the bug
### v1.13.3
I am trying to control the iris in Offboard flight mode, sending the thrust and torque setpoints with ROS 2 and the micrortps agent. The drone gets armed but the propellers don't spin, even if the setpoints are present on the topic.

## To Reproduce
Steps to reproduce the behavior:
1. Add 
```
- msg: vehicle_torque_setpoint
   receive: true
- msg: vehicle_thrust_setpoint
  receive: true
```
 into `msg/tools/urtps_bridge_topics.yaml` and generate the topic by launching the script.
2. Clean the ROS 2 workspace and build again with the script in the package `px4_ros_com`
3. Set `<enable_lockstep>0</enable_lockstep>` in the `.sdf.jinja` file of the desired drone (iris)
4. Start the simulation with `make px4_sitl_rtps gazebo`
5. Start the micrortps agent with `micrortps_agent -t UDP`
6. Check that the topics for the torques and thrust are available
7.  Open QGC 
8. Then I launch my offboard application, which consists just in arming the drone and after sending full thrust on Z
9. The drone gets armed but the propellers don't spin (but the topic is receiving the message)

## Expected behavior
The drone should arm and then the propellers should spin 

## Log Files and Screenshots
https://review.px4.io/plot_app?log=ebfc4251-ef61-4a8a-b592-deee7df20a88

## Drone 
- Iris sitl_gazebo

## Additional context
Is there an example of offboard with thrust and torque setpoints?

beniaminopozzan, 2023-04-26:
I don't know if that is the main issue, but you are missing a step when disabling the lockstep https://docs.px4.io/main/en/simulation/#disable-lockstep-simulation

How did you fill the OffboardControlMode message?

while you are in offboard mode and the drone should respond to your vehicle_thrust_setpoint and vehicle_torque_setpoints, what is the output of `listener vehicle_thrust_setpoint` and `listener vehicle_torque_setpoints`?

salvamarce, 2023-04-28:
> I don't know if that is the main issue, but you are missing a step when disabling the lockstep https://docs.px4.io/main/en/simulation/#disable-lockstep-simulation

Yes, I've also done that. I forgot to put that in the steps

> How did you fill the OffboardControlMode message?

This is my message
```
OffboardControlMode msg{};
msg.timestamp = timestamp_.load();
msg.position = false;
msg.velocity = false;
msg.acceleration = false;
msg.attitude = false;
msg.body_rate = false;
msg.actuator = true;

offboard_control_mode_publisher_->publish(msg);
```
> While you are in offboard mode and the drone should respond to your vehicle_thrust_setpoint and vehicle_torque_setpoints, what is the output of listener vehicle_thrust_setpoint and listener vehicle_torque_setpoints?

While the drone is arming I set everything to 0.0 and after it is armed, I sent full thrust
/fmu/vehicle_torque_setpoint/in:
```
timestamp: 1682701663360937
timestamp_sample: 0
xyz:
- 0.0
- 0.0
- 0.0
---
```

/fmu/vehicle_thrust_setpoint/in
```
timestamp: 1682701663360937
timestamp_sample: 0
xyz:
- 0.0
- 0.0
- 1.0
---
```
I've also tried to send `-1` on the Z, thinking that it was the problem but nothing changed.

beniaminopozzan, 2023-04-28:
@salvamarce , please fill `timestamp_sample` too and give it another try.

FYI, for standard quadrotors vehicle_thrust_setpoint.xyz[3] (the z) is negative as the frame is FRD

I should have asked this earlier, but you have the control allocation module active, right?

salvamarce, 2023-04-28:
I found the error. The problem was with the configuration of the control allocation parameters that were not all set (I forgot the PWM_MAIN_FUNC).
Actually, it seems that is not mandatory to change the boardconfig if the lockstep is disabled in the URDF.

Is there a documentation section about the offboard flight mode with torque and thrust setpoints? because I struggled a bit these days, just to reach pieces of information here and there.

Anyway, thanks for your kind help, and sorry if I opened a silly issue

beniaminopozzan, 2023-04-28:
Great!

The documentation for main is here: https://docs.px4.io/main/en/flight_modes/offboard.html#copter

FYI, as you may have noticed, it is very hard to solve issues about offboard control as PX4 is not the only component involved, therefore just reproducing the issue itself can be very problematic.

salvamarce, 2023-04-29:
Ok thanks.
I was using the documentation for the version v1.13, because I'm using the v1.13.3.

closed 2023-04-29 by salvamarce

rrajnidhi, 2023-05-16:
Hello all,
I was wondering if offboard with thrust and torque setpoint will work with fixed wing or with VTOL ?