PX4/PX4-Autopilot #21519: m1 mac build error

mengchaoheng, 2023-04-26:
```Console
➜  PX4-Autopilot git:(df-1.12.3) git submodule update --init --recursive 
Submodule 'Tools/flightgear_bridge' (https://github.com/PX4/PX4-FlightGear-Bridge.git) registered for path 'Tools/flightgear_bridge'
Submodule 'Tools/jMAVSim' (https://github.com/PX4/jMAVSim.git) registered for path 'Tools/jMAVSim'
Submodule 'Tools/jsbsim_bridge' (https://github.com/PX4/px4-jsbsim-bridge.git) registered for path 'Tools/jsbsim_bridge'
Submodule 'Tools/sitl_gazebo' (https://github.com/mengchaoheng/DF_gazebo.git) registered for path 'Tools/sitl_gazebo'
Submodule 'boards/atlflight/cmake_hexagon' (https://github.com/PX4/cmake_hexagon.git) registered for path 'boards/atlflight/cmake_hexagon'
Submodule 'mavlink/include/mavlink/v2.0' (https://github.com/mavlink/c_library_v2.git) registered for path 'mavlink/include/mavlink/v2.0'
Submodule 'platforms/nuttx/NuttX/apps' (https://github.com/PX4/NuttX-apps.git) registered for path 'platforms/nuttx/NuttX/apps'
Submodule 'platforms/nuttx/NuttX/nuttx' (https://github.com/PX4/NuttX.git) registered for path 'platforms/nuttx/NuttX/nuttx'
Submodule 'platforms/qurt/dspal' (https://github.com/ATLFlight/dspal.git) registered for path 'platforms/qurt/dspal'
Submodule 'src/drivers/gps/devices' (https://github.com/PX4/PX4-GPSDrivers.git) registered for path 'src/drivers/gps/devices'
Submodule 'src/drivers/uavcan/libuavcan' (https://github.com/PX4/libuavcan.git) registered for path 'src/drivers/uavcan/libuavcan'
Submodule 'src/drivers/uavcan_v1/legacy_data_types' (https://github.com/PX4/public_regulated_data_types.git) registered for path 'src/drivers/uavcan_v1/legacy_data_types'
Submodule 'src/drivers/uavcan_v1/libcanard' (https://github.com/UAVCAN/libcanard.git) registered for path 'src/drivers/uavcan_v1/libcanard'
Submodule 'src/drivers/uavcan_v1/public_regulated_data_types' (https://github.com/UAVCAN/public_regulated_data_types.git) registered for path 'src/drivers/uavcan_v1/public_regulated_data_types'
Submodule 'src/drivers/uavcannode_gps_demo/libcanard' (https://github.com/UAVCAN/libcanard.git) registered for path 'src/drivers/uavcannode_gps_demo/libcanard'
Submodule 'src/drivers/uavcannode_gps_demo/public_regulated_data_types' (https://github.com/UAVCAN/public_regulated_data_types.git) registered for path 'src/drivers/uavcannode_gps_demo/public_regulated_data_types'
Submodule 'src/lib/crypto/monocypher' (https://github.com/PX4/Monocypher.git) registered for path 'src/lib/crypto/monocypher'
Submodule 'src/lib/ecl' (https://github.com/PX4/PX4-ECL.git) registered for path 'src/lib/ecl'
Submodule 'src/lib/matrix' (https://github.com/PX4/PX4-Matrix.git) registered for path 'src/lib/matrix'
Submodule 'src/modules/micrortps_bridge/micro-CDR' (https://github.com/PX4/Micro-CDR.git) registered for path 'src/modules/micrortps_bridge/micro-CDR'
Submodule path 'Tools/flightgear_bridge': checked out '7c2c623da9f5dcb3f01d32830af1a2ed958de363'
Submodule path 'Tools/flightgear_bridge/models/Rascal': checked out 'cbdb2fc7dd44331da1dcd04a891ed7f8c27b1ef0'
Submodule path 'Tools/flightgear_bridge/models/TF-G1': checked out '4540de7ae8e71b9bd6e6f67d085695be0ab36bc3'
Submodule path 'Tools/flightgear_bridge/models/TF-G2': checked out '562ee71388c493675269210f7692d8b7a8bb800d'
Submodule path 'Tools/flightgear_bridge/models/TF-R1': checked out '158e429a0110589b1e2adfe868e166271470e144'
Submodule path 'Tools/jMAVSim': checked out '0a816d8100e9860a031ccfccfb9941bcd6eedc81'
Submodule path 'Tools/jMAVSim/jMAVlib': checked out 'c08380767b21938391bec676a43b36b837af9ba4'
Submodule path 'Tools/jsbsim_bridge': checked out 'e070c50d591541d440510b65c9da5cad1db5aa9a'
Submodule path 'Tools/jsbsim_bridge/models/ATI-Resolution': checked out '40571cd808a204f4062128daeb6bb6db8c134c48'
Submodule path 'Tools/jsbsim_bridge/models/Rascal': checked out 'd0d1db1961549e557885002cf9c3cff40cc27be2'
Submodule path 'Tools/sitl_gazebo': checked out '53ddbde987e0abb94e93d351b460ab914ef415e2'
Submodule path 'Tools/sitl_gazebo/external/OpticalFlow': checked out '28ef45c1fcb532b2bfd54c16b059c6a545143b2f'
Submodule path 'Tools/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker': checked out '9b5b567ae77f93d1ef4d36151529ba2506a46e7a'
Submodule path 'boards/atlflight/cmake_hexagon': checked out '08fd0a73045346448adf6969660196228b23e1fa'
Submodule path 'mavlink/include/mavlink/v2.0': checked out '636fae5ca34eff6007344718755d78fcdf94238e'
Submodule path 'platforms/nuttx/NuttX/apps': checked out '979e27408fa90a3c3d6432190c8f65cc80c3e6b6'
Submodule path 'platforms/nuttx/NuttX/nuttx': checked out 'bf660cba2af81f055002b3817c87b1f63a78fd09'
Submodule path 'platforms/qurt/dspal': checked out '0322a4e345e48ea28cb1cee14a33033cdaf0b16a'
Submodule path 'platforms/qurt/dspal/cmake_hexagon': checked out '07168bd5715818802b78f674816ec851307998a7'
Submodule path 'src/drivers/gps/devices': checked out 'f51267815db3331723b880ab20afc5a67a87c22b'
Submodule path 'src/drivers/uavcan/libuavcan': checked out '29e1aa6c12e3590252b102153d8c62155047a676'
Submodule path 'src/drivers/uavcan/libuavcan/dsdl': checked out '192295c4f9b67f4a20b0eabf74757b6597415f2b'
Submodule path 'src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan': checked out 'c58477a644d20ccf95a20c151f3a0402f271c3b8'
Submodule path 'src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan/dsdl': checked out 'fd12483ddd4e58242d61d74a163e7aeaa1e0f466'
Submodule path 'src/drivers/uavcan_v1/legacy_data_types': checked out '36a01e428b110ff84c8babe5b65667b5e3037d5e'
Submodule path 'src/drivers/uavcan_v1/libcanard': checked out '2a116170285fb47fcaae150ad21c2ccde0756a5f'
Submodule path 'src/drivers/uavcan_v1/public_regulated_data_types': checked out '0a773b93ce5c94e1d2791b180058cb9897fab7e1'
Submodule path 'src/drivers/uavcannode_gps_demo/libcanard': checked out '2a116170285fb47fcaae150ad21c2ccde0756a5f'
Submodule path 'src/drivers/uavcannode_gps_demo/public_regulated_data_types': checked out '0a773b93ce5c94e1d2791b180058cb9897fab7e1'
Submodule path 'src/lib/crypto/monocypher': checked out 'baca5d31259c598540e4d1284bc8d8f793abf83a'
Submodule path 'src/lib/ecl': checked out 'b3fed06fe822d08d19ab1d2c2f8daf7b7d21951c'
Submodule path 'src/lib/matrix': checked out 'b8568a89db8455d0ab2e2e391e2149ba2e5e10dd'
Submodule path 'src/modules/micrortps_bridge/micro-CDR': checked out '21d3cfe3ae570d1674da0105ab23b80958e0449a'
➜  PX4-Autopilot git:(df-1.12.3) make px4_sitl gazebo
-- PX4 version: v1.12.3-90-g602fa609ea
-- PX4 config file: /Users/mch/Proj/Mac_DF/PX4-Autopilot/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is AppleClang 13.1.6.13160021
-- The C compiler identification is AppleClang 13.1.6.13160021
-- The ASM compiler identification is Clang with GNU-like command-line
-- Found assembler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
[... 626 lines cut ...]
CMake Generate step failed.  Build files cannot be regenerated correctly.
[316/822] Building CXX object platforms/posix/src/px4/common/CMakeFiles/px4_layer.dir/print_load.cpp.o
/Users/mch/Proj/Mac_DF/PX4-Autopilot/platforms/posix/src/px4/common/print_load.cpp:119:11: warning: variable 'stat_thread' set but not used [-Wunused-but-set-variable]
        uint32_t stat_thread = 0;
                 ^
1 warning generated.
[318/822] Building CXX object src/lib/mixer_module/CMakeFiles/mixer_module.dir/mixer_module.cpp.o
In file included from /Users/mch/Proj/Mac_DF/PX4-Autopilot/src/lib/mixer_module/mixer_module.cpp:34:
/Users/mch/Proj/Mac_DF/PX4-Autopilot/src/lib/mixer_module/mixer_module.hpp:320:15: warning: private field '_B' is not used [-Wunused-private-field]
        const double _B[3][4] = { {-0.5,0.0,0.5,0.0}, {0.0,-0.5,0.0,0.5},{0.25,0.25,0.25,0.25}};
                     ^
1 warning generated.
[453/822] Building CXX object src/modules/flight_mode_manager/tasks/Orbit/CMakeFiles/FlightTaskOrbit.dir/FlightTaskOrbit.cpp.o
/Users/mch/Proj/Mac_DF/PX4-Autopilot/src/modules/flight_mode_manager/tasks/Orbit/FlightTaskOrbit.cpp:210:17: warning: unused variable 'center_to_position' [-Wunused-variable]
[... cut ...] Users/mch/Proj/Mac_DF/PX4-Autopilot/Tools/sitl_gazebo -B /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.26.3/bin/cmake -E touch /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
[509/822] Building CXX object src/modules/mavlink/CMakeFiles/modules__mavlink.dir/mavlink_main.cpp.o
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1
➜  PX4-Autopilot git:(df-1.12.3) 
```

mengchaoheng, 2023-04-26:
```Console
PX4-Autopilot git:(df-1.12.3) brew install px4-sim-gazebo

Warning: Use gstreamer instead of deprecated gst-libav
Warning: Use gstreamer instead of deprecated gst-plugins-bad
Warning: Use gstreamer instead of deprecated gst-plugins-base
Warning: Use gstreamer instead of deprecated gst-plugins-good
Warning: Use gstreamer instead of deprecated gst-plugins-ugly
Warning: Use gstreamer instead of deprecated gst-libav
Warning: Use gstreamer instead of deprecated gst-plugins-bad
Warning: Use gstreamer instead of deprecated gst-plugins-base
Warning: Use gstreamer instead of deprecated gst-plugins-good
Warning: Use gstreamer instead of deprecated gst-plugins-ugly
Warning: Use gstreamer instead of deprecated gst-libav
Warning: Use gstreamer instead of deprecated gst-plugins-bad
Warning: Use gstreamer instead of deprecated gst-plugins-base
Warning: Use gstreamer instead of deprecated gst-plugins-good
Warning: Use gstreamer instead of deprecated gst-plugins-ugly
Warning: Use gstreamer instead of deprecated gst-libav
Warning: Use gstreamer instead of deprecated gst-plugins-bad
Warning: Use gstreamer instead of deprecated gst-plugins-base
Warning: Use gstreamer instead of deprecated gst-plugins-good
Warning: Use gstreamer instead of deprecated gst-plugins-ugly
Warning: Use gstreamer instead of deprecated gst-libav
Warning: Use gstreamer instead of deprecated gst-plugins-bad
Warning: Use gstreamer instead of deprecated gst-plugins-base
Warning: Use gstreamer instead of deprecated gst-plugins-good
Warning: Use gstreamer instead of deprecated gst-plugins-ugly
Warning: Use gstreamer instead of deprecated gst-libav
Warning: Use gstreamer instead of deprecated gst-libav
Warning: Use gstreamer instead of deprecated gst-libav
Warning: Use gstreamer instead of deprecated gst-plugins-bad
Warning: Use gstreamer instead of deprecated gst-plugins-bad
Warning: Use gstreamer instead of deprecated gst-plugins-bad
Warning: Use gstreamer instead of deprecated gst-plugins-base
Warning: Use gstreamer instead of deprecated gst-plugins-base
Warning: Use gstreamer instead of deprecated gst-plugins-base
Warning: Use gstreamer instead of deprecated gst-plugins-good
Warning: Use gstreamer instead of deprecated gst-plugins-good
Warning: Use gstreamer instead of deprecated gst-plugins-good
Warning: Use gstreamer instead of deprecated gst-plugins-ugly
Warning: Use gstreamer instead of deprecated gst-plugins-ugly
Warning: Use gstreamer instead of deprecated gst-plugins-ugly
Warning: Use gstreamer instead of deprecated gst-libav
Warning: Use gstreamer instead of deprecated gst-libav
Warning: Use gstreamer instead of deprecated gst-plugins-bad
Warning: Use gstreamer instead of deprecated gst-plugins-bad
Warning: Use gstreamer instead of deprecated gst-plugins-base
Warning: Use gstreamer instead of deprecated gst-plugins-base
Warning: Use gstreamer instead of deprecated gst-plugins-good
Warning: Use gstreamer instead of deprecated gst-plugins-good
Warning: Use gstreamer instead of deprecated gst-plugins-ugly
Warning: Use gstreamer instead of deprecated gst-plugins-ugly
==> Fetching dependencies for px4/px4/px4-sim-gazebo: exiftool
==> Fetching exiftool
==> Downloading https://ghcr.io/v2/homebrew/core/exiftool/manifests/12.60
Already downloaded: /Users/mch/Library/Caches/Homebrew/downloads/e5f1e92121e2bbd5c9cde721f0787e943f59f9ee541db5572906c23c4064c5bb--exiftool-12.60.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/exiftool/blobs/sha256:7530b6e640f91294c223c2e97722145d91feea0917a6716da
Already downloaded: /Users/mch/Library/Caches/Homebrew/downloads/28da962283415bd413aab778d61cf3b5f0cc5a2a636568b54358cf2ebcead294--exiftool--12.60.monterey.bottle.tar.gz
==> Fetching px4/px4/px4-sim-gazebo
==> Downloading https://raw.githubusercontent.com/PX4/PX4-Autopilot/master/Tools/px4.py
Already downloaded: /Users/mch/Library/Caches/Homebrew/downloads/be4aff9b473b8b0c6bae0ff08a6611885de9d56ee3a477c05453bb6bbcaab246--px4.py
==> Installing px4-sim-gazebo from px4/px4
Warning: Use gstreamer instead of deprecated gst-libav
Warning: Use gstreamer instead of deprecated gst-plugins-bad
Warning: Use gstreamer instead of deprecated gst-plugins-base
Warning: Use gstreamer instead of deprecated gst-plugins-good
Warning: Use gstreamer instead of deprecated gst-plugins-ugly
Error: Operation already in progress for gstreamer.formula
Another active Homebrew process is already using gstreamer.formula.
Please wait for it to finish or terminate it to continue.
➜  PX4-Autopilot git:(df-1.12.3) 
```

mengchaoheng, 2023-04-26:
```
➜  ~ brew config
HOMEBREW_VERSION: 4.0.15
ORIGIN: https://github.com/Homebrew/brew
HEAD: 763c41f006ab9b44261e313b0aaa24a047431b06
Last commit: 3 days ago
Core tap origin: https://github.com/Homebrew/homebrew-core
Core tap HEAD: 225bc7f9819d190ab6952062dd63d0a530828652
Core tap last commit: 4 days ago
Core tap branch: master
Core tap JSON: 26 Apr 22:28 UTC
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.jOkN1HJp2R/org.xquartz:0
HOMEBREW_MAKE_JOBS: 10
Homebrew Ruby: 2.6.10 => /usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.6.10_1/bin/ruby
CPU: 10-core 64-bit westmere
Clang: 13.1.6 build 1316
Git: 2.32.0 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Curl: 7.79.1 => /usr/bin/curl
macOS: 12.3-x86_64
CLT: 13.3.0.0.1.1645755326
Xcode: 13.3
Rosetta 2: true
➜  ~ 
```

mengchaoheng, 2023-04-27:
but on a arm version brew, this is no problem, so we don't need to use rosseta to change the arm terminal to x86?
```
➜  ~ brew install  gst-libav
Running `brew update --auto-update`...
==> Fetching dependencies for gst-libav: libtiff, little-cms2, webp, freetype, glib, openjpeg, leptonica, graphene, faac, faad2, fdk-aac, gsettings-desktop-schemas, glib-networking, atk, gdk-pixbuf, hicolor-icon-theme, libepoxy, gtk+3, gtk4, json-glib, libpthread-stubs, libshout, libpsl, libsoup, libusrsctp, orc, gobject-introspection, py3cairo, pygobject3, rtmpdump, srtp, taglib, libcuefile, libreplaygain, musepack, gstreamer and gst-plugins-base
==> Fetching libtiff
==> Downloading https://ghcr.io/v2/homebrew/core/libtiff/manifests/4.5.0
Already downloaded: /Users/mch/Library/Caches/Homebrew/downloads/79217b88b5df9d2506bfb4fa1a3ecd0a9fe80b6a136a1325cc7aab8790d682fd--libtiff-4.5.0.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/libtiff/blobs/sha256:74f7135659b8204c8e67
==> Downloading from https://pkg-containers.githubusercontent.com/ghcr1/blobs/sha256:74f71
################################################################################### 100.0%
==> Fetching little-cms2
==> Downloading https://ghcr.io/v2/homebrew/core/little-cms2/manifests/2.14_1
Already downloaded: /Users/mch/Library/Caches/Homebrew/downloads/7640158537f6618004a28d170e2520e56a379de9320a52701358f5c117497875--little-cms2-2.14_1.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/little-cms2/blobs/sha256:3c4cc7b5b25f90f9
==> Downloading from https://pkg-containers.githubusercontent.com/ghcr1/blobs/sha256:3c4cc
################################################################################### 100.0%
==> Fetching webp
==> Downloading https://ghcr.io/v2/homebrew/core/webp/manifests/1.3.0_1
################################################################################### 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/webp/blobs/sha256:181852278a6be20e21b8bf2
==> Downloading from https://pkg-containers.githubusercontent.com/ghcr1/blobs/sha256:18185
^C
➜  ~ 
```

mengchaoheng, 2023-04-27:
fix by update the correct sitl_gazebo commit 4791211c5867b0461de93b8e6fb03ae248b0d9a2, since this is build for v1.12.3

closed 2023-04-27 by mengchaoheng

cj-chua, 2023-04-28:
@mengchaoheng How did you resolve this brew error?
```
Error: Operation already in progress for gstreamer.formula
Another active Homebrew process is already using gstreamer.formula.
```

mengchaoheng, 2023-04-28:
> @mengchaoheng How did you resolve this brew error?
> 
> ```
> 
> Error: Operation already in progress for gstreamer.formula
> 
> Another active Homebrew process is already using gstreamer.formula.
> 
> ```

restore the /opt folder's files by Time Machine of MacBook Pro.