PX4/PX4-Autopilot #21552: Problem with flashing PX4 boot loader for KakuteH7v2

Mitchell-Lee-93, 2023-05-04:
Hello,

I recently got KakuteH7v2 (which has STM32H743) and wanted to use it with PX4 firmware.
I was following this [documentation](https://docs.px4.io/main/en/flight_controller/kakuteh7v2.html) to do so. However, while flashing the PX4 boot loader KakuteH7v2, ([Hex file](https://raw.githubusercontent.com/PX4/PX4-user_guide/main/assets/flight_controller/kakuteh7v2/holybro_kakuteh7v2_bootloader.hex)) I got the following error which seems there is a compatibility issue with the hex file and the KakuteH7v2.

[image]

PS. I could force it to flash 'holybro_kakuteh7v2_bootloader.hex' using 'dfu-util', but either the newest QGC or manual uploading using 'make holybro_kakuteh7v2_default upload' wasn't able to recognize the bootloader.

julianoes, 2023-05-07:
I have the same problem when trying to flash the hex file using the Betaflight Configurator. However, when I flash the bootloader with `dfu-util` it's working fine.

Two things to check:
1. Have you flashed the `.bin` (not the .hex) file with `dfu-util`? Like so: `dfu-util -a 0 --dfuse-address 0x08000000:force -D boards/holybro/kakuteh7v2/extras/holybro_kakuteh7v2_bootloader.bin`
2. If that is still not working try the step to do a mass erase first: https://docs.px4.io/main/en/advanced_config/bootloader_update_from_betaflight.html#dfu-util

Mitchell-Lee-93, 2023-05-08:
It works!

I used 'boards/holybro/kakuteh7v2/extras/holybro_kakuteh7v2_bootloader.bin' instead of the Hex file and now everything works fine.

Thank you very much for the help!

closed 2023-05-08 by Mitchell-Lee-93