PX4/PX4-Autopilot #21555: Dangerous behavior with RTK management

Katawann, 2023-05-07:
## Description
We need to use RTK on our drones to bring even more accuracy in the positioning of our drone. We have had several issues between PX4 and RTK when switching to/from RTK that have forced us to disable it for the time being... I don't know if others have used RTK a lot with PX4, but either we don't configure some parameters well, or there is some work to be done at the software level to make the safeties more robust

## To Reproduce
You must have a drone equipped with a device with RTK functionality (in our case an ublox F9P RTK). You must then activate the RTK stream to switch the device in RTK mode. In our case we retrieve the RTK stream via the Internet and send it to the ublox through an companion computer, but it could do through QGC for example.

The issue are coming when the GPS is switching from or to the RTK mode. When this happens, the xyz correction of RTK is added (respectively removed) which leads to a replacement of the drone even in the air...

## Expected behavior
The expected behavior is up for discussion, but there are two scenarii to consider and protect I think:
   - Switch to RTK mode: I don't know if PX4 can do anything about this, maybe it's up to the user to disable the RTK stream once the drone is armed to ensure that the correction will not be applied in flight
   - Exit RTK mode: for this one the safest behavior would be to maintain the last correction received. Is it something possible to do in PX4 ?

## Log Files
We have had several flights (with sometime crash), which demonstrate the problems of switching to/from RTK mode. The logs come from 4 different drones using the same configuration

   - Switch to RTK mode while in air:
     - https://logs.px4.io/plot_app?log=a75e8cf0-8323-4ba9-84ff-85dcdab221b1
     - https://logs.px4.io/plot_app?log=dc1d7534-43e3-44d7-9698-73f84dd64f37
   - Switch to RTK around takeoff time (crash):
     - https://logs.px4.io/plot_app?log=6ebd3714-df2a-4c1c-afc2-d0cc9de2f8e0
   - Exit RTK while in air in Position mode:
     - https://logs.px4.io/plot_app?log=d5e54d95-cc07-482e-b787-d9125cfdfde1
   - Exit RTK while in Offobard mode (2x crash):
     - https://logs.px4.io/plot_app?log=4437d131-e906-472b-8a15-5c49e47ec1c1
     - https://logs.px4.io/plot_app?log=e69bc86d-f36a-4baa-8796-4b5f2bf259d4

## Drone
- Hexacopter with PX4 v1.13.2 and Pixhawk 4
- Companion computer
- GPS
  - Primary GPS: CUAV ublox F9P
  - Secondary GPS: Holybro M8N

AlexKlimaj, 2023-05-07:
This looks like it was caused by the change in GPS altitude when the fix type changed. I think in 1.14/main all height sources are fused and converge to your primary height source. That may prevent this from happening.

[image]

junwoo091400, 2023-06-29:
> When this happens, the xyz correction of RTK is added (respectively removed) which leads to a replacement of the drone even in the air...

Btw what was the actual problem then? That RTK correction got applied again during flight & it lead to unstable behavior? It does seem like local position coordinates changed abruptly when correction was applied, but unsure exactly why that led to a crash 🤷

Katawann, 2023-06-29:
When the correction of RTK is added or removed, the drone shift back to the position without the RTK correction in xyz. Two of our crashes had a correction of 1-2 meters up in z with RTK and when it lost the RTK, the drone fall down of 2 meters and in our case the drone crash because we are flying at low altitude. 

The last crash was on takeoff, the z correction of RTK has been added while we were taking off so the drone failed to understand that it was not really 2 meters up and still on the ground...

We will test loss of RTK with v1.14 probably next week to define if the height fusion is enough in our situation

Katawann, 2023-07-06:
So we did multiple test with the latest beta of v1.14 and height fusion, but the issue of drone moving when there is a regain/loss of RTK is still there.

Log: https://logs.px4.io/plot_app?log=b75bd4bc-d494-4f91-8e5f-251ab5267e6e

We take off with RTK (stream via internet), hover and deactivate internet. After one minute (3:24), the ublox F9P RTK exits RTK mode. At this point, the drone descends 1.5-2 meters and veers 1 meter sideways before stabilizing in DGPS. When we reactivate the internet (and therefore the RTK stream), the drone returns to the position it had at RTK loss (4:09), i.e. 1.5-2 meters up and 1 meter to one side.

During the entire test we didn't touch the joysticks, and the AMSL altitude on QGroundControl was always indicated at 448.5m, even when the drone moved location.

What could be done to avoid such behavior ? Should we always "lock" the last RTK correction in xyz and keep it once we leave RTK mode to avoid an unwanted and dangerous when there is a loss or regained of RTK ?

AlexKlimaj, 2023-07-06:
RTK corrections are time sensitive. You can't keep re-sending the last RTCM data to the GPS module and expect it to work. If its altitude you are worried about, you could switch to using the baro as the primary height source. 

[image]
[image]

Katawann, 2023-07-06:
But with this height fusion new feature in v1.14, the firmware should take the Fused Altitude Estimation no ?

At first, we were working with baro as primary height altitude with our drone but we had some issue of drone slowly moving up and down sometimes probably because we are flying most the time at 2-3 meters above the ground.

Yes my idea is not to send last RTCM data to the GPS module, but that PX4 to keep the correction (if it know it) if there is a loss of RTK. I think it's a dangerous behavior that the drone is moving alone in xyz with this loss/regain of correction of RTK.

Katawann, 2023-07-12:
@slgrobotics, agreed. We tested multiple range sensors, but the issue in our use case is that flying at 2-3 meters above plants + spraying products make it not trivial to detect properly the ground and it brings more issue than with the current situation where the drone is "blind" (and it works fine like this since 3 years). We have ongoing project with detection of distance with a camera but until that RTK would be a good improvement (except when the above situation happen...)

alireza787b, 2025-03-20:
Hello... is there any update on this issue?

We’re using Cube Orange+ with Here 4 GNSS modules and F9P RTK and datalink over internet in our drone show application, where precise low-altitude positioning is critical. We’ve experienced the same abrupt altitude shifts when switching to/from RTK mode. Our observations align with the issue described here—sudden 2–5 meter jumps when RTK corrections are applied or lost.

We’re particularly interested in any updates on the improved height fusion approach in PX4 v1.14 or any alternative strategies (such as locking the RTK correction or switching to barometer-dominant altitude control) that can help mitigate these dangerous transitions. Has anyone found a robust workaround or seen further development on this issue?

closed 2026-01-21 by dakejahl