PX4/PX4-Autopilot #21574: Inconsistent GPS instance ID with CAN for Dual GPS

wsolichin-sees, 2023-05-10:
## Describe the bug
When using multiple ARK-CAN RTK GPS, the sensor_gps uORB instance starts randomly. This can cause an issue when subscribing to MAVLink message GPS_RAW_INT / GPS2_RAW, or subscribing to sensor_gps instance.

## To Reproduce
Steps to reproduce the behavior:
1. Connect the ARK-CAN RTK GPS as shown below:
   [image]
2. Start QGroundControl and run dmesg. On first boot it will show as follows: 
   [image]
   As shown above, CAN node ID of 125 starts first, therefore sensor_gps with node ID 125 is set to uORB instance 0 (sensor_gps_0), and sensor_gps with node ID 124 is set to uORB instance 1 (sensor_gps_1).
3. Reboot the flight controller, and check dmesg output, as the node ID starts randomly, occasionally dmesg will show the following output as shown below:
   [image]
   As shown above, this time CAN node ID 124 starts first, therefore sensor_gps with node ID 124 is set to uORB instance 0 (sensor_gps_0), and vice-versa.

## Expected behavior
The expected behavior is when configuring the CAN component following the guide here: https://docs.px4.io/main/en/dronecan/ark_rtk_gps.html#ark-rtk-gps), is for the Rover to be the primary GPS instance (sensor_gps_0), and the Moving Base to be the secondary GPS instance (sensor_gps_1).

## Log Files and Screenshots
No log files needed to reproduce the issue.

## Drone (please complete the following information):
- Tested on the following flight controllers:
   - Pixhawk 4 Flight Controller
   - CubePilot Cube Orange
- Sensors:
   - 2x ARK-RTK CAN GPS

## Additional context
None.

AlexKlimaj, 2023-05-12:
Some potential thoughts on how to solve this are delaying publishing the gps message over CAN on the node either based on a parameter or the UBX_MODE param when it is set to a moving base. 

Maybe @dagar has a better idea.

wsolichin-sees, 2023-05-16:
I did try to delay publishing the GPS message over CAN with usleep(100 * 1000) before the GNSS Fix2 is published, but once in a while, on a vehicle reboot, the UAVCAN subscriber seems to initialized before the publisher gets initialized (as shown in the image below).

[image]