PX4/PX4-Autopilot #21577: The configuration information of flight control (like sensors, remote control, flight mode) cannot be saved.

jieren30, 2023-05-11:
hardware: holybro kakute H7mini
firmware version : PX4-Autopilot/1.14.0-beta2
bug:
When I received the flight control unit, I first used the daily build QGC to perform the initial configuration of the FCU. But the initial configuration of the FCU cannot be saved after power down. Every time I power up again, I need to configure the flight control again.
[image]

dagar, 2023-05-20:
@julianoes do you know anything about this board?

julianoes, 2023-05-20:
Yes, @jieren30 try to wipe everything using the steps here:
https://docs.px4.io/main/en/advanced_config/bootloader_update_from_betaflight.html#dfu-util