PX4/PX4-Autopilot #21586: Why use .sdf instead of .xacro files?

JohnTGZ, 2023-05-12:
Hi, I'm just curious why the earlier versions were using `.xacro` files for simulation models but have since switched to `.sdf` files. I would have thought that `.xacro` files would provide increased functionality. An instance of such functionality is being able to use ROS nodes such as `robot_state_publisher` to publish static transforms for sensor frames (depth camera, sonar, lidar, etc.) for instance, reducing the need to define static transforms externally in launch files.

Jaeyoung-Lim, 2023-05-12:
@JohnTGZ Two reasons
- xacro files generate urdf files, which have limited features compared to sdf.
- It makes the models less opaque making it easier to understand and configure compared to using xacro as additional tooling to generate a file that is used in simulation

JohnTGZ, 2023-05-13:
I see, that makes complete sense, thanks Jaeyoung for the detailed explanation.
I also observe the model description is also encapsulated within the .sdf files, making it more transparent as you said. 

Another question I have is about being able to publish the kinematic chains from SDFs as transformations under the /tf topic. I understand that with a '.xacro' robot description, you can use 'robot_state_publisher' to take in the 'robot_description' param (which loads the xacro file). Is there a similar functionality with SDFs? That would prevent one from having to publish their own static transforms independently (e.g. using the static tf transform node)

Jaeyoung-Lim, 2023-05-13:
@JohnTGZ I think we need to take a step back. We are mixing concepts such as xacro with sdf and urdf files.

xacro is a macro which generates a urdf file. Therefore xacro cannot be compared feature wise to a sdf file. 

I am not sure about the robot state publisher implementation, but if it is implemented specifically for an xacro file, this is specific for the implementation. It is not related to the fundamental limitation/feature of the file format itself

JohnTGZ, 2023-05-13:
Apologies for mixing up xacro with sdf and urdf files. What I meant to compare was `URDF` vs `SDF`. 
To further explain my question, I was wondering if it is recommended practice for robot joint transformations to be published using nodes like [static_transform_publisher](http://wiki.ros.org/static_transform_publisher) when using `.sdf` files. This is because `.urdf` files can already have all it's joints published by feeding it as a `robot_description` to the [robot_state_publisher](http://wiki.ros.org/robot_state_publisher) node, without a need for multiple `static_transform_publishers` to be instantiated just so we can have the frame transformations for our sensor units.

Jaeyoung-Lim, 2023-05-13:
> What I meant to compare was URDF vs SDF.
To further explain my question, I was wondering if it is recommended practice for robot joint transformations to be published using nodes like

Unfortunately, I dont have a good answer for you. I can imagine this would highly depend on what you are trying to do

JohnTGZ, 2023-05-16:
I see, thank you for explaining, I guess either extending the functionality of robot state publisher or having a newly written node to parse SDFs for joint states would be a better long term solution. I'm happy to close this issue.

closed 2023-05-16 by Jaeyoung-Lim