PX4/PX4-Autopilot #21588: Cannot build  SITL with ignition

fbenti, 2023-05-14:
## Describe the bug
I am trying to set up the sitl with gazebo.

## To Reproduce
Ubuntu 20.04
Ros2 foxy
Ignition Fortress

If I directly run `make px4_sitl ignition` I get the following error:
`ninja: error: unknown target 'ignition'`

I then tried to follow [https://github.com/Auterion/px4-simulation-ignition](https://github.com/Auterion/px4-simulation-ignition) but the last step (Creation and install of debian package on Linux) is returning the following error when I compile it with cmake.
```
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.58") found components: system thread filesystem 
-- Looking for ignition-plugin1 -- found version 1.4.0
-- Searching for dependencies of ignition-plugin1
-- Searching for <ignition-plugin1> component [register]
-- Looking for ignition-plugin1-register -- found version 1.4.0
-- Searching for dependencies of ignition-plugin1-register
CMake Warning at CMakeLists.txt:12 (find_package):
  By not providing "Findignition-gazebo5.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ignition-gazebo5", but CMake did not find one.

  Could not find a package configuration file provided by "ignition-gazebo5"
  with any of the following names:

    ignition-gazebo5Config.cmake
    ignition-gazebo5-config.cmake

  Add the installation prefix of "ignition-gazebo5" to CMAKE_PREFIX_PATH or
  set "ignition-gazebo5_DIR" to a directory containing one of the above
  files.  If "ignition-gazebo5" provides a separate development package or
  SDK, be sure it has been installed.

-- Looking for ignition-msgs8 -- found version 8.7.0
-- Searching for dependencies of ignition-msgs8
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found suitable version "3.6.1", minimum required is "3") 
-- Looking for ignition-math6 -- found version 6.14.0
-- Searching for dependencies of ignition-math6
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 6.2.0
-- Looking for ignition-gazebo6 -- found version 6.14.0
-- Searching for dependencies of ignition-gazebo6
-- Looking for sdformat12 -- found version 12.7.1
-- Searching for dependencies of sdformat12
-- Looking for ignition-math6 -- found version 6.14.0
-- Looking for ignition-utils1 -- found version 1.5.1
-- Searching for dependencies of ignition-utils1
-- Looking for ignition-plugin1 -- found version 1.4.0
-- Searching for dependencies of ignition-plugin1
-- Searching for <ignition-plugin1> component [loader]
-- Looking for ignition-plugin1-loader -- found version 1.4.0
-- Searching for dependencies of ignition-plugin1-loader
-- Searching for <ignition-plugin1> component [register]
-- Looking for ignition-transport11 -- found version 11.4.0
[... 175 lines cut ...]
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.6.1") 
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
_MAVLINK_INCLUDE_DIR
   used as include directory in directory /home/aut/px4-simulation-ignition
   used as include directory in directory /home/aut/px4-simulation-ignition
   used as include directory in directory /home/aut/px4-simulation-ignition
   used as include directory in directory /home/aut/px4-simulation-ignition
   used as include directory in directory /home/aut/px4-simulation-ignition
   used as include directory in directory /home/aut/px4-simulation-ignition
   used as include directory in directory /home/aut/px4-simulation-ignition
   used as include directory in directory /home/aut/px4-simulation-ignition
   used as include directory in directory /home/aut/px4-simulation-ignition
   used as include directory in directory /home/aut/px4-simulation-ignition
   used as include directory in directory /home/aut/px4-simulation-ignition
   used as include directory in directory /home/aut/px4-simulation-ignition
   used as include directory in directory /home/aut/px4-simulation-ignition

-- Configuring incomplete, errors occurred!
See also "/home/aut/px4-simulation-ignition/build/CMakeFiles/CMakeOutput.log".
See also "/home/aut/px4-simulation-ignition/build/CMakeFiles/CMakeError.log".
```
which looks-like it cannot find the MAVLINK package, which I have installed with `sudo apt-get install ros-foxy-mavros ros-foxy-mavros-extras`

I cannot find a complete and exhaustive documentation for it, can anybody help?

Jaeyoung-Lim, 2023-05-14:
@fbenti Please use the updated instructions. `ignition` has been renamed as `gz`

closed 2023-05-14 by Jaeyoung-Lim

fbenti, 2023-05-14:
Ok thanks. 

However I have this world model, [https://github.com/osrf/mbzirc/blob/main/mbzirc_ign/worlds/simple_demo.sdf](https://github.com/osrf/mbzirc/blob/main/mbzirc_ign/worlds/simple_demo.sdf) where all the plugins come from ignition. How can I export them in gazebo?