PX4/PX4-Autopilot #21593: Position estimation drifting when drone is stationary

Cavalletta98, 2023-05-15:
Hi, i'm facing an issue about position estimation when the drone is stationary on the ground. In particular i noticed that the red arrow on QGControl starts to drift from the original position. This cause that the estimate position is no more the real one and i can't perform a mission. I attached two plots of the local position estimation on X and Y axes. Can someone could help me? Which paramaters of the EKF2 should i have to ajust?
[image]
[image]

bresch, 2023-05-17:
Are you talking about the setpoint change in the beginning of the flight or the position tracking?
In both cases, I see deviations of max 40cm, which seems reasonable depending on the size of your drone. It can be improved but it really depends on the quality of the position measurements (GNSS I assume?) and the velocity/position controllers tuning.

Cavalletta98, 2023-05-18:
I'm talking about before flight starts. I put the drone on the ground, i wait for GPS fix and the i want to try to send a mission via QGcontrol. What i notice is that the red arrow,representing the drone, starts to move away from the position fixed. This cause that i cannot send the mission. I don't know if you understand the problem

bresch, 2023-05-19:
Ok, but your plot only shows a flight, not "before the flight starts".

Before the flight, it could well be that the GNSS solution is still converging and that's why you see the arrow moving around. Maybe the signal quality in your area is quite bad or the reception is jammed by some active components on your drone.

Cavalletta98, 2023-05-19:
The light on GPS is green when i see this problem. I also wait for 5 minutes before start to flight. I will never stop.
I attacched also the plot for GPS jamming
[image]