PX4/PX4-Autopilot #21596: uxrce_dds_client exposed ros2 message configurable as parameters

TianmingLi, 2023-05-16:
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or Discord (you can find an invite link on this project README).

## CPU and RAM usage very high after start uxrce_dds_client  
The higher cpu and ram usage maybe caused by UART high baud rate(921600) and so many ros message to transfer.

## Describe your preferred solution
Make exposed message in [dds_topics.yaml](https://github.com/PX4/PX4-Autopilot/blob/main/src/modules/uxrce_dds_client/dds_topics.yaml) configurable as parameters, so you can disable or enable it as you want.

## Describe possible alternatives
Reduce UART baud rate

## Additional context
[flight log](https://review.px4.io/plot_app?log=db474729-0699-4836-b376-f81aa702cbba)

dagar, 2023-05-20:
I'm just curious how did lowering the baud rate change the cpu usage?

As a start I was thinking we could add groups of topics (kind of like mavlink modes).

TianmingLi, 2023-05-20:
> I'm just curious how did lowering the baud rate change the cpu usage?
> 

My bad, I just forget UART use DMA to transmit data rather tha using CPU to do it.