PX4/PX4-Autopilot #21600: NTrip caster data is not working with GPS device

paujar, 2023-05-17:
## Describe the bug
My setup:
PX4 1.14.0 beta 2
MavSDK 1.4.5
ARK GPS RTK CAN version firmware version ; [here](https://github.com/PX4/PX4-Autopilot/pull/21595/files)
I am getting RTCM stream data from national ntrip caster endpoint and when I forward that data chunk directly towards F9P module, the device gets RTK FIX. But when I send that data to PX4 via UART from my Companion Computer that uses MavSdk Python version 1.4.5  with mavsdk rtk plugin method send_rtcm_data(data) to send the data to GPS. I have tried every possible approach to send the data, but I tested it with just sending the raw data (bytes) to the F9P module and that worked so I would assume that something goes wrong when using send_rtcm_data at some point.
It injects nicely towards the GPS, but it doesn't get used and there is no RTK FIX. 

## To Reproduce
Steps to reproduce the behavior:
1. Enable configuration of UAVCAN_SUB_RTCM (from FC settings)
2. Enable configuration of CANNODE_SUB_RTCM (from CAN device settings)
3. Get mavsdk drone connection to open the conversation between CC and FC
4. Get the data from ntrip client of your choise (stream)
5. Send the data stream towards MavSdk client with send_rtcm_data(str(bytes))
6. Look if your GPS changes to RTK FIX
7. See rtcm_msg_used from ARK RTK GPS

## Expected behavior
The GPS status is at maximum 3D DGPS Lock, but no higher. It should work with 2 GPS units; moving/fixed baseline and receiver, but this setup doesn't work. It's so close, but it doesn't. 

## Log Files and Screenshots
Here you can see the RTCM3 messages that are being sent to the PX4 via send_rtcm_data(
[image]

Here you can see the rtcm data being injected to the ARK RTK GPS / F9P:
[image]
and in the bottom of the image you can see the rtcm_crc_failed = false and rtcm_msg_used=0 <-- this should be "2"
And the data is validated so many times that it's not the issue. The issue is somewhere in between.

Add screenshots to help explain your problem.

## Drone (please complete the following information):
- This is not relevant
- This is not relevant

## Additional context
Below is zip package that you can try, you just need some ntrip provider for test_ssr.py to be set up and of course you need to have the environment.
[TestPythons.zip](https://github.com/PX4/PX4-Autopilot/files/11497241/TestPythons.zip)

paujar, 2023-05-19:
Forgot to mention that rtcm3 in the code is MAVProxy RTCM3.py file located in the location folder as the NTripClient as well ([https://github.com/ArduPilot/MAVProxy/blob/master/MAVProxy/modules/lib/rtcm3.py](url)

paujar, 2023-05-25:
I double checked - the RTCM3 library streaming that is on above doesn't work actually with ARK RTK with bypass mode. So I double checked that if I just receive the RTCM data (I didn't now use the RTCM3.py helper class) and pass that to the debug port in bypass mode;

`                           data=self.socket.recv(180)
                            if(not data):
                                print('not data')
                                break
                            else:
                                    print("sending..")
                                    self.serial.write(data)`

So that data is passed directly to the DEBUG port when ARK RTK is in bypass mode, it works, instantly goes to FLOAT and after a while to FIX RTK (I check it with GPGGA msg and with Wine). 

So now this simple model works directly from computer to debug port (bypassmode). Then I added the same stuff to Companion Computer and added the MAVSdk and PX4 to the game (versions are mentioned on the original post):

it got's the RTCM data stream pub/sub and **rtcm_inject rate** is 5-8 and **rtcm_crc_failed: False** and **rtcm_msg_used:0** in the ARK RTK GPS, so it doesn't work.

So I bet there is something wrong in the mavSDK, PX4 or ARK RTK GPS, because if it would just pass the RTCM data as is it would work. Is it possible that the problem is in MavSDK, because the ARK RTK GPS works when it talks directly via PX4 and uses same PUB/SUB methods. Could it be the encoding? Why it doesn't 

just accept bytes to send_rtcm_data and pass that @julianoes @AlexKlimaj or something else. I bet there is a small thing somewhere wrong, I also try to check but I don't know the dependencies of the mavsdk.

await drone.rtk.send_rtcm_data(RtcmData(str(data)))

Here is the updated ntripclient that you need to call from with the python test_ssr.py (in original post) and put this file to /location -folder
[NtripClient.zip](https://github.com/PX4/PX4-Autopilot/files/11561427/NtripClient.zip)

paujar, 2023-05-30:
> So far I'm not seeing any issue on the PX4 side.
>
> @julianoes <https://github.com/julianoes> did you have this working in
> some form?

Could there be some data sending issues in rtk plugin (mavsdk python) for
sending the rtcmdata?

Because now I noticed that if I boot the fc and gps,it sometimes it gets
some messages used in ublox (if to believe gps flags) and I thought that
could it be some data that gets sent too fast to mavsdk and the data goes
now and then corrupted. Though it uses async,but I dont know how it works
on deep level.

julianoes, 2023-05-30:
@dagar no I haven't tried this through MAVSDK-Python. The type conversion there might not be right.
I need to give it a go at some point.

julianoes, 2023-06-06:
Ok this is now [fixed](https://github.com/mavlink/MAVSDK/pull/2073) with mavsdk_server version [v1.4.16](https://github.com/mavlink/MAVSDK/releases/tag/v1.4.16) and MAVSDK-Python [v1.4.8](https://github.com/mavlink/MAVSDK-Python/releases/tag/1.4.8).

closed 2023-06-06 by julianoes