PX4/PX4-Autopilot #21604: ninja: error: unknown target 'jmavsim'

awkar, 2023-05-17:
When I make PX4 source code in Ubuntu 20.04, I get this error.
I run "make px4_sitl jmavsim" according to [this url](https://docs.px4.io/main/en/dev_setup/building_px4.html)
The source code is the newest.
Please tell me how to sovle this error, thanks.

djlisko01, 2023-05-18:
Hello, I have the same issue using Ubuntu 20.04.

dagar, 2023-05-20:
Do you have `java` and `ant` installed?

https://github.com/PX4/PX4-Autopilot/blob/bb0f2875a94960a4c7a937e9786946b10415e252/src/modules/simulation/simulator_mavlink/sitl_targets_jmavsim.cmake#L5

From a clean state please share the configure output (eg `make distclean && make px4_sitl jmavsim`).

awkar, 2023-05-22:
> Do you have `java` and `ant` installed?
> 
> https://github.com/PX4/PX4-Autopilot/blob/bb0f2875a94960a4c7a937e9786946b10415e252/src/modules/simulation/simulator_mavlink/sitl_targets_jmavsim.cmake#L5
> 
> From a clean state please share the configure output (eg `make distclean && make px4_sitl jmavsim`).

Thank you very much. I install `ant`  and `java` according to Line 202-220 in `PX4-Autopilot/Tools/setup/ubuntu.sh`, and it works.
But I have a question. I have installed the development environment by `bash ./PX4-Autopilot/Tools/setup/ubuntu.sh`, and I didn't find any waring or error. However, the `ant`  and `java` haven't been installed, and I fix the make error (ninja: error: unknown target 'jmavsim') by run these comands ( Line 202-220 in `PX4-Autopilot/Tools/setup/ubuntu.sh`) by myself. Is there any step I did wrong?

awkar, 2023-05-22:
> I install `ant` and `java` according to Line 202-220 in `PX4-Autopilot/Tools/setup/ubuntu.sh`, and it works.

 I install `ant` and `java` according to Line 202-220 in `PX4-Autopilot/Tools/setup/ubuntu.sh`, and it works. You can try it.

closed 2023-05-30 by julianoes