PX4/PX4-Autopilot #21607: Gazebo Simulation crash in VMware virtual machine

TianmingLi, 2023-05-18:
## Describe the bug
Gazebo Simulation crash in VMware virtual machine.
Ubuntu 2204
The last main branch px4

## To Reproduce
Steps to reproduce the behavior:
1. Install brand new ubuntu 22.04 on VMware virtual machine
2. Install development environment through `bash ./PX4-Autopilot/Tools/setup/ubuntu.sh`
3. Build and run simulation `make px4_sitl gz_x500`
4. See error

## Expected behavior
Gazebo sim GUI show up.

## Log Files and Screenshots

```
[GUI] [Err] [Ogre2RenderEngine.cc:1129]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at ./RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3434)
[GUI] [Err] [Ogre2RenderEngine.cc:1129]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at ./RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3434)
[GUI] [Err] [Ogre2RenderEngine.cc:1129]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at ./RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3434)
[GUI] [Err] [Ogre2RenderEngine.cc:1129]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at ./RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3434)
[GUI] [Err] [Ogre2RenderEngine.cc:1129]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at ./RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3434)
[GUI] [Err] [Ogre2RenderEngine.cc:1129]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at ./RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3434)
[GUI] [Err] [Ogre2RenderEngine.cc:1129]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at ./RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3434)
[GUI] [Err] [Ogre2RenderEngine.cc:1129]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at ./RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3434)
[GUI] [Err] [Ogre2RenderEngine.cc:1129]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at ./RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3434)
[GUI] [Err] [Ogre2RenderEngine.cc:1129]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at ./RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3434)
[GUI] [Err] [Ogre2RenderEngine.cc:1137] Unable to create the rendering window after [11] attempts.
[GUI] [Err] [Ogre2RenderEngine.cc:1046] Failed to create dummy render window.
[GUI] [Err] [Ogre2RenderEngine.cc:1047] Please see the troubleshooting page for possible fixes: https://gazebosim.org/docs/fortress/troubleshooting
```

Add screenshots to help explain your problem.

## Drone (please complete the following information):

## Additional context

After some searching, it can be solved by forcing specified render engine form `ogre2` to `ogre` in start script [px4-rc.simulator](https://github.com/PX4/PX4-Autopilot/blob/599a66c8a5f33fd9e139ab950e716d0e36a3e809/ROMFS/px4fmu_common/init.d-posix/px4-rc.simulator#LL73C6-L73C6)

After change 
``` code
		if [ -z "${HEADLESS}" ]; then
			# HEADLESS not set, starting gui
			${gz_command} ${gz_sub_command} -g &
		fi

```

to 

```
		if [ -z "${HEADLESS}" ]; then
			# HEADLESS not set, starting gui
			${gz_command} ${gz_sub_command} -g --render-engine ogre &
		fi

```

The error gone and gazebo gui pop up and whole simulation works.

dagar, 2023-05-20:
At the moment I don't have a preference between ogre vs ogre2 as long as it works. Would you like to open a pull request making this change?

dagar, 2023-05-20:
@MaEtUgR can you give this a try with WSL2?

TianmingLi, 2023-05-20:
> At the moment I don't have a preference between ogre vs ogre2 as long as it works. Would you like to open a pull request making this change?

[Request](https://github.com/PX4/PX4-Autopilot/pull/21617) has been submitted.

closed 2023-06-04 by TianmingLi

MaEtUgR, 2023-06-05:
In WSL(2) I also get an error but a different one:
```
INFO  [px4] Startup script returned successfully
pxh> libEGL warning: failed to open /dev/dri/renderD128: Permission denied

libEGL warning: failed to open /dev/dri/renderD128: Permission denied

terminate called after throwing an instance of 'Ogre::UnimplementedException'
  what():  OGRE EXCEPTION(9:UnimplementedException):  in GL3PlusTextureGpu::copyTo at ./RenderSystems/GL3Plus/src/OgreGL3PlusTextureGpu.cpp (line 685)
Stack trace (most recent call last):
#31   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fc62933b75a, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#30   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fc6293960b7, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#29   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fc626ed73e2, in g_main_context_iteration
#28   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fc626f2e6c7, in
#27   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fc626ed9d3a, in g_main_context_dispatch
#26   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fc629396a66, in
#25   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fc62933ff26, in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
#24   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fc62933ce39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#23   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc628a94712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#22   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fc62936a41d, in QObject::event(QEvent*)
#21   Object "/usr/lib/x86_64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0x7fc60402a513, in gz::gui::plugins::RenderWindowItem::Ready()
#20   Object "/usr/lib/x86_64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0x7fc60402a120, in gz::gui::plugins::RenderThread::Initialize[abi:cxx11]()
#19   Object "/usr/lib/x86_64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0x7fc604036c7d, in gz::gui::plugins::RenderThreadRhiOpenGL::Initialize[abi:cxx11]()
#18   Object "/usr/lib/x86_64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0x7fc60402dc87, in gz::gui::plugins::GzRenderer::Initialize[abi:cxx11]()
#17   Object "/lib/x86_64-linux-gnu/libgz-rendering7.so.7", at 0x7fc5f42b39e8, in gz::rendering::v7::BaseRenderEngine::CreateScene(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#16   Object "/lib/x86_64-linux-gnu/libgz-rendering7.so.7", at 0x7fc5f42bafee, in gz::rendering::v7::BaseRenderEngine::CreateScene(unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#15   Object "/lib/x86_64-linux-gnu/libgz-rendering7.so.7", at 0x7fc5f42ce9ea, in gz::rendering::v7::BaseScene::CreateMaterials()
#14   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre2.so", at 0x7fc59e67060a, in gz::rendering::v7::Ogre2Material::SetTextureMapImpl(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Ogre::PbsTextureTypes)
#13   Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7fc59e2db224, in Ogre::TextureGpuManager::_waitFor(Ogre::TextureGpu*, bool)
#12   Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7fc59e2dae4a, in Ogre::TextureGpuManager::_update(bool)
#11   Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7fc59e1a855c, in Ogre::ObjCmdBuffer::execute()
#10   Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7fc59e1a68cd, in Ogre::ObjCmdBuffer::NotifyDataIsReady::execute()
#9    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7fc59e2cad9b, in Ogre::TextureFilter::GenerateHwMipmaps::_executeSerial(Ogre::TextureGpu*)
#8    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/OGRE/RenderSystem_GL3Plus.so", at 0x7fc59dc48ac3, in
#7    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/OGRE/RenderSystem_GL3Plus.so", at 0x7fc59dc453da, in
#6    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fc62ad98517, in __cxa_throw
#5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fc62ad982b6, in std::terminate()
#4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fc62ad9824b, in
#3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fc62ad8cbbd, in
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fc62e7947f2, in abort
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fc62e7ae475, in raise
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fc62e802a7c, in pthread_kill
Aborted (Signal sent by tkill() 9400 1000)
INFO  [commander] Ready for takeoff!
```
I had to reinstall the new gazebo for this test since I'm happily using gazebo classic. Also it's a bit strange that if the wrong gazebo is installed you get a mistakable error saying the simulation target doesn't exist. Next I'll try the suggestion in https://github.com/PX4/PX4-Autopilot/pull/21617