PX4/PX4-Autopilot #21610: Pose Covariance

fbenti, 2023-05-19:
Hello, can anyone tell me where I can find the pose covariance of the drone? Is the EstimatorStates message what I'm looking for? [https://github.com/PX4/PX4-Autopilot/blob/main/msg/EstimatorStatus.msg](https://github.com/PX4/PX4-Autopilot/blob/main/msg/EstimatorStatus.msg)

bresch, 2023-05-19:
Hi, what your're looking for is defined in https://github.com/PX4/PX4-Autopilot/blob/599a66c8a5f33fd9e139ab950e716d0e36a3e809/msg/EstimatorStates.msg#L7

See https://docs.px4.io/main/en/advanced_config/tuning_the_ecl_ekf.html#state-variances for the numbering. I just saw that there is a mistake in the document, it should be:
> Refer to covariances[24] in [EstimatorStates](https://github.com/PX4/PX4-Autopilot/blob/main/msg/EstimatorStates.msg). The index map for covariances[24] is as follows: ...

I'll update the documentation.

Note tha the quaternion variance in our estimator is defined as a variance on each quaternion element (4d) and not as a perturbation on the tangent space (3d) as needed by http://docs.ros.org/en/lunar/api/geometry_msgs/html/msg/PoseWithCovariance.html so you would need to convert if first.

fbenti, 2023-05-19:
Thanks for the answer. I guess, the covariances are defined in NED coordinate, so to use them in ROS, for the orientation for instance,  I should first convert the quaternions to eulers, then apply [https://github.com/PX4/px4_ros_com/blob/0bcf68bcb635199adcd134e8932932054e863c0d/src/lib/frame_transforms.cpp#L200](https://github.com/PX4/px4_ros_com/blob/0bcf68bcb635199adcd134e8932932054e863c0d/src/lib/frame_transforms.cpp#L200) with `StaticTF::NED_TO_ENU`, right?

bresch, 2023-05-19:
> I should first convert the quaternions to eulers

No, the tangent space of quaternions is not the same as euler angles. But yes, you need to find a way to unwrap the quaternion covariance to its tangent space (and that's how you'll reduce its dimension correctly) and then transform it to ENU.
I guess some other ppl already faced the same issue so you might be able to find something online.

fbenti, 2023-05-19:
I'm gonna have a look at it, thanks. 

What about if I simply want the orientation covariance defined in euler angles? In this case, can I just use the conversion from quaternions to euler, and then from NED to ENU? 

Otherwise, this is how I would implement what you said above. If you are already familiar, could you give it a check?
```
Eigen::Vector3d quaternionToTangentSpace(const Eigen::Quaterniond& quaternion)
{
    Eigen::Vector3d tangent_vector;

    double angle = 2.0 * std::acos(quaternion.w()); // Compute the angle of rotation
    double axis_norm = quaternion.vec().norm(); // Compute the norm of the axis of rotation

    // Map the quaternion to the tangent space
    tangent_vector = (angle * quaternion.vec()) / axis_norm;

    return tangent_vector;
}

// Trasform pose covariance from px4 to ros
void transform(const std::shared_ptr<interfaces::srv::TransformPoseCovariance::Request> request,
          std::shared_ptr<interfaces::srv::TransformPoseCovariance::Response> response) {
    
    // Get covariance in px4
    Eigen::Vector3d cov_pos (request->cov.x, request->cov.y, request->cov.z);
    Eigen::Quaterniond cov_q (request->cov.qw, request->cov.qx, request->cov.qy, request->cov.qz);
    
    // Covert quaternion to tangent space
    Eigen::Vector3d cov_qt = quaternionToTangentSpace(cov_qt);
    
    // Initialize covariance matrix (in px4) 
    px4_ros_transform::frame_transforms::Covariance6d cov_px4;
    cov_px4[0] = cov_pos[0];
    cov_px4[7] = cov_pos[1];
    cov_px4[14] = cov_pos[2];
    cov_px4[21] = cov_qt[0];
    cov_px4[28] = cov_qt[1];
    cov_px4[35] = cov_qt[2];

    // Transform in ROS.
    px4_ros_transform::frame_transforms::Covariance6d cov_ros = px4_ros_transform::frame_transforms::transform_static_frame(cov_px4, px4_ros_transform::frame_transforms::StaticTF::NED_TO_ENU);

    
    RCLCPP_INFO(rclcpp::get_logger("rclcpp"), "TrasformTrajectorySetpoint server: request RECEIVED.");

    response->cov.x = cov_px4[0];
    response->cov.y = cov_px4[7];
    response->cov.z = ccov_px4[14];
    response->cov.rx = cov_px4[21];
    response->cov.ry = cov_px4[28];
    response->cov.rz = cov_px4[35];

}
```

bresch, 2023-05-31:
As you found in Discord, PX4 actually already does the conversion I mentioned previously. You can then directly use the result in https://github.com/PX4/PX4-Autopilot/blob/db18a9438277ace40112e139075e9ecc6ee1a87a/msg/VehicleOdometry.msg#L24

closed 2023-05-31 by bresch