PX4/PX4-Autopilot #21612: PX4 continous reboot and software crash in report

Chris-Bee, 2023-05-19:
## Describe the bug
At some after a number of flights, shortly after landing with a quadrotor setup, the PX4 seems to be stuck in a self reboot loop according to the beep patterns from the autopilot. This error occured twice in the last three weeks. The first time this error occured, the px4 settings where erased and needed to be fully setup again.

## To Reproduce
There is no deterministic way to reproduce the error.

## Expected behavior
Clean IDLE state after flight

## Log Files and Screenshots
Link to the log file https://review.px4.io/plot_app?log=c6774fdb-3c3d-4ee4-b522-f23423dd3e24

## Drone (please complete the following information):
Self-build 4 rotor drone

Let me know if specific additional data can be produced to help.

Thanks in advance,
Chris

dagar, 2023-05-20:
In your log it says your on `FW git-hash: 75801cd3bbcd5dd86e26f4a69b381f5b583ebdc5`, but I can't find that in the repository. Do you have any custom code?

Chris-Bee, 2023-05-20:
Hello @dagar, appologies for not including that info, yes the code has a modification:
- Repo corresponding to the git hash: https://github.com/aau-cns/PX4-Autopilot
- It's the same code as for this Pull-Request: https://github.com/PX4/PX4-Autopilot/pull/20163

We think that the error possibly leads to a corruption of the FW. After this error, the motors turn on much slower and one after the other. If we reflash the FW, then tis behavior is gone untill the next time the issue, described above, occures.