PX4/PX4-Autopilot #21615: PX4 fmu V5 cannot handle analog differential pressure sensors (i.e. MPXV7002)

gianni-carbone, 2023-05-19:
PX4 fmu V5 cannot handle analog differential pressure sensor (i.e. MPXV7002) even if it has an 6.6V ADC exposed.

A simple way of solution is to refactor code renaming ADC1_SPARE_1_CHANNEL macro to ADC_AIRSPEED_VOLTAGE_CHANNEL in board_config.h and the references to it in the sensors.cpp 
This code change also fixes the missing SENS_DPRES_ANSC parameter in QGC.

I don't know why V5 has not a standard handling of this type of sensor (like V2 has).  If the problem does not derive from a strategic or standard choice, I can post a commit that solves the problem on fmu V5.

dagar, 2023-05-20:
I actually haven't seen anyone using analog differential pressure sensors for years, but if there's still interest I'd be happy to help get a solution in place.

gianni-carbone, 2023-05-21:
Dagar, my need arises from the fact that I had a px4_fmu-v2 with an analog differential pressure sensor and now I switched to v5 (cuav 5 nano). Investigating the code I saw that, in the v2 boardconfig, the reference to ADC 6.6V is with the macro ADC_AIRSPEED_VOLTAGE_CHANNEL. The same macro is used in some boardconfigs, not in others. For example in V5 the ADC1_SPARE_1_CHANNEL macro is used instead. The result is that conditionally compiling sensor.c removes the analog differential sensor handling code (which is already hard-coded!). So just rename the ADC1_SPARE_1_CHANNEL macro to ADC_AIRSPEED_VOLTAGE_CHANNEL in the relevant boarconfig file (and all its references in the same file) and the rest goes by itself. Furthermore, it will also be true that a digital sensor is immune to disturbances, but it is also true that there are analog sensors around at a lower price. For example, my MPXV7002 can be found for 25 euros at national distributors.
If you want I can do a pull request, like I've done in the past for other modules, which solves the problem for V5.

tubeme, 2023-06-29:
Better get CAN airspeed sensor. There are a lot of sensors available. Analog is not good!