PX4/PX4-Autopilot #21618: Camera plugin namespace in gazebo

fbenti, 2023-05-20:
Ros Humble
Gazebo Garden

I'd would like to spawn multi-drones, each equipped with a camera (x500_depth). The problem is that on gazebo side the camera topics `/camera` and `/camera_info` are not remapped with some prefix, which is what I need since I'd like, then, to bridge them into ros2. How can I change that?

rg-28, 2024-05-29:
Hi, were you able to find the solution to this?

halak0013, 2024-05-30:
Hi, we are using gazebo garden and customized fixed plane
We add 3 camera for top, back and front. And we install `ros-humble-ros-gzgarden` for ros2 and gazebo communication.
when we start launch file with this bridge
```python
    bridge_back = Node(
        package="ros_gz_bridge",
        executable="parameter_bridge",
        arguments=["/camera_back@sensor_msgs/msg/Image@ignition.msgs.Image"],
    )

  ld.add_action(camera_back)
```
it give camera image form rqt image