PX4/PX4-Autopilot #21621: Trajectory Setpoint - yawspeed

fbenti, 2023-05-22:
I'm sending TrajectorySetpoint to the drone in order to rotate around its z axis, with low yaw speed, I set yawspeed to 0.1. However, it seems like the yawspeed parameter is not considered.  

[image]
[06_57_44.zip](https://github.com/PX4/PX4-Autopilot/files/11528079/06_57_44.zip)

I'm sending TrajectorySetpoint and OffboarddControlMode with only position set to True (but I also tried with attitude true, but nothing changed.

bresch, 2023-05-22:
Do you also set an attitude or just a yawrate?
Could you upload the log to https://logs.px4.io/ and share the link here please

fbenti, 2023-05-22:
I've also set the attitude. 
[https://logs.px4.io/plot_app?log=9e036756-6219-4e61-8f88-1aa75e3c9383](https://logs.px4.io/plot_app?log=9e036756-6219-4e61-8f88-1aa75e3c9383)

bresch, 2023-05-22:
The yawspeed is a feedforward, so you need to send a yaw setpoint that is the integral of that 0.1 rad/s if you want a ramp with constant speed.

fbenti, 2023-05-22:
Ok, I cannot just set the desired yaw only once, but I have to change it at every sampling step according to the sampling time, right?

I mean, if I start at `yaw=0` and set the `yaw_setpoint =1` (rad) and `yawspeed = 0.1` (rad/s), it wouldn't reach the target in 10 seconds ?

fbenti, 2023-05-22:
In the first log attached, in the TrajectorySetpoint msg, I was giving `yaw = desired_yaw` and `yawspeed = yawspeed`. 

Now, instead [https://review.px4.io/plot_app?log=3e32b266-4225-47ba-8ba9-cbd9dfb27adb](https://review.px4.io/plot_app?log=3e32b266-4225-47ba-8ba9-cbd9dfb27adb)?  So now in the trajectory msg, at every iteration I'm giving a yaw setpoint as `yaw = curr_yaw + yawspeed` and set `yawspeed = yawspeed`.  Is this the right way of doing it? If so, what the purpose of setting the yawspeed?

bresch, 2023-05-23:
> Ok, I cannot just set the desired yaw only once, but I have to change it at every sampling step according to the sampling time, right?

Yes.

> I'm giving a yaw setpoint as `yaw = curr_yaw + yawspeed` and set `yawspeed = yawspeed` . Is this the right way of doing it? 

Yes, but you should also consider the sampling time and use the previous yaw setpoint, not the current value: `yaw = yaw + yawspeed * dt`

> If so, what the purpose of setting the yawspeed?

Yawspeed is a feedforward, it helps to have a better tracking. If you just set the `yaw`, the drone will slightly lag behind the setpoint.

BTW, if you just want to control the yaw rate, you can also send `yaw = NAN` and `yawspeed = yawspeed`. This will open the yaw loop and only control the yaw rate.