PX4/PX4-Autopilot #21626: PX4 wrong Altitude after rotating vehicle at startup

Edekheh, 2023-05-23:
## Describe the bug
When replacing the drone's battery, I often rotate the drone, and then after setting the drone to the takeoff position, I see that the drone has the wrong amsl altitude. I use mavros and look at topic /mavros/altitude 

## To Reproduce
Steps to reproduce the behavior:
1. Drone switched on
2. Rotate drone a few times
3. Place a drone for start
4. Drone have wrong altitude (AMSL)
5. Resetting PX4
6. Drone have correct altitude (AMSL)

## Expected behavior
The drone should detect that it is not calibrating altitude correctly by rotating it, so it should wait for a stable position and then recalibrate altitude. Alternatively, have an interface to reset altitude calibration (preferably as a mavros service)

bresch, 2023-05-26:
Do you maybe have a log to share? (logs.px4.io)

Sidzeppelin95, 2023-05-27:
Here's a possible solution:
The bug is that when replacing the drone's battery, rotating the drone and
then setting it to the takeoff position causes the drone to show the wrong
AMSL altitude on the /mavros/altitude topic. Resetting PX4 fixes this issue.
To avoid this issue, the drone should have a mechanism to detect incorrect
altitude calibration caused by rotating the drone, and recalibrate the
altitude once it is stabilized. Additionally, a mavros service for
resetting the altitude calibration could be implemented as an alternative
solution.
Implementing an interface to reset the altitude calibration via a mavros
service would provide a convenient way of addressing the issue for users,
without having to reset the entire PX4 system. It should be noted that this
bug might also be related to the drone's IMU calibration, so it would be
worth checking the IMU calibration status as well.
Checking the IMU calibration according to the drone's manufacturer's
instructions should be done routinely as part of pre-flight checks. It is
also important to ensure that the drone is on a flat and stable surface
prior to takeoff, as this can affect its altitude reading and calibration.
Additionally, the use of GPS altitude in addition to barometric altitude
can increase the accuracy of altitude readings.
In summary, to address the issue of the drone showing incorrect AMSL
altitude after battery replacement and rotation, the drone should have a
mechanism to detect incorrect altitude calibration and recalibrate the
altitude once stabilized. A mavros service for resetting the altitude
calibration could also be implemented as an alternative solution. Routine
IMU calibration checks and ensuring the drone is on a flat and stable
surface prior to takeoff should also be done. Using GPS altitude in
conjunction with barometric altitude can also improve altitude reading
accuracy.
Thanks & Regards
Siddhartha Sharma

Edekheh, 2023-05-27:
I will have logs on monday. 

As for now I'm resseting by simple calling : 
rosservice call /mavros/cmd/command "{broadcast: false, command: 246, confirmation: 0, param1: 1, param2: 0.0, param3: 0.0, param4: 0.0, param5: 0.0, param6: 0.0, param7: 0.0}"

This command will only reset px4, not mavros, after that everything works correctly.
I wonder if there is any possibility of automatic detection of amsl height calibration error, so that I do not have to call the service manually.
Thanks for fast responses !