PX4/PX4-Autopilot #21627: 1.14 beta - Unable to arm & position mode altitude not holding

junwoo091400, 2023-05-23:
This is from the original comment from @MDecarabas: https://github.com/PX4/PX4-Autopilot/issues/21358#issuecomment-1483580800

## Describe the bug
Manual Mode
Unable to arm because throttle is high when using sbus rc.
-Troubleshooting:
--Tested using telemetry radio with joystick and issue does not occur.
--Tested on 1.13.3 using sbus rc and issue does not occur

Altitude Mode
Only able to arm if throttle is pushed all the way down when using sbus rc. Once armed it will not lower the rpm of the motors
-Troubleshooting:
--Tested using telemetry radio with joystick and behavior does not occur.
--Tested on 1.13.3 using sbus rc and behavior does not occur

Stablized Mode
Same error and troubleshooting as in manual mode

Position Mode
Unable to reduce altitude when in mode. Drone will either keep level if sticks are fully down or rise if sticks are centered.
-Troubleshooting:
--Tested using telemetry radio with joystick and issue does not occur.
--Tested on 1.13.3 using sbus rc and issue does not occur

Attached below are flight logs for the altitude mode and position mode behavior described above. Let me know if I can provide any more clarifying information.

## Log Files and Screenshots
Altitude Mode: https://logs.px4.io/plot_app?log=c597875e-feab-40d4-a633-a709a72dbe99
Position Mode: https://logs.px4.io/plot_app?log=aaeae7c4-4171-472f-8c0e-34a35c91d326

## Drone (please complete the following information):
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

## Additional context
Add any other context about the problem here.

bresch, 2023-05-25:
Sounds like a problem with his RC calibration, no?

MaxBertus, 2023-05-25:
I have the same issue and the calibration procedure doesn't resolve the problem

Jaeyoung-Lim, 2023-05-25:
@junwoo091400 Are you sure virtual joystick is disabled in QGC?

abarcis, 2023-05-29:
Check https://github.com/PX4/PX4-Autopilot/pull/21576, maybe it resolves the problem. For me, the symptoms were the same when the throttle axis was inverted.

MaEtUgR, 2023-06-06:
Thanks for the reports!
I'm pretty sure this problem is fixed with: https://github.com/PX4/PX4-Autopilot/pull/21692
Let me double-check in the logs 👀
Sorry that it took so long 😇

MaEtUgR, 2023-06-06:
The log is really useful 👍
From the log:
```
RC_MAP_THROTTLE, 2
RC2_REV, -1.0
RC2_MAX, 1932
RC2_TRIM, 1932
```
It fits exactly the case described in https://github.com/PX4/PX4-Autopilot/issues/21575 and fixed in https://github.com/PX4/PX4-Autopilot/pull/21682 and soon on 1.14 in https://github.com/PX4/PX4-Autopilot/pull/21692

closed 2023-06-06 by MaEtUgR