PX4/PX4-Autopilot #21650: Unable to calibrate ESC with UAVCAN power module

TianmingLi, 2023-05-29:
## Describe the bug
Unable to calibrate ESC with UAVCAN power module. 
For cuav's v6x flight controller, it only provide a uavcan power module. follow the official esc calibration instructions will lead a failure esc calibration.

## To Reproduce

Follow official  esc calibration [document](https://docs.px4.io/main/en/advanced_config/esc_calibration.html#steps)

## Expected behavior

ESC calibration complete

## Log Files and Screenshots
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/main/en/getting_started/flight_reporting.html)).
- Upload the log to the [PX4 Flight Review](http://logs.px4.io/)
- Share the link to the log (Copy and paste the URL of the log)

Add screenshots to help explain your problem.

## Drone (please complete the following information):
- Flight controller: CUAV V6x
- QGC: V4.2.6.

## Additional context

Not sure this issue should be submited on PX4 or QGC

**[#5622](https://github.com/mavlink/qgroundcontrol/issues/5622)**

Sidzeppelin95, 2023-05-29:
Hi Tianming Li,
A possible solution for ESC calibration issue on cuav's v6x flight
controller is to perform the calibration manually using UAVCAN GUI Tool or
via the Mavlink console, as the official ESC calibration instructions may
not work. It's also recommended to ensure that the ESCs are properly
connected and powered, and that the firmware on both the flight controller
and ESCs are up to date.
Additionally, if the issue persists, it's best to seek help from the PX4
community for further assistance. It's important to provide a detailed
description of the problem and any relevant information, such as flight
logs and screenshots, to help others understand the issue and provide a
better solution.
It's worth noting that some ESCs may require specific calibration methods
or settings depending on their brand or model. As such, it's recommended to
consult the ESC manual or manufacturer's website for more detailed
guidance. Additionally, if the ESCs are damaged or malfunctioning,
calibration may not resolve the issue and they may need to be replaced.
Ultimately, proper calibration of the ESCs is crucial for safe and stable
drone operation.
Finally, it's important to ensure that all safety precautions are taken
when working with drones and related equipment. This includes wearing
appropriate personal protective equipment, such as safety glasses and
gloves, and following all relevant safety regulations and guidelines. By
taking the necessary precautions and following appropriate guidance, it's
possible to resolve the ESC calibration issue and achieve optimal drone
performance.
Thanks & Regards
Siddhartha Sharma

julianoes, 2023-05-29:
I'm not sure the LLM answer above is helpful.

@TianmingLi are you sure you actually need to calibrate your ESCs? I would not expect that UAVCAN ESCs need calibration. They might need setting up the min/max params and some bench testing to check it's working as expected but they don't need the sort of PWM max/min calibration that basic PWM ESCs require.

TianmingLi, 2023-05-29:
>  I would not expect that UAVCAN ESCs need calibration. 

Sorry for my unclear information.

Power module is UAVCAN protocol.

But ESC module is PWM protocol.

TianmingLi, 2023-05-29:
> Hi Tianming Li, A possible solution for ESC calibration issue on cuav's v6x flight controller is to perform the calibration manually using UAVCAN GUI Tool or via the Mavlink console, as the official [...]

[image]

By now, I used `actuator testing` pannel to calibrate ESCs.

1. Disconnect battery
2. Slid all motor to the top
3. Connect  battery
4. Wait `beep beep`
5. Slid all motor to the buttom

closed 2023-06-04 by TianmingLi

ghavan0517, 2023-09-19:
Hi Li,
We are also facing same error with you. 
After manually calibrating, how do you know that it is calibrated?

ghavan0517, 2023-09-26:
Thank you for the information above. 
Do you know what to do if all the info for 4motors are different even if the drone is horizontal?