PX4/PX4-Autopilot #21654: Airspeed failure in high wind cases

Jaeyoung-Lim, 2023-05-30:
## Describe the bug
I wanted to test a scenario for fixedwings, where there was excess wind (18m/s)

However, the vehicle fails a preflight check during flight and the airspeed is invalidated. This eventually ends up in the vehicle crashing.

@sfuhrer @tstastny FYI

## To Reproduce
Steps to reproduce the behavior:
1. Set the `windVelocityMean` to 18 in `windy.world` (https://github.com/PX4/PX4-SITL_gazebo-classic/blob/0d53638452e6da4371c12599e1a649608361811e/worlds/windy.world#L18)
2. Run SITL with `plane` model
```
make px4_sitl gazebo-classic_plane__windy
```
3. In the px4 shell, type in the following command
```
make px4_sitl gazebo-classic_plane__windy
```
5. See error
[image]
6. Vehicle crashes into the ground

## Expected behavior
NPFG is built to handle high wind cases and as long as the airspeed sensor is "working" it should not invalidate the airspeed sensor

## Log Files and Screenshots
Tested in SITL Gazebo classic
- Flight log: [log](https://review.px4.io/plot_app?log=0750a5d6-8769-42a6-b118-5c05293f3bb8)

ryanjAA, 2023-07-05:
This is interesting. Highest sustained wind I have flown in is about 16 m/s. It’s always a pita to takeoff since you need to cover the pitot, arm and turn quite a few checks off. One of the reasons airspeed checks are routinely disabled is to cope with high winds.

sfuhrer, 2023-07-05:
>It’s always a pita to takeoff since you need to cover the pitot, arm and turn quite a few checks off

You mean because of the preflight airspeed checks? The range of acceptable airspeeds was extended to FW_AIRSPD_MAX in 1.14. 
If you have airspeed failure false positives in high wind then let me know about them, they shouldn't happen.

sfuhrer, 2023-07-05:
@Jaeyoung-Lim have you also tried taking of in headwind? That would be a more realistic scenario in that much wind. 
Here it seems like the wind estimator doesn't initialize properly and then doesn't recover, likely because the airspeed gate size prevents further airspeed fusion.
[image]

This is heavily related to https://github.com/PX4/PX4-Autopilot/pull/21764.

closed 2023-11-07 by sfuhrer