PX4/PX4-Autopilot #21655: Is the Multicopter Control Architecture Diagram updated?

Deltarr350, 2023-05-30:
Hello, I would like to ask whether the diagram below found from the following link (under section Multicopter Control Architecture): (https://docs.px4.io/main/en/flight_stack/controller_diagrams.html) reflects the latest updated implementation in PX4 (using control allocation). 

[image]

The reason I am not sure is due to the output of the Angular Rate Controller consisting of deflection set points being passed to a mixer block. From my understanding (please correct me if I'm wrong), the output from the controllers should be thrust and torque setpoints which are to be passed to the control allocation module. If the diagram is outdated, can someone point me to where I can find the updated one? Thank you very much in advance.

bresch, 2023-05-31:
Hi @Deltarr350 , 
The variables are maybe not the best but the meaning is still correct:
- `/delta A_{sp}` = "torque_x"
- `/delta E_{sp}` = "torque_y"  
- `/delta R_{sp}` = "torque_z" 

For the thrust, `/delta T_{sp}` = "[0, 0, thrust_z]"

Deltarr350, 2023-06-21:
Hi @bresch, thank you very much for your response, helped me clarify things.

closed 2023-06-21 by Deltarr350