PX4/PX4-Autopilot #21658: PX4 Standard VTOL Gazebo Classic HITL Sensor Error

Deltarr350, 2023-05-31:
## Describe the bug
Hello I am currently trying to run a HITL simulation on Gazebo Classic 11 and Ubuntu 20.04 of the [standard_vtol_hitl](https://github.com/PX4/PX4-SITL_gazebo-classic/tree/2e3ed9bfb04d8865e59380afa575dd37b0d6c8e0/models/standard_vtol_hitl) model. Following the instructions found here: https://docs.px4.io/main/en/simulation/hitl.html , I was able to launch my custom model on Gazebo and also start QGroundControl successfully. The QGroundControl was also able to connect to my Pixhawk 4 Mini Board. However, the status on QGroundControl is 'Not Ready', with the following errors:
1. Critical: Preflight Fail: Airspeed invalid
2. Critical: Preflight Fail: height estimate error
3. Critical: Preflight Fail: Found 0 compass (required: 1)
4. EMERGENCY: Baro sensor #0 failure: Timeout

Additionally, the side pane indicates Magnetometer and AHRS are showing error status

## To Reproduce
Steps to reproduce the behavior:
1. Connect pixhawk board via usb
2. Modify the <serialDevice></serialDevice> field in the standard_vtol_hitl.sdf file accordingly
3. Set <hil_mode>1</hil_mode>
4. On terminal, run: 
- cd PX4-Autopilot
- DONT_RUN=1 make px4_sitl_default gazebo-classic
- source Tools/simulation/gazebo-classic/setup_gazebo.bash $h $(pwd) $(pwd)/build/px4_sitl_default
- gazebo Tools/simulation/gazebo-classic/sitl_gazebo-classic/worlds/hitl_standard_vtol.world
5. On a separate terminal window, run:
- ./QGroundControl.AppImage

## Expected behavior
QGroundControl shows 'Ready' status, no sensor errors

## Log Files and Screenshots
[image]
[image]

## Drone (please complete the following information):
- Standard VTOL HITL

## Additional context
N/A

vacoff, 2023-09-25:
I also have the exact issue on my setup: gazebo and Ubuntu 22. Can't find any solution.

NanyongGit, 2024-03-15:
Having this issue now on my jmavsim and Ubuntu 22

FilBk, 2024-04-29:
I'm still facing the same issues on gazebo classic and ubuntu 22.0
 I moved to HITL in  matlab simulink instead and i'm finding similar problems with sensor fails being displayed and navigator therefore changing it's state leading to messy simulation.

Zhenyu1Wu, 2025-03-16:
Having this issue now on my gazebo and Ubuntu 20.04

AzzamALMEKHLAFI, 2025-11-01:
Hi guys,Having this issue now on my gazebo and Ubuntu 22.04. Has anyone solved the problem?
@Deltarr350 @mrpollo @vacoff @FilBk @Zhenyu1Wu @