PX4/PX4-Autopilot #21668: ROS2 PX4 Colcon build notes/ideas

dagar, 2023-06-01:
PX4-Autopilot fully building in colcon (optionally)
- use msgs/ directly instead of px4_msgs (decide both datatype naming and topic mapping)
- in tree Nodes
 - Node that wraps micro XRCE DDS agent
 - Node that translates existing PX4 topics to ROS2 standard (odometry, sensors, etc)
   - could be a good place to handle logging, parameters bidirectionally
 - Node that "manages" existing PX4
   - full SITL use case
   - builds and deploys embedded PX4 (eg px4_fmu-v5x)
 - Node with OFFBOARD example as a starting point (pull in Jae's stuff)
- launch files for a few common scenarios
 - basic SITL + gazebo (+xrce agent)
 - real vehicle deployment (Linux + NuttX)

bradenwagstaff, 2023-06-01:
I took Jae's stuff and expanded on it a little bit. The main change I made is you can launch a simulation without QGC, fly that quad using keyboard teleop controls to control velocity setpoints instead of position setpoints. It also all runs under one launch file and runs a node that manages MicroDDS and PX4

I'm currently making an in depth tutorial targeted at new ROS2/PX4 users to help them get this example running and will film a tutorial walk through within the next week. A lot of this is motivated from the recent ROS call and the expressed need for tutorials that help people get started

https://github.com/ARK-Electronics/ws_ROS2_PX4_Simulation

Current example is setup as a new workspace, but I'm currently retrofitting it to be a package you can pull into any workspace

beniaminopozzan, 2023-06-14:
A few thoughts about having ROS 2 nodes in tree and building them with colcon

- We should have different packages in tree:
  - Core package: the current PX4-firmware
    > Node that "manages" existing PX4
full SITL use case
builds and deploys embedded PX4 (eg px4_fmu-v5x)

    in @dagar initial post
  - Communication packages: node wrapping micro XRCE DDS agent, ROS msgs / PX4 msgs conversions
    > Node that wraps micro XRCE DDS agent
Node that translates existing PX4 topics to ROS2 standard (odometry, sensors, etc)
could be a good place to handle logging, parameters bidirectionally
  
    in the initial post.
  - Examples / tutorial package
  - Simulation package (maybe)

  in this way we can keep everything well organized without splitting everything in different repos. Long term goal, having the possibility to do `sudo apt install ros-"some version"-px4` would be pretty nice. 
- Different packages must be placed in different folders and cannot be nested. I don't know if this can be accomplished without changing completely the PX4 tree like
  ```bash
  PX4-Autopilot
  ├── PX4-core                   # the current PX4-Autopilot
  ├── PX4-coms                 # communication package
  ├── PX4-tutorials             # tutorial package
  ```
  
FYI: At the moment we have the `ros2` target, and if we place PX4-Autopilot in a colcon workspace, `colcon build` will compile a package called `px4` containing the PX4 msgs (a copy of the `px4_msgs` package but with different name). Well, this is true if you are using ROS Foxy but the compilation is broken in Humble (see #21128) the issue seems to arise from calling `add_subdirectory` in the CMakeLists.txt file.

Slightly off-topic: this action https://github.com/ros-tooling/action-ros-ci may be useful to test future ROS 2 nodes (see https://github.com/PX4/px4_msgs/pull/26)

Jaeyoung-Lim, 2023-06-15:
> Different packages must be placed in different folders and cannot be nested. I don't know if this can be accomplished without changing completely the PX4 tree like

@beniaminopozzan Probably we can try to make each of the px4 module a colcon package? we can then have a small meta package that runs the rcs