PX4/PX4-Autopilot #21675: Dual RTK Heading

mrclwtsn, 2023-06-04:
I tried to use Dual RTK Heading function with 2 Holybro H-RTK F9P but failed. 

Here is details about my comfiguration:
Autopilot: Cube Orange
PX4 Version: 1.13.3 Stable
GPS_1_CONFIG: 201
GPS_2_CONFIG:202
GPS_UBX_MODE: Heading
EKF2_AID_MASK:1

 In documantion [https://docs.px4.io/main/en/gps_compass/u-blox_f9p_heading.html] its written that for GPS heading we should set EKF2_GPS_CTRL to 3 which is dual antenna heading. The problem is there is no parameters appers in QGC like that. Maybe the parameter name has changed?  I think that the documentation is not enough clear. Beside that should we disabled CAL_MAGX_PRIO parameters to use RTK heading function or not needed?

bresch, 2023-06-05:
Hi, [EKF2_GPS_CTRL](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#EKF2_GPS_CTRL) is a new parameter that got introduced in version 1.14. In 1.13, the GPS heading was activated using `EKF2_AID_MASK` (now deprecated). It's not the only thing that changes, we also improved the failure handling logic and I would highly encourage you to try the 1.14 beta version.

> Beside that should we disabled CAL_MAGX_PRIO parameters to use RTK heading function or not needed?

You can disable it if you don't want to fallback into mag heading if the GNSS heading data stops. Otherwise, when there is valid GNSS heading data, the mag is not used at all. You can also disable the mag fusion in [EKF2_MAG_TYPE](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#EKF2_MAG_TYPE) instead of disabling it at the driver level.

slgrobotics, 2023-06-17:
I was able to make dual-GPS Heading work (dual F9P), but had to tweak code a bit, here is related PR:
https://github.com/PX4/PX4-GPSDrivers/pull/131
Also, my settings are:
EKF2_GPS_CTRL 15
EKF2_MAG_TYPE 5
SENS_GPS_MASK 0  (Disable GPS blending)
SENS_GPS_PRIME 0  (always use Main GPS)
GPS_YAW_OFFSET 180  (Moving Base antenna in front of the Rover)
I set other parameters to disable IMU etc, but that's not so relevant.

slgrobotics, 2023-06-17:
Just in case, there's a related HPPOSLLH pull request - but it seems to be hanging in limbo so far:

https://github.com/PX4/PX4-GPSDrivers/pull/132
https://github.com/PX4/PX4-Autopilot/pull/21678

PepperOnioio, 2023-07-13:
@slgrobotics, I tried using 1.14 beta with the suggested GPS Parameters. But I can't achieve GPS heading, The UAV is only using Estimation and only achieving Fix type 5.
Current HW is the following:
FC: mRo H7 Zero
GPS: 2x mRo u-Blox ZED-F9 RTK L1/L2 GPS

The GPS' are connected to each other and each is connected to the FC(Main on GPS1 and Secondary on Serial 3).
Maybe I'm missing something. GPS heading worked on 1.12.3, but the error handling was bad.
Parameter File:
[GPS_test_071323.txt](https://github.com/PX4/PX4-Autopilot/files/12040203/GPS_test_071323.txt)

slgrobotics, 2023-07-13:
@PepperOnioio  - please make sure that you pulled the latest from https://github.com/PX4/PX4-Autopilot/tree/main  -as the important changes to _GPSDriver_ submodule and converting lat/lon/alt to "_double_" were merged in just hours ago (https://github.com/PX4/PX4-Autopilot/pull/21678)

Try "gps status" - it should show both GPS in fix_type "6" for "heading" to become valid. Secondary GPS should be your Moving Base, main - your Rover. Look at coordinates for both antennas - does the position shift agree with your antennas location?
Your Moving Base GPS must be connected to the RTCM feed from the Land base station.

It would be helpful to see a diagram of your connections, like in https://github.com/PX4/PX4-GPSDrivers/pull/131  - all ports, feeds etc.

And see if anything in ubx.cpp / ubx.h looks not applicable to your setup.

My startup file is here, maybe you find some useful settings there: 

https://github.com/slgrobotics/PX4-Autopilot/blob/main/ROMFS/px4fmu_common/init.d/airframes/50003_lawnMower1_config

PepperOnioio, 2023-07-14:
Thanks for getting back to me.
For context, we just want to achieve GPS Yaw, so we don't have a Land Base station.
GPS1 UART1 - Main is connected to the flight controller on UART4
GPS2 UART1 - Secondary is connected to the flight controller on UART8
GPS1 UART2 is connected to GPS2 UART2
And the GPS' are 70cm apart and the Main GPS is at 270 degree, whilst Secondary GPS is at 90 degrees.

[image]

As previously mentioned, we used this setup successfully with 1.12.3 Stable and even 1.13.3 Stable, but every time we jammed the GPS, the estimated yaw started rotating CCW and would stop, even after regaining GPS. Only after a restart, it would work correctly again.

I'll try again today with the newest firmware, and post the result.

slgrobotics, 2023-07-14:
@PepperOnioio  - Thanks, that makes things a bit more clear.
I would be specifically interested if _"gps status"_ shows Main GPS in _fix_type 6_, and _heading_ showing up as a number, not **nan**.

I am pretty sure that PX4 sends all necessary setup commands to F9P, but just in case I reset my two units (SparkFun F9P's) to factory settings.

I ran my lawnmower yesterday with freshly compiled code from the **"main"** branch, and everything worked fine. I run on Raspberry Pi 3B, so I can put tracing _printf_'s in the code and see, for example, if certain UBX messages arrive and are being processed by PX4 in ubx.cpp. Not sure if you have that option with your FC.

PepperOnioio, 2023-07-18:
@slgrobotics - I haven't been able to make it work yet. I did check the status on the GPS1 - and the heading is 0.
I don't think I can read the UBX messages through the flight controller.
[image]

slgrobotics, 2023-07-22:
@PepperOnioio  - I see _fix-type_ being 5 in your screen shot, which means "float". F9P would only deliver heading when _fix_type_ is 6 ("fixed"). Basically, until your Rover GPS is in "fixed" mode, you cannot expect heading to show up.

It's a rainy day here, and my lawnmower is parked. When I run it I will tale a look at MAVLINK messages and see what they show. So far I haven't done that, as I was looking at console messages - one of the advantages of having a Raspberry Pi as a controller.

slgrobotics, 2023-07-23:
@PepperOnioio  - I did two experiments with my setup.

In my case, same as yours, UART2 of both devices are connected together, TX->RX, RX<-TX - to allow RTCM corrections to flow from Moving Base to Rover.

1. With the Land Station ON, both Moving Base and Rover GPS'es have _fix_type_ 6. Heading ("yaw") appears on the main GPS. I have RTK precision (1.4 cm) on my Main GPS, and can see the same on the Moving Base.

```
pxh> gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] status: OK, port: /dev/ttyACM1, baudrate: 460800
INFO  [gps] sat info: disabled
INFO  [gps] rate reading: 		  2173 B/s
INFO  [gps] rate position: 		  4.75 Hz
INFO  [gps] rate velocity: 		  4.75 Hz
INFO  [gps] rate publication:		  4.75 Hz
INFO  [gps] rate RTCM injection:	  0.00 Hz
 sensor_gps
    timestamp: 1327926743 (0.003507 seconds ago)
    timestamp_sample: 0
    altitude_ellipsoid_m: 116.155800
    time_utc_usec: 1690140718874546
		...
	heading: -0.69779
    heading_offset: -3.14159
    heading_accuracy: 0.00988
    rtcm_injection_rate: 0.00000
    automatic_gain_control: 0
    fix_type: 6
    jamming_state: 2
    spoofing_state: 1
    vel_ned_valid: True
    satellites_used: 29
    selected_rtcm_instance: 0
    rtcm_crc_failed: False
    rtcm_msg_used: 2

INFO  [gps] 
INFO  [gps] Secondary GPS
INFO  [gps] protocol: UBX
INFO  [gps] status: OK, port: /dev/ttyACM0, baudrate: 460800
INFO  [gps] sat info: disabled
INFO  [gps] rate reading: 		  1878 B/s
INFO  [gps] rate position: 		  8.72 Hz
INFO  [gps] rate velocity: 		  8.72 Hz
INFO  [gps] rate publication:		  8.72 Hz
INFO  [gps] rate RTCM injection:	  0.00 Hz
 sensor_gps
      ...
    heading: nan
    heading_offset: -3.14159
    heading_accuracy: 0.00000
    rtcm_injection_rate: 0.00000
    automatic_gain_control: 0
    fix_type: 6
    jamming_state: 1
    spoofing_state: 1
    vel_ned_valid: True
    satellites_used: 28
    selected_rtcm_instance: 0
    rtcm_crc_failed: False
    rtcm_msg_used: 2

pxh> 
```

2. With the  Land Station OFF, Moving Base has _fix_type_ 4 and Rover GPS'es have _fix_type_ 6. Heading ("yaw") appears on the main GPS.

```
pxh> gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] status: OK, port: /dev/ttyACM1, baudrate: 460800
INFO  [gps] sat info: disabled
INFO  [gps] rate reading: 		  2150 B/s
INFO  [gps] rate position: 		  4.76 Hz
INFO  [gps] rate velocity: 		  4.76 Hz
INFO  [gps] rate publication:		  4.76 Hz
INFO  [gps] rate RTCM injection:	  0.00 Hz
 sensor_gps
      ...
    heading: -0.70024
    heading_offset: -3.14159
    heading_accuracy: 0.00989
    rtcm_injection_rate: 0.00000
    automatic_gain_control: 0
    fix_type: 6
    jamming_state: 2
    spoofing_state: 1
    vel_ned_valid: True
    satellites_used: 27
    selected_rtcm_instance: 0
    rtcm_crc_failed: False
    rtcm_msg_used: 2

INFO  [gps] 
INFO  [gps] Secondary GPS
INFO  [gps] protocol: UBX
INFO  [gps] status: OK, port: /dev/ttyACM0, baudrate: 460800
INFO  [gps] sat info: disabled
INFO  [gps] rate reading: 		  1752 B/s
INFO  [gps] rate position: 		  8.72 Hz
INFO  [gps] rate velocity: 		  8.72 Hz
INFO  [gps] rate publication:		  8.72 Hz
INFO  [gps] rate RTCM injection:	  0.00 Hz
 sensor_gps
		...
    heading: nan
    heading_offset: -3.14159
    heading_accuracy: 0.00000
    rtcm_injection_rate: 0.00000
    automatic_gain_control: 0
    fix_type: 4
    jamming_state: 1
    spoofing_state: 1
    vel_ned_valid: True
    satellites_used: 28
    selected_rtcm_instance: 0
    rtcm_crc_failed: False
    rtcm_msg_used: 2
```
[image]
[image]

slgrobotics, 2023-07-23:
@PepperOnioio  - BTW, fix_type 5 (float) in your case means that the Main GPS is receiving RTCMs from the Secondary, so the UART2's in both devices are initialized properly by GPSDriver code. But your GPSes don't seem to get to the state when they computed all necessary corrections and can deliver RTK precision (mode 6).

Takayama75, 2023-10-25:
> I was able to make dual-GPS Heading work (dual F9P), but had to tweak code a bit, here is related PR: [PX4/PX4-GPSDrivers#131](https://github.com/PX4/PX4-GPSDrivers/pull/131) Also, my settings are: EKF2_GPS_CTRL 15 EKF2_MAG_TYPE 5 SENS_GPS_MASK 0 (Disable GPS blending) SENS_GPS_PRIME 0 (always use Main GPS) GPS_YAW_OFFSET 180 (Moving Base antenna in front of the Rover) I set other parameters to disable IMU etc, but that's not so relevant.

Could you share the other parameters you set? I tried to disable compass and/or IMU. However, the arm was rejected by 0 Compass error etc.  Even if dual antenna GPS installed and EKF2_MAG_TYPE is set to 5, is compass still required for arm?

slgrobotics, 2023-10-25:
@Takayama75  - my guess is that EKF2 and arming checks are independent. There may be a flag/bit in a parameter that disables compass check when arming. My robot has magnetometer(s), but I make EKF2 ignore them. Arming check still sees them though.

darshankt, 2023-12-05:
Hi, I'm working with 1.13.0  & trying setup C-RTK dual antennas device as heading system. Based on my version of firmware i don't have EKF2_GPS_CTRL param and I have to EKF2_AID_MASK param for heading. Even though i have set EKF2_AID_Mask param for heading, my system still using compass magnetometers as heading. I don't know I need to setup any other parameters to along with EKF2_AID_MASK to enable heading system. Also how can I verify heading system using Dual antenna for Yaw estimation.

darshankt, 2023-12-10:
> Hi @darshankt <https://github.com/darshankt> I am interested in using the
> C-RTK duak antenna (CUAV?) too. Is the device now supported by PX4? The
> parameter you mentioned appears in firmware v1.14. Do you plan to test it?
>
> As it has the same function as the H-RTK Unicore UM982 (Dual Antenna), it
> is interesting to read these posts too.
> #22163 <https://github.com/PX4/PX4-Autopilot/issues/22163>
> https://discuss.px4.io/t/um982-isnt-providing-heading-data/35195

I'm still figuring it out, I have tested in v1.13.0 but no luck and now
planning move to v1.14.0. If you have PX4 with 1.14.0 please test with
parameters and let me know.

Thank you
Darshan

slgrobotics, 2024-05-19:
I filed a new issue related to EKF2 handling of RTK GPS data - https://github.com/PX4/PX4-Autopilot/issues/23157

This cured my rover of weaving around straight lines.

Also, just for anybody who may be guessing how to properly run GPS driver with dual F9P devices (it isn't that obvious ;-)):

``` gps start -d /dev/ttyACM1 -b 460800 -e /dev/ttyACM0 -g 460800```