PX4/PX4-Autopilot #21676: JSBSim-bridge Simulation Rate Question

JeongseokHyun, 2023-06-04:
Hello.

Currently, I want to put my custom module in JSBSim and run the simulation with PX4-Autopilot.

So, I try to run the JSBSim-PX4 SITL using JSBSim-bridge code.

I have a question about this code.

When I check the JSBSim-bridge.h file, I saw the _dt variable set 0.004 sec (250Hz).

As I know, PX4 and JSBSim can run based on 1kHz.

Is it absolute value? or not?

If I want to run the simulation in 1 kHz, Can I change it?

<jsbsim_bridge.h>
class JSBSimBridge {
 public:
  JSBSimBridge(JSBSim::FGFDMExec *fdmexec, ConfigurationParser &cfg);
  ~JSBSimBridge();
  void Run();

 private:
  bool SetFdmConfigs(ConfigurationParser &cfg);
  bool SetMavlinkInterfaceConfigs(std::unique_ptr<MavlinkInterface> &interface, TiXmlHandle &config);

  JSBSim::FGFDMExec *_fdmexec;  // FDMExec pointer
  ConfigurationParser &_cfg;

  std::unique_ptr<MavlinkInterface> _mavlink_interface;
  std::unique_ptr<SensorImuPlugin> _imu_sensor;
  std::unique_ptr<SensorGpsPlugin> _gps_sensor;
  std::unique_ptr<SensorBaroPlugin> _baro_sensor;
  std::unique_ptr<SensorMagPlugin> _mag_sensor;
  std::unique_ptr<SensorAirspeedPlugin> _airspeed_sensor;
  std::unique_ptr<ActuatorPlugin> _actuators;

  double _dt{0.004};
  double _realtime_factor{1.0};
  bool _result{true};
};

<jsbsim_bridge.cpp>
  // Config JSBSim FDM
  SetFdmConfigs(_cfg);
  _fdmexec->Setdt(_dt);
  _fdmexec->RunIC();

Jaeyoung-Lim, 2023-06-04:
> If I want to run the simulation in 1 kHz, Can I change it?

@JeongseokHyun Have you tried setting the timestep higher in the following?

https://github.com/PX4/px4-jsbsim-bridge/blob/68de2cc63ded9a0d6641d45e9eb3ed2b43454cba/src/jsbsim_bridge.cpp#L51

JeongseokHyun, 2023-06-05:
@Jaeyoung-Lim Yes, But I only tried to run JSBSim in windows without PX4.

Jaeyoung-Lim, 2023-06-05:
@JeongseokHyun Okay...but if you do what I suggest, it would answer your question?

closed 2023-06-10 by JeongseokHyun