PX4/PX4-Autopilot #21686: EKF log analysis scripts crashing

tylertian123, 2023-06-06:
## Describe the bug
Following the documentation [here](https://docs.px4.io/main/en/advanced_config/tuning_the_ecl_ekf.html#how-do-i-check-the-ekf-performance) I tried using the scripts `process_logdata_ekf.py` and `batch_process_logdata_ekf.py` at `Tools/ecl_ekf` to analyze EKF performance in my log files. However, when I ran it on my log file, the script exited with an exception about `'reject_mag_x'`.

I'm on PX4 v1.14.0-beta2 (commit 18898f1). Both the script and the logfile are from the same version.

## To Reproduce
Steps to reproduce the behavior:
1. Download the logfile attached
2. Run the script `python3 Tools/ecl_ekf/process_logdata_ekf.py <logfile>`
3. Script crashes without creating any output

## Expected behavior
I expected plots and metadata to be generated according to the documentation.

## Log Files and Screenshots
Log file: https://logs.px4.io/plot_app?log=eb988feb-08ae-4957-89a9-2257532c0916

If I modify the script to print out the exception stacktrace, this is what I get (by default the script exits after the `'reject_mag_x'` line):
```
found estimator_selector_status (multi-ekf) data
6 ekf instances
Using test criteria loaded from /home/tyler/utat/sae/sae_fs/PX4-Autopilot/Tools/ecl_ekf/check_level_dict.csv

estimator instance: 0
'reject_mag_x'
Traceback (most recent call last):
  File "process_logdata_ekf.py", line 186, in <module>
    main()
  File "process_logdata_ekf.py", line 168, in main
    test_results = process_logdata_ekf(
  File "process_logdata_ekf.py", line 124, in process_logdata_ekf
    master_status, check_status, metrics, airtime_info = analyse_ekf(
  File "/home/tyler/utat/sae/sae_fs/PX4-Autopilot/Tools/ecl_ekf/analyse_logdata_ekf.py", line 75, in analyse_ekf
    metrics = calculate_ecl_ekf_metrics(
  File "/home/tyler/utat/sae/sae_fs/PX4-Autopilot/Tools/ecl_ekf/analysis/metrics.py", line 22, in calculate_ecl_ekf_metrics
    innov_fail_metrics = calculate_innov_fail_metrics(
  File "/home/tyler/utat/sae/sae_fs/PX4-Autopilot/Tools/ecl_ekf/analysis/metrics.py", line 128, in calculate_innov_fail_metrics
    innov_fail_metrics[result] = calculate_stat_from_signal(
  File "/home/tyler/utat/sae/sae_fs/PX4-Autopilot/Tools/ecl_ekf/analysis/metrics.py", line 193, in calculate_stat_from_signal
    return stat_function(data[variable][in_air_det.get_airtime(dataset)])
KeyError: 'reject_mag_x'
```

## Drone (please complete the following information):
Large fixed wing. FCU is a CubePilot Orange, using the internal IMUs.

## Additional context
The main propulsion motor is quite large so it likely created a lot of magnetic interference, so there were a lot of EKF magnetometer errors when throttle changes significantly as can be see in the logs.

duckythescientist, 2023-06-07:
I don't know if this is correct, but I'm wondering if the logging format changed but the analysis script hasn't been updated.  `reject_mag_x` isn't a key in the log datasets, but `fs_bad_mag_x` is and from a cursory glance performs a similar function. If I change all references from `reject_mag_*` to `fs_bad_mag_*` under `Tools/ecl_ekf` then I can get the PDFs to generate without complaints. 

I also had to use the suggested fix in #21429

tylertian123, 2023-06-09:
@duckythescientist Thanks a lot for the tip! I managed to get it working for the time being by adding the changes you suggested. Hope this gets updated soon but at least in the meantime I can proceed with the analysis.

bresch, 2023-06-11:
@duckythescientist Yes, we forgot to update the analysis script. Could you please make a PR with your changes?

PR #21717 (2023-06-22): Fix reject_mag* and imu_coning* key errors
Replace reject_mag* with fs_bad_mag*
Apply the fix from #21429
Should also fix #21686
closed 2023-06-22 by dagar